srsran/srsRAN_4G #268: Some latency consideration in eNB software

kewinrausch, 2018-11-20:
Hi SRS,

I just want to report a curious behavior while running some performance tests in the eNB side of the srsLTE software suite. The behavior happens on the last version of the code. I tested it without any additional functionality which I included in the project (so in its otiginal state). [Link to the source here.](https://github.com/kewinrausch/empower-srsLTE/tree/upgrade)

First I finally have been able to connect and sustain traffic from more than three UEs! :smiley:

Second, I noticed that the RTT (well, latency in general) seems to scale less or more with the number of UE attached and committed to perform traffic. While they are silently connected I have the following situation:
```
64 bytes from 10.20.30.4: icmp_seq=289 ttl=64 time=16.4 ms
64 bytes from 10.20.30.4: icmp_seq=290 ttl=64 time=15.9 ms
64 bytes from 10.20.30.4: icmp_seq=291 ttl=64 time=34.5 ms
64 bytes from 10.20.30.4: icmp_seq=292 ttl=64 time=33.1 ms
64 bytes from 10.20.30.4: icmp_seq=293 ttl=64 time=32.9 ms
64 bytes from 10.20.30.4: icmp_seq=294 ttl=64 time=32.0 ms
64 bytes from 10.20.30.4: icmp_seq=295 ttl=64 time=31.7 ms
64 bytes from 10.20.30.4: icmp_seq=296 ttl=64 time=30.9 ms
64 bytes from 10.20.30.4: icmp_seq=297 ttl=64 time=29.8 ms
64 bytes from 10.20.30.4: icmp_seq=298 ttl=64 time=29.1 ms
64 bytes from 10.20.30.4: icmp_seq=299 ttl=64 time=28.5 ms
64 bytes from 10.20.30.4: icmp_seq=300 ttl=64 time=28.1 ms
64 bytes from 10.20.30.4: icmp_seq=301 ttl=64 time=27.8 ms
64 bytes from 10.20.30.4: icmp_seq=310 ttl=64 time=41.0 ms
```
Nothing strange here.

When one UE start to perform a bandwidth test with a common App or Iperf, the background (concurrent with the traffic generator) ping becomes like:

```
64 bytes from 10.20.30.7: icmp_seq=21 ttl=64 time=237 ms
64 bytes from 10.20.30.7: icmp_seq=22 ttl=64 time=207 ms
64 bytes from 10.20.30.7: icmp_seq=23 ttl=64 time=221 ms
64 bytes from 10.20.30.7: icmp_seq=24 ttl=64 time=209 ms
64 bytes from 10.20.30.7: icmp_seq=25 ttl=64 time=217 ms
64 bytes from 10.20.30.7: icmp_seq=26 ttl=64 time=167 ms
64 bytes from 10.20.30.7: icmp_seq=27 ttl=64 time=197 ms
64 bytes from 10.20.30.7: icmp_seq=28 ttl=64 time=186 ms
64 bytes from 10.20.30.7: icmp_seq=29 ttl=64 time=188 ms
```

With two UEs I have:
```
64 bytes from 10.20.30.7: icmp_seq=17 ttl=64 time=450 ms
64 bytes from 10.20.30.7: icmp_seq=18 ttl=64 time=409 ms
64 bytes from 10.20.30.7: icmp_seq=19 ttl=64 time=412 ms
64 bytes from 10.20.30.7: icmp_seq=20 ttl=64 time=431 ms
64 bytes from 10.20.30.7: icmp_seq=21 ttl=64 time=372 ms
64 bytes from 10.20.30.7: icmp_seq=22 ttl=64 time=372 ms
64 bytes from 10.20.30.7: icmp_seq=23 ttl=64 time=392 ms
64 bytes from 10.20.30.7: icmp_seq=24 ttl=64 time=431 ms
64 bytes from 10.20.30.7: icmp_seq=25 ttl=64 time=433 m
```

While with three UE I finally measured:
```
64 bytes from 10.20.30.7: icmp_seq=20 ttl=64 time=819 ms
64 bytes from 10.20.30.7: icmp_seq=21 ttl=64 time=933 ms
64 bytes from 10.20.30.7: icmp_seq=22 ttl=64 time=941 ms
64 bytes from 10.20.30.7: icmp_seq=23 ttl=64 time=911 ms
64 bytes from 10.20.30.7: icmp_seq=24 ttl=64 time=967 ms
64 bytes from 10.20.30.7: icmp_seq=25 ttl=64 time=946 ms
64 bytes from 10.20.30.7: icmp_seq=26 ttl=64 time=808 ms
```

**The ping was always working from the EPC (situated on the same machine of the eNB) and the first UE connected.** I expected the latency to increase, but not to follow the number of UE connected/transmitting.

Just in case I tried a similar approach using only laptops instead of srsLTE and UEs. I've got one client (in the EPC-like setup) pushing to two different server machines. The outcome of this esxperiment is the following:
```
64 bytes from 192.168.0.107: icmp_seq=43 ttl=64 time=0.252 ms
64 bytes from 192.168.0.107: icmp_seq=44 ttl=64 time=0.268 ms
64 bytes from 192.168.0.107: icmp_seq=45 ttl=64 time=0.251 ms
64 bytes from 192.168.0.107: icmp_seq=46 ttl=64 time=0.266 ms
64 bytes from 192.168.0.107: icmp_seq=47 ttl=64 time=0.256 ms
64 bytes from 192.168.0.107: icmp_seq=48 ttl=64 time=0.279 ms
64 bytes from 192.168.0.107: icmp_seq=49 ttl=64 time=2.02 ms
64 bytes from 192.168.0.107: icmp_seq=50 ttl=64 time=1.77 ms
64 bytes from 192.168.0.107: icmp_seq=51 ttl=64 time=1.93 ms
64 bytes from 192.168.0.107: icmp_seq=52 ttl=64 time=2.39 ms
64 bytes from 192.168.0.107: icmp_seq=53 ttl=64 time=1.91 ms
64 bytes from 192.168.0.107: icmp_seq=54 ttl=64 time=1.97 ms
64 bytes from 192.168.0.107: icmp_seq=55 ttl=64 time=1.76 ms
64 bytes from 192.168.0.107: icmp_seq=56 ttl=64 time=1.44 ms
64 bytes from 192.168.0.107: icmp_seq=57 ttl=64 time=1.98 ms
64 bytes from 192.168.0.107: icmp_seq=58 ttl=64 time=1.50 ms
64 bytes from 192.168.0.107: icmp_seq=59 ttl=64 time=1.92 ms
64 bytes from 192.168.0.107: icmp_seq=60 ttl=64 time=2.00 ms
64 bytes from 192.168.0.107: icmp_seq=61 ttl=64 time=1.80 ms
64 bytes from 192.168.0.107: icmp_seq=62 ttl=64 time=1.83 ms
64 bytes from 192.168.0.107: icmp_seq=63 ttl=64 time=2.47 ms
64 bytes from 192.168.0.107: icmp_seq=64 ttl=64 time=2.49 ms
64 bytes from 192.168.0.107: icmp_seq=65 ttl=64 time=1.81 ms
64 bytes from 192.168.0.107: icmp_seq=66 ttl=64 time=1.91 ms
64 bytes from 192.168.0.107: icmp_seq=67 ttl=64 time=1.38 ms
64 bytes from 192.168.0.107: icmp_seq=68 ttl=64 time=1.89 ms
64 bytes from 192.168.0.107: icmp_seq=69 ttl=64 time=1.09 ms
64 bytes from 192.168.0.107: icmp_seq=70 ttl=64 time=1.54 ms
64 bytes from 192.168.0.107: icmp_seq=71 ttl=64 time=2.51 ms
64 bytes from 192.168.0.107: icmp_seq=72 ttl=64 time=2.42 ms
64 bytes from 192.168.0.107: icmp_seq=73 ttl=64 time=1.68 ms
```

Basically you can't tell exatcly when the second iperf client is operating since the latency is always less or more the same.

While I'm not totally sure that this is a real issue, what worries me is that the performance degradates with the number of UE connected. This is something which can become worriesome in the future and probably identifies some sort of bottleneck in the stack.

If you want me to do additional tests just ask.

kewinrausch, 2018-11-21:
I have an additional guess here: could it be that this depends by the scheduler? RR metric assign one TTI per UE (If I recall it right), and thus every connecting UE cause the latency to increase.

igorskh, 2018-11-28:
Hi Kewin,

We are working with srsLTE eNB/UE, but normally only with one UE, later we can check in detail the latency with two UEs and post some results here. 

Correct me if I'm wrong but UEs are allocated transport blocks not TTIs, which means per 1 TTI more than 1 UE can be scheduled in this sense maybe decreasing a packet size on the traffic generator might reveal some interesting effects.

I think in this case the reason is in generated traffic and as for us it makes sense that the RTT might increase. 

However, it is not clear 
- how much traffic is generated (throughput per UE)
- what kind of traffic (TCP/UDP) 
- radio resources utilization (such as scheduled number of resource blocks and allocated TTIs). 
- do you achieve maximum throughput with a speed test for the current system bandwidth? 

Also, interesting to see, however, the end-to-end latency of traffic packets at least on the application layer.

My guess that since there is no prioritizing of traffic, so that ICMP packets of ping and UDP/TCP (whatever you use for the speed test) are equal, probably traffic generator packets are taking over. 

Regards,
Igor Kim
Research Assistant
Deutsche Telekom University of Applied Sciences Leipzig (HfTL)

kewinrausch, 2018-11-29:
> Correct me if I'm wrong but UEs are allocated transport blocks not TTIs, which means per 1 TTI more than 1 UE can be scheduled in this sense maybe decreasing a packet size on the traffic generator [...]

Yeah, in my case I can allocate multiple UE per single TTI. In Such case thus more than one UE can communicate at the same time and the latency should be affected so much. Imagine for example if you have 10 UE; with the current scheduler you will have to wait 10 ms before you get again an opportunity to transmit some data. Allocating multiple UE in the same TTI should avoid this.

> how much traffic is generated (throughput per UE)
> what kind of traffic (TCP/UDP)

TCP traffic in a 25 PRBs carrier with limited DL MCS to 16, which at best conditions gives around 7 Mbps. I use this fixed scenario to have a stable base to confront various measurements I'm doing.
Lately I started to use also UDP to see the effects.

> radio resources utilization (such as scheduled number of resource blocks and allocated TTIs).

I'm splitting the spectrum in two slices for the moment, so at the end one UE get 7 RBGs and the other 6.

> do you achieve maximum throughput with a speed test for the current system bandwidth?

Yeah, with only one UE I can achieve max throughput given the limitations I inserted in the eNB configuration (DL/UL MCS limitation).

> Also, interesting to see, however, the end-to-end latency of traffic packets at least on the application layer. My guess that since there is no prioritizing of traffic, so that ICMP packets of ping [...]

You are right; this is actually also my guess. Competition for generating IP packets can delay everything, but I wasn't expecting it to have such a behavior with the increasing number of UE. I repeated the experiment with an ethernet connection (Iperf TCP tests while ping was going), and the latency is stable regardless the number of clients.

This because TCP use the entire BW, which means that more generators just split the resources but should not add more latency: the system is already using all the resources. If you have one generator you will get X mbps, with 2 you get less or more X/2 for each. So the latency of the ping should be stable (maybe bad, but at least stable).

igorskh, 2018-12-06:
We did some evaluation of end-to-end latency, specifically eNB PDCP to UE PDCP, to evaluate the latency in more detail. 

In the figure below, you can see that in UL the latency has some pattern changing from higher to lower values. The time frames between 30 and 50 seconds and 60-80 s are when the second UE is producing the downlink traffic and the latency increases, not twice but for some value, while we also have the frequency domain separation as shown in #272. Time frame before 20 s - no traffic, 20-30, 50-60 one UE has traffic (the one which is measured).

[image]

We trace the latency only on one of two UEs, because we injected probes to the srsLTE and the other one is a commercial UE.

kewinrausch, 2018-12-07:
Hey there, nice job.

Since it seems you have a good setup already running, can you try give a run with two and three UE connected to the same eNB? You can run a ping (towards every UE or towards a single one) on the EPC machine, while having IPERF or similar traffic generator produce the data in the meantime.

andrepuschmann, 2019-01-02:
Hey guys, just a few thoughts on your delay observations. In summary, you're observing high RTT (i.e. ping) when more UEs are connected and when other traffic is running. First of all, that is roughly what I would have expected. Note that the eNB schedules all UEs in a round-robin fashion. So the "access delay" for each UE is a function of the number of active UEs. Now if a UE gets UL resources, it's up to the UE what data it sends. Since all data traffic goes through DRB1 we should have a look at RLC and PDCP of that bearer, and in particular, any queuing that is taking place. There is no queue in our PDCP but a fixed-size TX queue in all RLC entities. Since all traffic goes through that RLC queue it is obvious that a ICMP request/reply gets delayed by, e.g. UDP traffic generated by iperf, on the same bearer. Currently, the default queue size is 128 PDUs. That means if there is a full-rate UDP transmission taking place, i.e. full RLC Tx queue when the ping request is sent, this ping request sits in the RLC queue for 127 UL opportunities (assuming that each opportunity handles one RLC SDU). Feel free to play with default SDU queue length in rlc_am.h or rlc_um.h (at compile time at the moment).

kewinrausch, 2019-01-07:
So this is not an issue but rather a behavior triggered by default features that could be improved in the future. Thank you for the info!

closed 2019-01-21 by andrepuschmann