srsran/srsRAN_4G #272: Missed subframes with 2 UEs

igorskh, 2018-12-03:
Hello,

We have noticed some strange behavior while running downlink test on 2 UEs with iPerf3 and another traffic generator, second connected UE has some subframes missed (further evaluation showed that every 20th subframe is missing).

### Configuration
- srsLTE eNB and srsLTE EPC commit from Mon Oct 15 10:09:39 2018 +0200
- UEs: Samsung Galaxy Tab 3 and OnePlus 6 (and srsUE in the second scenario)
- Traffic generator: iperf3

### First scenario
we have tested the srsLTE eNB and commercial UEs and run iperf3 on the second connected UE:
`iperf3 -c 10.0.0.1 -P10`

The spectrum has a repeating pattern with empty subframes like in the figure below:
[image]

After we reconnect the second UE one or sometime two times the issue disapears:
[image]

### Second scenario
We tested with a branch with our changes in a code to use bandwidth parts and collect more information.
We used bandwidth parts to separate UEs in frequency domain and get the same issue (2nd connected UE is on the right hand side):
[image]

After reconnecting the second UE two times there is no missing subframes anymore:
[image]

We have collected subframes information and results shown that for the 2nd UE every 20th subframe is missing, while for the 1st UE every TTI is allocated. Zero denotes that there were two successive TTI and the value on the Y axis shows the index of the missed subframe (how many successive TTIs were before the missed one).
[image]

Have you noticed this issue before and what could be the reason?

Configuration file:
[enb_remote.conf.zip](https://github.com/srsLTE/srsLTE/files/2640146/enb_remote.conf.zip)

kewinrausch, 2018-12-05:
I am doing some measurement and testing in a similar situation and just reported an issue. I'm not sure is related somehow with what you have, but with 3 commercial UE I just found out that there was some kind of collision in DCI allocation (or at least is what I isolated right now with testing). 

Take a look to the new issue I [just open](https://github.com/srsLTE/srsLTE/issues/273).

You think that this could be related? Have you checked if you have in the log some trace about the scheduled complaining that is not possible to generate UL/DL DCI?

Cheers,
Kewin R.

igorskh, 2018-12-06:
> You think that this could be related? Have you checked if you have in the log some trace about the scheduled complaining that is not possible to generate UL/DL DCI?
>

Here is what we got from the logs:
```
10:36:30.135111 [MAC ] [W] [02041] SCHED: Could not schedule DL DCI for rnti=0x47, pid=2, cfi=3
10:36:30.155194 [MAC ] [W] [02061] SCHED: Could not schedule DL DCI for rnti=0x47, pid=2, cfi=3
10:36:30.175113 [MAC ] [W] [02081] SCHED: Could not schedule DL DCI for rnti=0x47, pid=2, cfi=3
10:36:30.195048 [MAC ] [W] [02101] SCHED: Could not schedule DL DCI for rnti=0x47, pid=2, cfi=3
10:36:30.215129 [MAC ] [W] [02121] SCHED: Could not schedule DL DCI for rnti=0x47, pid=2, cfi=3
```
A lot of these messages. When we reconnect the UE and it gets the RNTI 0x48 or 0x49 for example, we do not see this warning so often and do not observe this problem anymore. But I am not yet sure how it can be related to the RNTI as you described in your issue, but we will continue to look into this problem.

andrepuschmann, 2019-04-25:
Closing issue. Please reopen if it still exists in new release.

andrepuschmann, 2019-04-26:
I was wrong. The issue hasn't been addressed in new version yet.

andrepuschmann, 2019-05-26:
Solved in next release. Thanks again for reporting.

closed 2019-05-26 by andrepuschmann