srsran/srsRAN_4G #273: Collisions in DL DCI allocations

kewinrausch, 2018-12-05:
Hey srsLTE guys,

I am currently doing some though measurements with multiple legacy UEs connected through srsENB. The core network is provided by srsEPC. My setup is default (last version), eNB and EPC in the same machine, CFI 3, 25 PRBs, with DL max MCS locked at 16 (the problem still is visible with unlocked DL MCS, so MCS are not affecting this).

I developed a MAC scheduler for Vertical slice support (5G concept), that allow to reserve part of the spectrum to different slices ([link here](https://github.com/5g-empower/empower-srsLTE/blob/agent/srsenb/src/mac/scheduler_RAN.cc)). Every slice then schedule UE belonging to them in a RR fashion.

Measurement with 2 UE with combination of 2 slices and different slice size allocation where all good and the result were the expected ones. But when I connect three differents UE in three different slices a strange behavior happens. The three UEs are scheduled all in the same TTI, each of them gets around 4 RBGs (4 + 4 + 5 = 13 RBG, the full spectrum in 5MHz).

Despite every UE having similar resource allocation, one of the UEs (the 1st) was always doing really bad (picture 1). At some point I realized that shutting down UE3 allowed UE1 to reach its perfect bandwidth (suspicious behavior). After some try and print attempts I noticed that DL DCI will generate lof of collisions and detection of "_no free locations_" for PDCCH space where the DCIs should be stored to describe the data profile (see [here](https://github.com/srsLTE/srsLTE/blob/master/srsenb/src/mac/scheduler.cc#L1050)).

I digged up till `srslte_pdcch_ue_locations_ncce_L` procedure in `pdcch.c` of srsLTE library and confronted the algorithm there with the specifications. Everything seems fine at a first look, but somehow (I still have to find out why) location collides with one another for RNTIs that are somehow contiguous. In my case I found out through testing that 0x46 and 0x48 have such problem while pre-generating the NCCE locations for various aggregation levels.

I did a quick fix in the `mac.cc` file of the eNB, switching the RNTI selection from continuous to another behavior like `last_rnti += 3` and saw that now performance in the same exact conditions are proper and less or more stable (picture 2).

I hope my description is clear enough; I'm at the end of some heavy days. 😄 
Feels free to ask me additional information about this problem.

[image]
[image]

igorskh, 2018-12-06:
We also noticed that DCI collision happens in downlink, specifically in function `sched_ue::pucch_sr_collision(uint32_t current_tti, uint32_t n_cce)` which is called from `sched::generate_dci` function for the RNTI 0x47. This indeed explains missed subframe in our issue #272 and degraded throughput that you described.

We've checked with srsUE it is sending the scheduling request, but i cannot figure out why is it causing the collision and only with specific RNTIs.

andrepuschmann, 2018-12-07:
Thanks for reporting and the analysis provided. We'll look into ASAP.

andrepuschmann, 2019-04-25:
Closing issue. New release contains an improved scheduler. Please reopen if it still persists.

andrepuschmann, 2019-04-26:
I was wrong. The issue hasn't been addressed in new version yet.

andrepuschmann, 2019-05-26:
Solved in next release. Thanks again for reporting.

closed 2019-05-26 by andrepuschmann