srsran/srsRAN_4G #279: Cell_search cannot change frequency

xieyaxiongfly, 2018-12-14:
Dear all,

I got a problem running the cell_search example. 

I am using USRP X310 (I am not sure, whether srsLTE works with X310 or not). 

When I run the cell_search -b 1 - v, I get the error message as follow:

Opening RF device...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.13.0.1-release
Opening USRP with args: type=x300,master_clock_rate=184.32e6
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1299 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1300 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
nof_freqs: 599
[INFO]: Set rf_freq to 2110.000 MHz
[  0/599]: EARFCN 0 Freq. 2110.00 MHz looking for PSS.

[INFO]: Setting sampling frequency 1.92 MHz for PSS search
[INFO]: Starting receiver...
[INFO]: CP-CFO: estimated=0.066649, mean=0.066649
[INFO]: PSS: id=0, peak_pos=670, peak_value=1.339536
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.214404, mean=0.081424
[INFO]: PSS: id=0, peak_pos=7065, peak_value=1.215625
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.195840, mean=0.092866
[INFO]: PSS: id=0, peak_pos=488, peak_value=1.096967
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=-0.024149, mean=0.081164
[INFO]: PSS: id=0, peak_pos=5851, peak_value=1.312697
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=-0.180149, mean=0.055033
[INFO]: PSS: id=0, peak_pos=5932, peak_value=1.135440
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=-0.059304, mean=0.043599
[INFO]: PSS: id=0, peak_pos=6846, peak_value=1.482998
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.455896, mean=0.084829
[INFO]: PSS: id=0, peak_pos=8074, peak_value=1.089502
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.226347, mean=0.098981
[INFO]: PSS: id=0, peak_pos=8230, peak_value=1.140006
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.127405, mean=0.101823
[INFO]: PSS: id=0, peak_pos=3097, peak_value=1.189355
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.161552, mean=0.107796
[INFO]: PSS: id=0, peak_pos=2662, peak_value=1.014584
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.053675, mean=0.053675
[INFO]: PSS: id=1, peak_pos=8655, peak_value=1.010826
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.154269, mean=0.063734
[INFO]: PSS: id=1, peak_pos=595, peak_value=1.028761
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.381641, mean=0.095525
[INFO]: PSS: id=1, peak_pos=7932, peak_value=1.038318
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.181101, mean=0.104082
[INFO]: PSS: id=1, peak_pos=1918, peak_value=1.067029
[... 166 lines cut ...]
[INFO]: CP-CFO: estimated=0.063659, mean=0.025552
[INFO]: PSS: id=2, peak_pos=3806, peak_value=1.101132
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.012675, mean=0.024265
[INFO]: PSS: id=2, peak_pos=6016, peak_value=1.101061
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: CP-CFO: estimated=0.018485, mean=0.023687
[INFO]: PSS: id=2, peak_pos=8765, peak_value=1.090525
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
[INFO]: Set rf_freq to 2110.200 MHz
[  1/2]: EARFCN 2 Freq. 2110.20 MHz looking for PSS.

[INFO]: Setting sampling frequency 1.92 MHz for PSS search
[INFO]: Starting receiver...
Error timed out while receiving samples from UHD.
Error receiving samples
Error calling srslte_ue_sync_work()
Error searching cell
Error searching cell

From the message, I found that the it is not about the 2110.20MHz frequency. The cell search just cannot loop. If the program reconfigure the center frequency, the USRP cannot work and collect the IQ data. 

Can anyone kindly help me with this problem?

Thanks in advance.

andrepuschmann, 2018-12-17:
Doesn't look like a srsLTE error. Rather UHD or USRP in general. Please check if your daugtherboard supports that band at all.

closed 2018-12-17 by andrepuschmann

xieyaxiongfly, 2018-12-17:
Thanks for your advice. I have checked my hardware. The daughterboard (UBX160) I am using supports 10MHz-6GHz frequency band. 

Can I ask a few more questions?

1, can srsLTE works with USRP X310? 

2, As you can see that I can search the first frequency, but fails when switching to the next frequency. Is there any possible reason that will cause such a problem?

xieyaxiongfly, 2018-12-18:
I also tried multiple different USRP X310. I got the same error messages.

xieyaxiongfly, 2018-12-18:
I also tried  "./pdsch_ue -f 1800e6" which searches for the cell at the frequency of 1.8GHz repeatly, I got the following message: 

Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.13.0.1-release
Opening USRP with args: type=x300,master_clock_rate=184.32e6
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1305 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1320 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Starting AGC thread...
Tunning receiver to 1800.000 MHz
Searching for cell...
Could not find any cell in this frequency
Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 1 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 2 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 3 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 4 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 5 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 6 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 7 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 8 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 9 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 10 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Could not find any cell in this frequency
Cell not found after 11 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Error timed out while receiving samples from UHD.
Error receiving samples
Error calling srslte_ue_sync_work()
Error searching cell
Error searching cell
Error searching for cell

I repeat the above process multiple times and found that the error appears after a random number of trials (in the above example, it is 11 trials). 

I feel that the code has some problem when running on X310 USRP hardware. 

Could you kindly provide some advices for it? Is chaning the hardware the only solution for it?

andrepuschmann, 2018-12-18:
We've the X310 running in many setups. Note that it usually requires different power levels than, for example, the B210. Also, we're running on UHD 3.9.7 LTS.

igorskh, 2018-12-18:
We use USRP X310 for srsLTE UE and eNB without any problems with UHD 3.10, it runs stable for hours.

There are might be several issues, but it doesn't seem to be related to srsLTE:
1. Try to run with a lower version of UHD (3.9/3.10).
2. Check the connection from the PC to the USRP.
3. Might a performance issue of the PC itself (low performance in general, other processes running).
4. At least with the version of UHD we use the application should be run as root e.g. `sudo ./pdsch_ue -f 1800e6`.

xieyaxiongfly, 2018-12-19:
I change the UHD to 3.9 and it works!! 

Thanks for your advice. It really helps a lot!