srsran/srsRAN_4G #285: Unexpected dependancy of uplink BLER on MCS.

legden, 2018-12-20:
In our test equipment we use srsLTE 18.09 with b200mini and commercial UE, enb.conf file attached. We have no errors on downlink channel, but very high BLER on uplink which vary from 0% up to 70% and more while MCS vary from 19 to 23.

------DL------------------------------UL----------------------------------
rnti  cqi    ri   mcs  brate   bler   snr   phr   mcs  brate   bler    bsr
46   15.0     0  27.0  23.7M     0%  28.4  40.0  21.6  1.05M    17%    0.0
46   15.0     0  27.0  24.4M     0%  29.2  40.0  21.1  1.06M    17%    0.0
46   15.0     0  26.9  3.13M     0%  29.9  40.0  20.0  38.6k    90%    0.0

Trying to assign different fixed values to pusch_mcs in configuration file we observe that BLER depends on MCS in unexpected manner. For example with MCS fixed to 23 we have no errors at all and good constant throughput.

------DL------------------------------UL----------------------------------
rnti  cqi    ri   mcs  brate   bler   snr   phr   mcs  brate   bler    bsr
49   15.0     0  9.26   514k     0%  34.5  22.7  23.0  22.8M     0%   128k
49   15.0     0  9.12   434k     0%  33.9  21.1  23.0  22.8M     0%   128k
49   15.0     0  9.24   457k     0%  33.8  21.4  23.0  22.8M     0%   132k

But when MCS is 20 there are errors.

------DL------------------------------UL----------------------------------
rnti  cqi    ri   mcs  brate   bler   snr   phr   mcs  brate   bler    bsr
46   15.0     0  9.00  1.37k     0%  30.3  40.0  20.0  72.3k    50%    0.0
46   15.0     0  9.00  1.37k     0%  30.0  40.0  20.0  56.9k    55%    0.0
46   15.0     0  9.00  1.37k     0%  30.5  40.0  20.0  72.3k    50%    0.0

This behaviour is same for two UE of different manufactures.

[enb.conf.txt](https://github.com/srsLTE/srsLTE/files/2699073/enb.conf.txt)

andrepuschmann, 2018-12-21:
Can you send a short log for both cases with PHY level set to info, please?

legden, 2018-12-24:
Sorry for delay. There logs for cases with mcs fixed to 23 and 20.
[enb_mcs23.log](https://github.com/srsLTE/srsLTE/files/2706898/enb_mcs23.log)
[enb_mcs20.log](https://github.com/srsLTE/srsLTE/files/2706900/enb_mcs20.log)

akhila-s-rao, 2020-03-11:
What was the accepted solution before this was closed ? I have the same problem.

andrepuschmann, 2020-03-11:
Do you see the same issue with the 19.12? If so, could you send a set of logs?

legden, 2020-03-12:
It seems to be fixed in 18.12 in my case. But rather not exactly by commit 2bc4d70. Unfortunely have no possibility to check it right now with 19.12 due to lack of hardware.