srsran/srsRAN_4G #290: Error calling srslte_ue_sync_work()

674342860qq, 2019-01-07:
Hi,
There are no problems when I run srsepc and srsenb,But when I run srsue,I got errors like:

Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
.Error timed out while receiving samples from UHD.
Error receiving samples
Error calling srslte_ue_sync_work()
Error searching cell
Resetting Radio...
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX
.Error timed out while receiving samples from UHD.
Error receiving samples
Error calling srslte_ue_sync_work()
Error searching cell
Resetting Radio...
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX
.Error timed out while receiving samples from UHD.
Error receiving samples
Error calling srslte_ue_sync_work()
Error searching cell
Resetting Radio...

Any advice is welcome!

andrepuschmann, 2019-01-07:
Can you please provide more info regarding this? What version are you using? What RF front end?

674342860qq, 2019-01-07:
Hi,
thank you for your quick reply.
I install the srsLTE from https://github.com/srsLTE/srsLTE ,I guess it is the latest version.
I am using USRP x300,SBX 120(subdev A RF front 0),Ubuntu 16.04.
I think UHD block is normal because it works when I use GNURadio.When I run srsepc and srsenb ,two LEDs of the SBX flash normally,but when I run srsue,the two LED don't flash.

674342860qq, 2019-01-09:
Could you please give me more help?.. @andrepuschmann 
The errors are:

Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
.Error timed out while receiving samples from UHD.
Error receiving samples
Error calling srslte_ue_sync_work()
Error searching cell
Resetting Radio...
Error timed out while receiving samples from UHD.
SYNC:  Receiving from radio while in IDLE_RX

frankist, 2019-01-10:
> Sorry,I forget to tell that I git the code from the master banch and UHD
> version is 3.13.0

Hey,

Can you try with the UHD 3.9 LTS version?

Regards,
Francisco

674342860qq, 2019-01-11:
Hi,
Now I can run srsUE by running another laptop with i7 cpu...
But I can not connect to the srsENB on the other PC.
I got errors about fail to attach

Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-0.4 KHz
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=27, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=1
Scheduling request failed: releasing RRC connection...
Random Access Transmission: seq=43, ra-rnti=0x2
Random Access Transmission: seq=32, ra-rnti=0x2
Random Access Transmission: seq=6, ra-rnti=0x2
Random Access Transmission: seq=31, ra-rnti=0x2
Random Access Transmission: seq=17, ra-rnti=0x2
Random Access Transmission: seq=27, ra-rnti=0x2
Random Access Transmission: seq=32, ra-rnti=0x2
Random Access Transmission: seq=32, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=28, ra-rnti=0x2
Warning: Detected Radio-Link Failure
RRC Connection Reestablishment
Random Access Transmission: seq=4, ra-rnti=0x2
Random Access Transmission: seq=47, ra-rnti=0x2
RRC IDLE
Failed to Attach
  
Could anyone give me some advice?
Thanks..

674342860qq, 2019-01-11:
sorry ,here is the ue.log

[ue.log](https://github.com/srsLTE/srsLTE/files/2748971/ue.log)

andrepuschmann, 2019-01-21:
You have isues with your radio and hardware. Loads of lates.

closed 2019-01-21 by andrepuschmann