srsran/srsRAN_4G #301: Building on Ubuntu 14.04.5: error: redefinition of 'struct in6_pktinfo'

axilirator, 2019-01-26:
First of all, yes, I know I am using an ancient version of Ubuntu. Still in a process of moving to Arch Linux.

## Issue Description ##
Building the latest (release_18_12 tag) srsLTE from source fails with the following messages:

> Scanning dependencies of target srsue_upper
> [ 78%] Building CXX object srsue/src/upper/CMakeFiles/srsue_upper.dir/gw.cc.o
> In file included from /home/foobar/srsLTE/srsue/src/upper/gw.cc:35:0:
> /usr/include/linux/ipv6.h:19:8: error: redefinition of ‘struct in6_pktinfo’
>  struct in6_pktinfo {
>         ^
> In file included from /home/foobar/srsLTE/lib/include/srslte/phy/io/netsink.h:40:0,
>                  from /home/foobar/srsLTE/lib/include/srslte/srslte.h:81,
>                  from /home/foobar/srsLTE/lib/include/srslte/common/common.h:81,
>                  from /home/foobar/srsLTE/lib/include/srslte/common/buffer_pool.h:42,
>                  from /home/foobar/srsLTE/srsue/hdr/upper/gw.h:31,
>                  from /home/foobar/srsLTE/srsue/src/upper/gw.cc:28:
> /usr/include/netinet/in.h:534:8: error: previous definition of ‘struct in6_pktinfo’
>  struct in6_pktinfo
>         ^
> In file included from /home/foobar/srsLTE/srsue/src/upper/gw.cc:35:0:
> /usr/include/linux/ipv6.h:24:8: error: redefinition of ‘struct ip6_mtuinfo’
>  struct ip6_mtuinfo {
>         ^
> In file included from /home/foobar/srsLTE/lib/include/srslte/phy/io/netsink.h:40:0,
>                  from /home/foobar/srsLTE/lib/include/srslte/srslte.h:81,
>                  from /home/foobar/srsLTE/lib/include/srslte/common/common.h:81,
>                  from /home/foobar/srsLTE/lib/include/srslte/common/buffer_pool.h:42,
>                  from /home/foobar/srsLTE/srsue/hdr/upper/gw.h:31,
>                  from /home/foobar/srsLTE/srsue/src/upper/gw.cc:28:
> /usr/include/netinet/in.h:541:8: error: previous definition of ‘struct ip6_mtuinfo’
>  struct ip6_mtuinfo
>         ^
> make[2]: *** [srsue/src/upper/CMakeFiles/srsue_upper.dir/gw.cc.o] Error 1
> make[1]: *** [srsue/src/upper/CMakeFiles/srsue_upper.dir/all] Error 2
> make: *** [all] Error 2

Please see both cmake and make logs attached.

[srslte_cmake.log](https://github.com/srsLTE/srsLTE/files/2798612/srslte_cmake.log)
[srslte_make.log](https://github.com/srsLTE/srsLTE/files/2798611/srslte_make.log)

andrepuschmann, 2019-01-26:
Yes that's an incompatibility with and libc header defining that struct twice. Try this patch if that helps.
[ipv6.txt](https://github.com/srsLTE/srsLTE/files/2799469/ipv6.txt)

closed 2019-01-28 by andrepuschmann

Keskebeu, 2019-12-12:
I have the same issue in Ubuntu 14.04.