srsran/srsRAN_4G #304: UHD code 11, libsrsrlte_rf.so missing

CloudAntoni, 2019-02-01:
We have made the srsLTE and when we execute " sudo srsenb" or "srsue" and gives these errors: (not happening with the srsepc).

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/lab138/.config/srslte/enb.conf...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.13.1.0-release
Opening USRP with args: 
**Error opening UHD: code 11
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto**

We just founded the uhd error code: 44 on the community site, but we don't know how to solve it.
Also, we don't know how to find the library: **"libsrslte_rf.so".**

We are using lUbuntu 18.04 and we are using a N210, the last release from the srsLTE and the latest versions for the libraries.

In addition we found the same problem as others when we do the cmake ../ : 
**"no package polarssl"
"no package mbedtls"
"no package libpcslite"
"Could not find MKL"** 
But the compilation and all the test were 100% passed.

Sorry for the long text...

Thank you,

Antoni

gabriel-lando, 2019-02-04:
Run `sudo ldconfig` and try again

GonCasMor, 2019-02-05:
It worked with that command.
Thank you so much @gabriel-lando.

GonCasMor, 2019-02-06:
Sorry but we still have errors with our srsLTE.
When we execute "**sudo srsenb**" we get:

Built in Release mode using 18.12.0.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/lab138/.config/srslte/enb.conf...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.13.1.0-release
Opening USRP with args: 
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
[ERROR] [DBMGR] **The daughterboard manager encountered a recoverable error in init.**
Loading the "unknown" daughterboard implementations to continue.
The daughterboard cannot operate until this error is resolved.
LookupError: KeyError: key "0" not found in dict(i, N14adf4360_regs_t17prescaler_value_tE)
[WARNING] [MULTI_USRP] The hardware does not support the requested RX sample rate:
Target sample rate: 1.920000 MSps
Actual sample rate: 1.923077 MSps

[WARNING] [MULTI_USRP] The hardware does not support the requested TX sample rate:
Target sample rate: 1.920000 MSps
Actual sample rate: 1.923077 MSps

Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
Setting Sampling frequency 11.52 MHz

Warning TX/RX time offset has not been calibrated for device uhd_unknown. Set a value manually

==== eNodeB started ===
Type <t> to view trace
Failed to connect to MME - retrying in 10 seconds
OError timed out while receiving samples from UHD.
LLLLLLLLLLLLLLLLLLLLLLLLLLLLError timed out while receiving samples from UHD.
(and so on..)
LLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLFailed to connect to MME - retrying in 10 seconds
Error timed out while receiving samples from UHD.

Could you help us?

Thank you so much for your support

andrepuschmann, 2019-02-06:
The N210 is not supported.

closed 2019-02-06 by andrepuschmann