srsran/srsRAN_4G #306: memory issue

bluecat1024, 2019-02-02:
The asn.1 conversion to pre-defined arrays in struct(bit string, Octet,string, dynamic seq of, etc), may cause the size of an asn.1 PDU very large.

sizeof a s1ap pdu, is over a million bytes even if you limited the array length.

That means you have to flushes all these bytes to zero before filling the packet, and that's a huge over head; Moreover it may incur Segmentation Fault, and srsLTE crashes.

frankist, 2019-02-04:
Hey,
Considering that the new asn1 library performs heap allocations to save memory space, you should avoid calling "unsafe" bzero(...) and memcpy(...) operations. Use instead the C++ overloaded copy operators/constructors.

We measured the space consumed by each rrc asn1 msg structure and it never went over a million of bytes, unless you allocate bit/octet strings of that size.

alvasMan, 2019-02-04:
Hi, you are talking specifically about the S1AP only correct? Not RRC, not anything else?

Which version of the software are you using and what is your system setup?

frankist, 2019-02-04:
Are you talking specifically about the s1ap or the rrc? The s1ap currently does not contain pre-defined array types (e.g. bitstrings, octet strings, etc.)

With regard to the rrc asn1 library, it performs heap allocations to save memory space. We measured the space consumed by each rrc asn1 msg structure and it never went over a million of bytes, unless you allocate bit/octet strings of that size. 

You should avoid calling "unsafe" bzero(...) and memcpy(...) operations with the new rrc asn1 lib. Use instead the C++ overloaded copy operators/constructors. This is the likely the cause for the SegFault and the unbounded memory consumption you are witnessing.

bluecat1024, 2019-02-05:
Actually the s1ap asn.1 structures do have pre-defined arrays? for example a[16314] or something
And I believe the following sentense is still in the newest s1ap:
bzero(&pdu, sizeof(LIBLTE_S1AP_S1AP_PDU_STRUCT));
If you choose to output the sizeof, sizeof(LIBLTE_S1AP_S1AP_PDU_STRUCT) is over a million.

bluecat1024, 2019-02-05:
The new RRC is free of this issue; but I suppose other asn1 libs and their related upper codes are still not.

frankist, 2019-02-06:
Yes, the current s1ap asn1 lib still utilises the stack quite intensively
for simple arrays. We will look into the issues with its pdu sizes.

You also mentioned having crashes. Can you give a bit more context (e.g.
code snippet) regarding this issue?

Regards,
Francisco

closed 2019-04-24 by andrepuschmann