srsran/srsRAN_4G #307: Error compiling srsLTE under Ubuntu 16.04

GinesGarcia, 2019-02-04:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
[Describe the issue in detail]
I cannot compile srsLTE under Ubuntu 16.04

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Ubuntu 16.04

## Expected Behavior ##
[What you expect to happen]
Compile srsLTE

## Actual Behaviour ##
[What happens instead e.g. error message]

make -j 2 

[ 51%] Linking C executable pss_usrp
/usr/bin/ld: warning: libqwt.so.6, needed by /usr/local/lib/libsrsgui.so, not found (try using -rpath or -rpath-link)
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::setTrackerPen(QPen const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotCurve::setStyle(QwtPlotCurve::CurveStyle)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotCanvas::staticMetaObject'
/usr/local/lib/libsrsgui.so: undefined reference to `vtable for QwtLinearScaleEngine'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::qt_metacall(QMetaObject::Call, int, void**)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotPicker::trackerTextF(QPointF const&) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtEventPattern::setMousePattern(unsigned int, int, int)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::setZoomBase(QRectF const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::event(QEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::widgetKeyReleaseEvent(QKeyEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::axisScaleDiv(int)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::qt_metacall(QMetaObject::Call, int, void**)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtLinearColorMap::colorIndex(QwtInterval const&, double) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::minimumSizeHint() const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::enableAxis(int, bool)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtRasterData::setInterval(Qt::Axis, QwtInterval const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::legendItemClicked()'
/usr/local/lib/libsrsgui.so: undefined reference to `typeinfo for QwtLinearColorMap'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::pickRect() const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::updateDisplay()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtRasterData::initRaster(QRectF const&, QSize const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotCurve::QwtPlotCurve(QString const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::widgetLeaveEvent(QEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::setZoomBase(bool)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::resizeEvent(QResizeEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::widgetMouseMoveEvent(QMouseEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::stretchSelection(QSize const&, QSize const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::qt_metacast(char const*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::adjustedPoints(QPolygon const&) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::~QwtPlot()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::drawRubberBand(QPainter*) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtLinearColorMap::QwtLinearColorMap(QColor const&, QColor const&, QwtColorMap::Format)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::QwtPlotZoomer(QwtPlotCanvas*, bool)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::updateLayout()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::axisWidget(int)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::setAxis(int, int)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotSpectrogram::setData(QwtRasterData*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtEventPattern::mouseMatch(QwtEventPattern::MousePattern const&, QMouseEvent const*) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::~QwtPlotZoomer()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::canvas()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotPicker::trackerText(QPoint const&) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::widgetMouseDoubleClickEvent(QMouseEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtColorMap::colorTable(QwtInterval const&) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::zoom(QRectF const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `typeinfo for QwtRasterData'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtText::~QwtText()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotPanner::QwtPlotPanner(QwtPlotCanvas*)'
[... 64 lines cut ...]
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlot::setAxisScaleDiv(int, QwtScaleDiv const&)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtRasterData::QwtRasterData()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::widgetMouseReleaseEvent(QMouseEvent*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPicker::reset()'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotPanner::setAxisEnabled(int, bool)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::accept(QPolygon&) const'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::qt_metacast(char const*)'
/usr/local/lib/libsrsgui.so: undefined reference to `QwtPlotZoomer::widgetKeyPressEvent(QKeyEvent*)'
collect2: error: ld returned 1 exit status
lib/src/phy/sync/test/CMakeFiles/pss_usrp.dir/build.make:101: recipe for target 'lib/src/phy/sync/test/pss_usrp' failed
make[2]: *** [lib/src/phy/sync/test/pss_usrp] Error 1
CMakeFiles/Makefile2:3177: recipe for target 'lib/src/phy/sync/test/CMakeFiles/pss_usrp.dir/all' failed
make[1]: *** [lib/src/phy/sync/test/CMakeFiles/pss_usrp.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
mkdir build 
cmake ../
make -j 2

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
I tried with a fresh install of Ubuntu 16.04 and the error persist.

andrepuschmann, 2019-02-04:
That's an issue with srsgui and the Qt libs you've installed (or not installed).

suttonpd, 2019-02-04:
Hi Gines,
srsLTE has found srsGUI and is trying to link against it. srsGUI requires libqwt but it is either not installed or can't be found. 
The error message "warning: libqwt.so.6, needed by /usr/local/lib/libsrsgui.so, not found" tells you this. Either remove srsGUI and do a full, clean build of srsLTE, or install libqwt.

suttonpd, 2019-02-04:
Remove all srsGUI header files and libraries from your install folders. Closing this thread as it's not an issue - please use the mailing list for further questions: http://www.softwareradiosystems.com/mailman/listinfo/srslte-users

closed 2019-02-04 by suttonpd

GinesGarcia, 2019-02-04:
ok Thanks