srsran/srsRAN_4G #316: UE cannot attach anymore

trocotronic, 2019-03-05:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
The last commit that worked smoothly is [e88e1bb4d2ecca27cb49ed3cfbbef7c2a7bf0d18](https://github.com/srsLTE/srsLTE/commit/e88e1bb4d2ecca27cb49ed3cfbbef7c2a7bf0d18). With this commit, my UE (a commercial P20 Lite) can attach to eNB and EPC and it can navigate in internet normally.
But when apply the next commit [4867b16e3877c0e0fcc9c1ef5e48664db90e69b6](https://github.com/srsLTE/srsLTE/commit/4867b16e3877c0e0fcc9c1ef5e48664db90e69b6) it seems that something is broken. My UE does not attach to eNB anymore. No PRACH or other signal is processed. 

## Setup Details ##
Commit `4867b16e3877c0e0fcc9c1ef5e48664db90e69b6` and following.
Debian 9.7.
UE is Huawei P20 Lite.
USRP B210, USB3.

## Expected Behavior ##
UE should attach to eNB as with previous commit `e88e1bb4d2ecca27cb49ed3cfbbef7c2a7bf0d18`

## Actual Behaviour ##
UE does not attach or send any signal to eNB. 

## Steps to reproduce the problem ##
Commit `4867b16e3877c0e0fcc9c1ef5e48664db90e69b6`

Used .conf files
[srs_conf.zip](https://github.com/srsLTE/srsLTE/files/2930173/srs_conf.zip)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2019-03-05:
Please try the current master. The issue should be fixed there.

closed 2019-03-05 by andrepuschmann

trocotronic, 2019-03-05:
Sorry if I did not explained carefully the steps. 
I started with master copy today. As I saw that my UE could not attach, I rolled back commit by commit until the last it worked, `e88e1bb4d2ecca27cb49ed3cfbbef7c2a7bf0d18`. Thus, current master still does not work.

andrepuschmann, 2019-03-05:
Can you run the eNB with RRC debug logs and provide the log and PCAP?

trocotronic, 2019-03-05:
Here I attach the rrc.log and enb.pcap. RRC is set to debug, all remaining to warning.
[srs_rrc.zip](https://github.com/srsLTE/srsLTE/files/2930291/srs_rrc.zip)

andrepuschmann, 2019-03-05:
That looks okish, can you do the same for the last commit that worked?

trocotronic, 2019-03-05:
Here I attach the last working commit (e88e1bb4d2). The UE attaches successfully and some seconds later it enters to airplane mode (detach).
[srs_rrc_e88e1bb4d2.zip](https://github.com/srsLTE/srsLTE/files/2931101/srs_rrc_e88e1bb4d2.zip)

andrepuschmann, 2019-03-06:
Thanks for the logs but I am afraid I have to ask you to run with log.phy_level=info again. I'd like to look at the SIBs and they are not in the PCAP or RRC logs in pre-18.12 releases. If you have a chance, you could also try to attach srsUE to that eNB and send the logs and PCAP from the UE

trocotronic, 2019-03-07:
Here I attach phy in info, the rest (rrc included) in warning. This is for the last working commit.
I could not run srsUE because I only have 1 USRP, dedicated to eNB. PCAP included is for the P20.

[srs_phy_e88e1bb4d2.zip](https://github.com/srsLTE/srsLTE/files/2940261/srs_phy_e88e1bb4d2.zip)

andrepuschmann, 2019-03-07:
Sorry, but can you please run again and uncomment  "#define WRITE_SIB_PCAP" in srsenb/src/mac/mac.cc? Otherwise the SIBs are not in the PCAP. Thanks a lot.

trocotronic, 2019-03-07:
Sure. Here you have the PCAP with SIB traces.

[srs_phy_sib.zip](https://github.com/srsLTE/srsLTE/files/2941529/srs_phy_sib.zip)

andrepuschmann, 2019-03-07:
The SIB1 is corrupted. You are using the old config file for SIB1. Please update to the current sib.conf.example. Also, you should see a warning on the console when running srsENB.

trocotronic, 2019-03-07:
Perfect! This was the issue, an old sib.conf file. As you mentioned, I got some RRC warnings:

>The condition lb <= n <= ub (-70 <= 126 <= -22) was not met
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21058] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21375] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21255] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21405] Encoding failure.
The condition lb <= n <= ub (-1 <= 8 <= 6) was not met
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][12652] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][13375] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][16584] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][20512] Encoding failure.
[/home/me/srsLTE/lib/include/srslte/asn1/asn1_utils.h][714] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][20703] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][20893] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][20786] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21372] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21255] Encoding failure.
[/home/me/srsLTE/lib/src/asn1/rrc_asn1.cc][21405] Encoding failure.

but I never realized that were caused by an old sib.conf. I thought they were related with the new ASN1 encoder introduced few weeks ago.
Anyway, finally it works as always did. 
Thank you for your time and patience.

andrepuschmann, 2019-03-07:
That's fine. Glad you got it working in the end.

dominicantonyv, 2019-03-22:
Hi

I successfully installed srsLTE. I ran srsEPC and srsENB, but my smartphone failed to detect the network. 
My OS is Ubuntu 14.04 LTS and the hardware is NI USRP-2901.

May I ask where are all the configuration files saved? how can I check the bandwidth and other parameters.

Also when I ran the command "sudo srslte_install_configs.sh", I got the response like 

Please call script with either user or service as first parameter.

E.g. ./srslte_install_configs.sh user
 .. to install all config files to /home/maxeyepc13/.config/srslte

E.g. ./srslte_install_configs.sh service
 .. to install all config files to /etc/srslte

So then I ran "sudo srslte_install_configs.sh service" and "sudo srslte_install_configs.sh user" for which I got the following responses respectively

$ sudo srslte_install_configs.sh service

Installing srsLTE configuration files:
 - Creating srsLTE config folder /etc/srslte
 - Installing ue.conf.example in /etc/srslte/ue.conf
 - Installing enb.conf.example in /etc/srslte/enb.conf
 - Installing sib.conf.example in /etc/srslte/sib.conf
 - Installing rr.conf.example in /etc/srslte/rr.conf
 - Installing drb.conf.example in /etc/srslte/drb.conf
 - Installing epc.conf.example in /etc/srslte/epc.conf
 - Installing mbms.conf.example in /etc/srslte/mbms.conf
 - Installing user_db.csv.example in /etc/srslte/user_db.csv
Done.

$ sudo srslte_install_configs.sh user
Installing srsLTE configuration files:
 - Creating srsLTE config folder /home/maxeyepc13/.config/srslte
 - Installing ue.conf.example in /home/maxeyepc13/.config/srslte/ue.conf
 - Installing enb.conf.example in /home/maxeyepc13/.config/srslte/enb.conf
 - Installing sib.conf.example in /home/maxeyepc13/.config/srslte/sib.conf
 - Installing rr.conf.example in /home/maxeyepc13/.config/srslte/rr.conf
 - Installing drb.conf.example in /home/maxeyepc13/.config/srslte/drb.conf
 - Installing epc.conf.example in /home/maxeyepc13/.config/srslte/epc.conf
 - Installing mbms.conf.example in /home/maxeyepc13/.config/srslte/mbms.conf
 - Installing user_db.csv.example in /home/maxeyepc13/.config/srslte/user_db.csv
Done.

Are those correct or did I do anything wrong? 
Could someone please guide me to find the configuration files and guide me on how to change the configurations?

Thanks for your consideration.