srsran/srsRAN_4G #319: Build failure on CentOS 7 due to zero initializer

ttsou, 2019-03-16:
## Issue Description ##

Build failure on CentOS 7 due to zero initializer.

It's not clear to me whether the base requirement comes from C99, C++11, C++14, or strictly the gcc version, but the C form `(struct timeval){ 0 }` compiles (through triggers new warnings).

```
diff --git a/lib/include/srslte/common/metrics_hub.h b/lib/include/srslte/common/metrics_hub.h
index eda6d09..81ab4d4 100644
--- a/lib/include/srslte/common/metrics_hub.h
+++ b/lib/include/srslte/common/metrics_hub.h
@@ -37,7 +37,7 @@ public:
     :m(NULL)
   {
     for (int i = 0; i < 3; ++i) {
-      sleep_period_start[i] = (struct timeval){};
+      sleep_period_start[i] = (struct timeval){0};
     }
   }
   bool init(metrics_interface<metrics_t> *m_, float report_period_secs_=1.0) {
```

## Setup Details ##
```
CentOS Linux release 7.6.1810 (Core)
Linux version 3.10.0-862.11.6.rt56.819.el7.x86_64 (mockbuild@koji35.cern.ch) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-28) (GCC) ) #1 SMP PREEMPT RT Wed Aug 15 09:40:31 CEST 2018

```

```
commit 44b20d4abab763e6627156b59af7c8b9e5675dac
Author: Andre Puschmann <andre@softwareradiosystems.com>
Date:   Wed Nov 14 15:07:21 2018 +0100

    fix compile-time error with glibc for CentOS for ipv6 headers

commit 3cc4ca851a18b15234d849a5a4a8f9bf0768d30f
Author: Andre Puschmann <andre@softwareradiosystems.com>
Date:   Wed Feb 20 15:25:31 2019 +0100

    move setting of EARFCNs straight into parse_args function
```

## Expected Behavior ##
Finish build.

## Actual Behaviour ##
```
/srsLTE/lib/include/srslte/common/metrics_hub.h: In constructor ‘srslte::metrics_hub<metrics_t>::metrics_hub()’:
/srsLTE/lib/include/srslte/common/metrics_hub.h:40:32: error: expected primary-expression before ‘struct’
       sleep_period_start[i] = (struct timeval){};
                                ^
/home/ni/lte-standalone/srsLTE/lib/include/srslte/common/metrics_hub.h:40:32: error: expected ‘)’ before ‘struct’
```

andrepuschmann, 2019-03-18:
Thanks for reporting this Tom. We gonna change the build flags in the next release and expect this to be working down to gcc 4.4 or so.

andrepuschmann, 2019-04-24:
Closing this issue. Please reopen if it still exists in the new release.

closed 2019-04-24 by andrepuschmann