srsran/srsRAN_4G #322: LTE network is not detected in smartphone

dominicantonyv, 2019-03-26:
Hi
I successfully installed srsLTE. I ran srsEPC and srsENB, but my smartphone failed to detect the network.
My OS is Ubuntu 14.04 LTS and the hardware is NI USRP-2901 and the smartphone used as UE is Mi A1.
May I ask where are all the configuration files saved? how can I check the bandwidth and other parameters of current generation.

Also when I ran the command "_sudo srslte_install_configs.sh_", I got the response like

> Please call script with either user or service as first parameter.
> E.g. ./srslte_install_configs.sh user
> .. to install all config files to /home/maxeyepc13/.config/srslte
> E.g. ./srslte_install_configs.sh service
> .. to install all config files to /etc/srslte

So then I ran "_sudo srslte_install_configs.sh service_" and "_sudo srslte_install_configs.sh user_" for which I got the following responses respectively

> $ sudo srslte_install_configs.sh service
> Installing srsLTE configuration files:
> Creating srsLTE config folder /etc/srslte
> Installing ue.conf.example in /etc/srslte/ue.conf
> Installing enb.conf.example in /etc/srslte/enb.conf
> Installing sib.conf.example in /etc/srslte/sib.conf
> Installing rr.conf.example in /etc/srslte/rr.conf
> Installing drb.conf.example in /etc/srslte/drb.conf
> Installing epc.conf.example in /etc/srslte/epc.conf
> Installing mbms.conf.example in /etc/srslte/mbms.conf
> Installing user_db.csv.example in /etc/srslte/user_db.csv
> Done.

> $ sudo srslte_install_configs.sh user
> Installing srsLTE configuration files:
> Creating srsLTE config folder /home/maxeyepc13/.config/srslte
> Installing ue.conf.example in /home/maxeyepc13/.config/srslte/ue.conf
> Installing enb.conf.example in /home/maxeyepc13/.config/srslte/enb.conf
> Installing sib.conf.example in /home/maxeyepc13/.config/srslte/sib.conf
> Installing rr.conf.example in /home/maxeyepc13/.config/srslte/rr.conf
> Installing drb.conf.example in /home/maxeyepc13/.config/srslte/drb.conf
> Installing epc.conf.example in /home/maxeyepc13/.config/srslte/epc.conf
> Installing mbms.conf.example in /home/maxeyepc13/.config/srslte/mbms.conf
> Installing user_db.csv.example in /home/maxeyepc13/.config/srslte/user_db.csv
> Done.

Are those correct or did I do anything wrong?
Could someone please guide me to find the configuration files and guide me on how to change the configurations?
Thanks for your consideration.

andrepuschmann, 2019-03-26:
Looks good. Your question above includes your answer already. Please follow the git issue submission guidelines and share logs.

closed 2019-03-26 by andrepuschmann

Usman-Josh, 2022-06-06:
I have the same issue, my mobile phone is not picking the network, I have changed the MCC and MNC in the EPC and ENB config files, and also the APN name, could the issue be from not knowing the IMSI  and the values of other parameters that are in the user_db.csv file?