srsran/srsRAN_4G #323: srsue can't connect to srsenb

MIKEXU2017, 2019-04-01:
## Issue Description ##
srsue can't connect to srsenb

## Setup Details ##

**device：**
Ubuntu16.04
srsepc&srsenb:bladeRFx40
srsue:bladeRFx40

**PS**
Both computers use a virtual machine and use the same Ubuntu.vdi
----------------------------------------------------------

**srsepc**
zpt4x@zpt4x-VirtualBox:~/srsLTE/srsepc$ sudo srsepc epc.conf
[sudo] password for zpt4x:

Built in Release mode using commit 3cc4ca8 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response

----------------------------------------------------------

**srsenb**
zpt4x@zpt4x-VirtualBox:~/srsLTE/srsenb$ sudo srsenb enb.conf
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d

Built in Release mode using commit 3cc4ca8 on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...
Opening USRP with args:
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
set TX frequency to 2684999936
set RX frequency to 2564999936
Starting plot for worker_id=0
Setting Sampling frequency 11.52 MHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz

==== eNodeB started ===
Type <t> to view trace

----------------------------------------------------------
**srsue**
zpt4x@zpt4x-VirtualBox:~/srsLTE/srsue$ sudo srsue ue.conf
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d
Reading configuration file ue.conf...

Built in Release mode using commit 3cc4ca8 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
Opening USRP with args:
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.....
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=0.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.5 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.5 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.4 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.4 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.4 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.4 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.5 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.8 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.4 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.9 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.5 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
Stopping srsUE... Press Ctrl+C 9 more times to force stop
^C.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.9 KHz
---  exiting  ---

----------------------------------------------------------

If I exchange two bladeRFx40 to use,the following results will appear.

> (The value of CFO is the opposite.)

**srsue**
zpt4x@zpt4x-VirtualBox:~/srsLTE/srsue$ sudo srsue ue.conf
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d
Reading configuration file ue.conf...

Built in Release mode using commit 3cc4ca8 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
Opening USRP with args:
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.4 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.7 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.3 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.1 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.2 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.2 KHz
Stopping srsUE... Press Ctrl+C 9 more times to force stop
^C.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.4 KHz
---  exiting  ---

## Expected Behavior ##
I want srsue to connect to srsenb.

## Actual Behaviour ##
Nothing happened.

[image]

andrepuschmann, 2019-04-01:
Hey, please send the UE logs along. In general we suggest you should make sure that this setup works with pdsch_enodeb and pdsch_ue first.

MIKEXU2017, 2019-04-02:
_The problem seems to have become more serious?_
zpt4x@zpt4x-VirtualBox:~/srsLTE/srsue$ sudo srsue ue.conf
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d
Reading configuration file ue.conf...

Built in Release mode using commit 3cc4ca8 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.5 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.8 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.6 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.3 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.7 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=1.8 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.........^CStopping srsUE... Press Ctrl+C 9 more times to force stop
.---  exiting  ---
[ue.log](https://github.com/srsLTE/srsLTE/files/3032499/ue.log)

MIKEXU2017, 2019-04-02:
_This is another run log with the opposite value：_
zpt4x@zpt4x-VirtualBox:~/srsLTE/srsue$ sudo srsue ue.conf
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d
Reading configuration file ue.conf...

Built in Release mode using commit 3cc4ca8 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.7 KHz
...
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.6 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.9 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.8 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.6 KHz
.
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.5 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.6 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.4 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.5 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Stopping srsUE... Press Ctrl+C 9 more times to force stop
^CSet RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.---  exiting  ---
[ue.log](https://github.com/srsLTE/srsLTE/files/3032511/ue.log)

MIKEXU2017, 2019-04-04:
### Now testing pdsch_enodeb and pdsch_ue:
**pdsch_enodeb:**
zpt4x@zpt4x-VirtualBox:~/srsLTE/build/lib/examples$ ./pdsch_enodeb -f 1800
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d
Opening RF device...
Opening USRP with args:
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Setting sampling rate 5.76 MHz
Set TX gain: 60.0 dB
[INFO @ fpga_common/src/lms.c:2227] Clamping frequency to 237500000Hz
set TX frequency to 237500000
Set TX freq: 0.00 MHz
- Resource Allocation Type:        Type 0
+ Resource Block Group Size:        2
+ RBG Bitmap:            0x1fff
- HARQ process:            0
- TPC command for PUCCH:        --
- Transport blocks swapped:        false
- Transport block 1 enabled:        true
+ Modulation and coding scheme index:    1
+ New data indicator:            No
+ Redundancy version:            0
- Transport block 2 enabled:        false
- PRB Bitmap Assignment 0st slot:
0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24,
- PRB Bitmap Assignment 1st slot:
0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24,
- Number of PRBs:            25
- Number of TBs:            1
- Transport block:            0
-> Modulation type:            QPSK
-> Transport block size:        904
Type new MCS index (0-28) and press Enter: ^CSIGINT received. Exiting...
Done
----------------------------------------------------------------------------------------
**pdsch_ue:**
zpt4x@zpt4x-VirtualBox:~/srsLTE/build/lib/examples$ ./pdsch_ue -r 1234 -f 1800
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-128-g379f922d
Opening RF device with 1 RX antennas...
Opening USRP with args:
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Starting AGC thread...
Tunning receiver to 0.002 MHz
[INFO @ fpga_common/src/lms.c:2227] Clamping frequency to 237500000Hz
set RX frequency to 237500000
Searching for cell...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
*Found Cell_id:   0 CP: Normal  , DetectRatio=100% PSR=15.34, Power=63.1 dBm
Found Cell_id:   0 CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Found Cell_id:   0 CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 0 (N_id_2=0)
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Setting sampling rate 5.76 MHz
Set RX sampling rate 5.76 Mhz, filter BW: 5.00 Mhz
- PCI:             0     3.7, FrameCnt: 0, State: 1 0
- Nof ports:       1
- CP:              Normal 
- PRB:             25
- PHICH Length:    Normal
- PHICH Resources: 1
- SFN:             784
Decoded MIB. SFN: 784, offset: 0
  Tx scheme: Single
nof layers: 1
nof codewords: 1
  CFO: +141.88 Hz
  RSRP: +38.8 dBm |  -inf dBm
  SNR: +22.3 dB |  -inf dB
  Rb:   0.64 /   0.70 /   2.20 Mbps (net/maximum/processing)
PDCCH-Miss: 26.81%
PDSCH-BLER:  0.00%
  TB 0: mcs=1; tbs=904
  TB 1: mcs=0; tbs=0

Press enter maximum printing debug log of 1 subframe.

  Tx scheme: Single
nof layers: 1
nof codewords: 1
  CFO:  -16.04 Hz
  RSRP: +39.4 dBm |  -inf dBm
  SNR: +21.0 dB |  -inf dB
  Rb:   0.70 /   0.70 /   2.23 Mbps (net/maximum/processing)
PDCCH-Miss: 26.05%
PDSCH-BLER:  0.00%
  TB 0: mcs=1; tbs=904
  TB 1: mcs=0; tbs=0

Press enter maximum printing debug log of 1 subframe.

  Tx scheme: Single
nof layers: 1
nof codewords: 1
  CFO: -183.75 Hz
  RSRP: +40.5 dBm |  -inf dBm
  SNR: +11.9 dB |  -inf dB
  Rb:   0.72 /   0.70 /   1.85 Mbps (net/maximum/processing)
PDCCH-Miss: 27.31%
PDSCH-BLER:  0.01%
  TB 0: mcs=1; tbs=904
  TB 1: mcs=0; tbs=0

Press enter maximum printing debug log of 1 subframe.

  Tx scheme: Single
nof layers: 1
nof codewords: 1
  CFO: -195.56 Hz
  RSRP: +40.8 dBm |  -inf dBm
  SNR: +11.3 dB    Tx scheme: Single
SIGINT received. Exiting....70 /   2.20 Mbps (net/maximum/processing)
nof layers: 1
nof codewords: 1
  CFO: -1922.21 Hz
  RSRP: +38.9 dBm |  -inf dBm
  SNR:  +0.9 dB |  -inf dB
  Rb:   0.40 /   0.00 /   0.01 Mbps (net/maximum/processing)
PDCCH-Miss: 30.73%
PDSCH-BLER:  0.00%
  TB 0: mcs=1; tbs=904
  TB 1: mcs=0; tbs=0

Press enter maximum printing debug log of 1 subframe.

Hangup
[image]
[image]

MIKEXU2017, 2019-04-05:
Hi,

Can anyone help me with this?This has confused me for a long time, or someone told me what is wrong with the tests of pdsch_enodeb and pdsch_ue?

Best regards,
Mike

andrepuschmann, 2019-04-24:
Closing this here because the discussion seems to go nowhere. Please use the list for config issues like this.

closed 2019-04-24 by andrepuschmann