srsran/srsRAN_4G #326: Error while installing srsLTE

Zethernos, 2019-04-09:
Before filing an issue, search for solutions here:
- srsLTE users mailing list ()

## Issue Description ##
Hi everyone, I am very new to this project. I would like to ask if there is anything that I am missing out.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Currently using B210, two laptops. Ubutnu 18.04.1 LTS, 64 bit(currently resolving this issue) using this website for instructions. https://github.com/srsLTE/srsLTE

## Expected Behavior ##
I expect to download and build srsLTE smoothly. However when doing the next step0 "make" command it seems to be hang/stuck.  

## Actual Behaviour ##
pi@pi-ubuntu:~/srsLTE/build$ cmake ../
-- CMAKE_SYSTEM: Linux-4.15.0-42-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Could NOT find mkl (missing: MKL_LIBRARIES MKL_CORE MKL_ILP MKL_SEQ) 
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Checking for module 'libpcsclite'
--   No package 'libpcsclite' found
-- Could NOT find PCSCLITE (missing: PCSCLITE_LIBRARY PCSCLITE_INCLUDE_DIR) 
-- PCSC LIBRARIES: PCSCLITE_LIBRARY-NOTFOUND
-- PCSC INCLUDE DIRS: PCSCLITE_INCLUDE_DIR-NOTFOUND
-- UHD LIBRARIES /usr/local/lib/libuhd.so
-- UHD INCLUDE DIRS /usr/local/include
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- FINDING SOAPY.
-- Checking for module 'SoapySDR'
--   No package 'SoapySDR' found
-- libSOAPYSDR not found.
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
-- SRSGUI LIBRARIES SRSGUI_LIBRARIES-NOTFOUND
-- SRSGUI INCLUDE DIRS SRSGUI_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SRSGUI (missing: SRSGUI_LIBRARIES SRSGUI_INCLUDE_DIRS) 
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- CMAKE_C_FLAGS is  -Wall -Wno-comment -Wno-write-strings -Winline -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -mfma -DLV_HAVE_FMA -Ofast -funroll-loops
-- CMAKE_CXX_FLAGS is  -Wall -Wno-comment -Wno-reorder -Wno-unused-but-set-variable -Wno-unused-variable -Wformat -Wmissing-field-initializers -Wtype-limits -std=c++03 -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE
-- Using install prefix: /usr/local
-- Building for version: 18.12.0
-- SRSGUI LIBRARIES SRSGUI_LIBRARIES-NOTFOUND
-- SRSGUI INCLUDE DIRS SRSGUI_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SRSGUI (missing: SRSGUI_LIBRARIES SRSGUI_INCLUDE_DIRS) 
-- SRSGUI LIBRARIES SRSGUI_LIBRARIES-NOTFOUND
-- SRSGUI INCLUDE DIRS SRSGUI_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SRSGUI (missing: SRSGUI_LIBRARIES SRSGUI_INCLUDE_DIRS) 
--    examples will be installed.
-- No post-build command defined
-- Building with srsUE
-- No post-build-UE command defined
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/srsLTE/build

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
References -
sudo apt-get install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev
http://www.sharetechnote.com/html/SDR_srsLTE_Build.html (references for uhd installation, srsGUI installation), https://files.ettus.com/manual/page_build_guide.html, http://files.ettus.com/manual/page_install.html. These websites I have followed on the Linux section.
## Additional Information ##
As I am new, I have read that we need two devices and one USRP. Please do let me know if there is anything I am missing thanks and cheers.

Zethernos, 2019-04-09:
pi@pi-ubuntu:~$ sudo apt-get update
Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease       
Ign:3 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Err:4 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Get:5 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]    
Hit:6 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Hit:7 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease         
Get:8 http://id.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]   
Get:9 http://id.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB] 
Reading package lists... Done                                  
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
I am not sure of this bionic error when using this website : http://files.ettus.com/manual/page_install.html

Zethernos, 2019-04-09:
pi@pi-ubuntu:~/srsLTE/build$ sudo srsue
Reading configuration file /home/pi/.config/srslte/ue.conf...

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto
I an currently stuck at this problem with the same bionic error above.

andrepuschmann, 2019-04-09:
Looks like you forgot to connect the USRP. What does uhd_usrp_probe say?

Zethernos, 2019-04-09:
uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/local/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   revision: 4
|   |   product: 2
|   |   serial: 3136D68
|   |   name: MyB210
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX2
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: B
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B210 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX2
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

andrepuschmann, 2019-04-09:
No idea why the UHD is not picking it up then. Perhaps use the older LTS release 3.9.7.

closed 2019-04-09 by andrepuschmann