srsran/srsRAN_4G #327: Using

Zethernos, 2019-04-10:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
Having some difficulties regarding the srsUE when using this laptop, Currently I have the another laptop that's able to run the function sudo srsepc and sudo srsenb. Besides that, regarding the SIM card for the UE, it is only for typical USIM cards that are programmable?

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Currently using B210, two laptops. Ubutnu 18.04.1 LTS, 64 bit(currently resolving this issue) using this website for instructions.

## Expected Behavior ##
I tried to build the srsUE. However, it displays error. More information is at the additional information below.

## Actual Behaviour ##
pi@pi-ubuntu:~/srsLTE$ sudo add-apt-repository ppa:ettusresearch/uhd-3.9.lts
 This is the long-term support (LTS) 3.9 version of UHD.
 More info: https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd-3.9.lts
Press [ENTER] to continue or Ctrl-c to cancel adding it.

Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 https://download.docker.com/linux/ubuntu bionic InRelease                
Ign:3 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic InRelease
Ign:4 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Hit:5 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Hit:6 http://security.ubuntu.com/ubuntu bionic-security InRelease              
Err:7 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Hit:8 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease       
Hit:9 http://id.archive.ubuntu.com/ubuntu bionic-updates InRelease             
Hit:10 http://id.archive.ubuntu.com/ubuntu bionic-backports InRelease          
Hit:11 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease  
Err:12 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done                      
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
pi@pi-ubuntu:~/srsLTE$ sudo apt-get update
Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 https://download.docker.com/linux/ubuntu bionic InRelease                
Ign:3 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic InRelease
Ign:4 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Hit:5 http://security.ubuntu.com/ubuntu bionic-security InRelease              
Hit:6 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Err:7 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Hit:8 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease 
Hit:9 http://id.archive.ubuntu.com/ubuntu bionic-updates InRelease       
Hit:10 http://id.archive.ubuntu.com/ubuntu bionic-backports InRelease          
Hit:11 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease        
Err:12 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done                       
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
https://github.com/srsLTE/srsLTE/issues/326

> No idea why the UHD is not picking it up then. Perhaps use the older LTS release 3.9.7. I tried using this method. However I have two errors instead of one. Is this something I should ignore?

Zethernos, 2019-04-10:
Thanks in advance

Zethernos, 2019-04-10:
pi@pi-ubuntu:~$ uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/local/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   revision: 4
|   |   product: 2
|   |   serial: 3136D68
|   |   name: MyB210
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX2
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: B
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B210 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX2
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

Zethernos, 2019-04-10:
> pi@pi-ubuntu:~$ uhd_usrp_probe
> [INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
> [INFO] [B200] Loading firmware image: [...]

pi@pi-ubuntu:~$ uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/local/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO....
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.

/
| Device: B-Series Device
| _____________________________________________________
| /
| | Mboard: B210
| | revision: 4
| | product: 2
| | serial: 3136D68
| | name: MyB210
| | FW Version: 8.0
| | FPGA Version: 16.0
| |
| | Time sources: none, internal, external, gpsdo
| | Clock sources: internal, external, gpsdo
| | Sensors: ref_locked
| | _____________________________________________________
| | /
| | | RX DSP: 0
| | |
| | | Freq range: -8.000 to 8.000 MHz
| | _____________________________________________________
| | /
| | | RX DSP: 1
| | |
| | | Freq range: -8.000 to 8.000 MHz
| | _____________________________________________________
| | /
| | | RX Dboard: A
| | | _____________________________________________________
| | | /
| | | | RX Frontend: A
| | | | Name: FE-RX2
| | | | Antennas: TX/RX, RX2
| | | | Sensors: temp, rssi, lo_locked
| | | | Freq range: 50.000 to 6000.000 MHz
| | | | Gain range PGA: 0.0 to 76.0 step 1.0 dB
| | | | Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
| | | | Connection Type: IQ
| | | | Uses LO offset: No
| | | _____________________________________________________
| | | /
| | | | RX Frontend: B
| | | | Name: FE-RX1
| | | | Antennas: TX/RX, RX2
| | | | Sensors: temp, rssi, lo_locked
| | | | Freq range: 50.000 to 6000.000 MHz
| | | | Gain range PGA: 0.0 to 76.0 step 1.0 dB
| | | | Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
| | | | Connection Type: IQ
| | | | Uses LO offset: No
| | | _____________________________________________________
| | | /
| | | | RX Codec: A
| | | | Name: B210 RX dual ADC
| | | | Gain Elements: None
| | _____________________________________________________
| | /
| | | TX DSP: 0
| | |
| | | Freq range: -8.000 to 8.000 MHz
| | _____________________________________________________
| | /
| | | TX DSP: 1
| | |
| | | Freq range: -8.000 to 8.000 MHz
| | _____________________________________________________
| | /
| | | TX Dboard: A
| | | _____________________________________________________
| | | /
| | | | TX Frontend: A
| | | | Name: FE-TX2
| | | | Antennas: TX/RX
| | | | Sensors: temp, lo_locked
| | | | Freq range: 50.000 to 6000.000 MHz
| | | | Gain range PGA: 0.0 to 89.8 step 0.2 dB
| | | | Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
| | | | Connection Type: IQ
| | | | Uses LO offset: No
| | | _____________________________________________________
| | | /
| | | | TX Frontend: B
| | | | Name: FE-TX1
| | | | Antennas: TX/RX
| | | | Sensors: temp, lo_locked
| | | | Freq range: 50.000 to 6000.000 MHz
| | | | Gain range PGA: 0.0 to 89.8 step 0.2 dB
| | | | Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
| | | | Connection Type: IQ
| | | | Uses LO offset: No
| | | _____________________________________________________
| | | /
| | | | TX Codec: A
| | | | Name: B210 TX dual DAC
| | | | Gain Elements: None
____________________________________________
> | | /
> | | | RX DSP: 1
> | | |
> | | | Freq range: -8.000 to 8.000 MHz
> | | _____________________________________________________
> | | /
> | | | RX Dboard: A
> | | | [...]

May I know how do you tell that the UHD is not picking up based from uhd_usrp_probe from this :https://github.com/srsLTE/srsLTE/issues/326? Sorry for the inconvenience and really appreciate it if you can help to identify what issue did I miss out to rectify or forget to add in?

Zethernos, 2019-04-10:
Using my other laptop that's able to run sudo srsepc and sudo srsenb
pi@pi:~/srsLTE/build$ sudo srsepc
[sudo] password for pi: 

Built in Release mode using commit 1c015aa on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /home/pi/.srs/epc.conf...
HSS Initialized.
MME GTP-C Initialized
MME Initialized. 
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:525, MNC:01, PLMN: 2487568
S1 Setup Request - TAC 739, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/pi/.srs/enb.conf...
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.13.0.1-release
Opening USRP with args: type=b200,master_clock_rate=30.72e6
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 30.720000 MHz... 
[INFO] [B200] Actually got clock rate 30.720000 MHz.
Error opening RX stream: 30
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto

Sometimes sudo srsenb is able to smoothly but sometimes it keep looping and displaying error message UHD :30 iirc.

Zethernos, 2019-04-10:
I have used sudo apt-get update followed by sudo apt-get upgrade. 
pi@pi-ubuntu:~$ sudo apt-get install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.10.2-1ubuntu2).
libboost-program-options-dev is already the newest version (1.65.1.0ubuntu1).
libfftw3-dev is already the newest version (3.3.7-1).
libsctp-dev is already the newest version (1.0.17+dfsg-2).
libconfig++-dev is already the newest version (1.5-0.4).
libmbedtls-dev is already the newest version (2.8.0-1).
0 upgraded, 0 newly installed, 0 to remove and 21 not upgraded.
Initially, I had 500 not upgraded however after using those it cutted down to 21 not upgraded.

andrepuschmann, 2019-04-10:
You are having UHD issues. Note that 18.10 does work with the Ettus LTS PPA.

closed 2019-04-10 by andrepuschmann

Zethernos, 2019-04-10:
Can you help me how to approach it as I unable to find the solution? Thanks in advance.