srsran/srsRAN_4G #328: srsUE display error message

Zethernos, 2019-04-10:
## Issue Description ##
Having some difficulties regarding the srsUE when using this laptop, Currently I have the another laptop that's able to run the function sudo srsepc and sudo srsenb. 

## Setup Details ##
Currently using B210, two laptops. Ubutnu 18.04.1(currently using to build sudo srsue), ubuntu 16.04. LTS, 64 bit(currently resolving this issue) using this website for instructions.

## Expected Behavior ##
Running the sudo srsue command without any errors

## Actual Behaviour ##
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto

pi@pi-ubuntu:~/srsLTE/build$ uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   revision: 4
|   |   product: 2
|   |   serial: 3136D68
|   |   name: MyB210
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX2
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
[... 260 lines cut ...]
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

## Steps to reproduce the problem ##
By running sudo srsepc and sudo srsenb in the first laptop with B210 with two seperate terminals. 

## Additional Information ##
Can you please help me how to find the solution by adding some ubuntu commands? Forgive me but I have been trying to resolve this issue for several days. I went to use this commands
sudo add-apt-repository ppa:ettusresearch/uhd-3.9.lts
sudo apt-get update
by given advice here in this forum. I am relatively new to this. Thank you! I am not sure why I am unable to UHD version to 3.9.7

Zethernos, 2019-04-11:
Update: ---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto

pi@pi-ubuntu:~/srsLTE/build$ sudo apt-get update
[sudo] password for pi: 
Get:1 https://download.docker.com/linux/ubuntu bionic InRelease [64.4 kB]
Hit:2 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Ign:3 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic InRelease
Get:4 http://id.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]   
Get:5 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]    
Ign:6 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Hit:7 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease       
Hit:8 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Err:9 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Get:10 http://id.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Hit:11 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease        
Get:12 http://id.archive.ubuntu.com/ubuntu bionic-updates/main i386 Packages [487 kB]
Err:13 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.83 80]
Get:14 http://id.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [574 kB]
Get:15 http://security.ubuntu.com/ubuntu bionic-security/main i386 Packages [216 kB]
Get:16 http://id.archive.ubuntu.com/ubuntu bionic-updates/main amd64 DEP-11 Metadata [278 kB]
Get:17 http://id.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 48x48 Icons [66.7 kB]
Get:18 http://id.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 64x64 Icons [123 kB]
Get:19 http://id.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [755 kB]
Get:20 http://id.archive.ubuntu.com/ubuntu bionic-updates/universe i386 Packages [744 kB]
Get:21 http://id.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 DEP-11 Metadata [209 kB]
Get:22 http://id.archive.ubuntu.com/ubuntu bionic-updates/universe DEP-11 48x48 Icons [183 kB]
Get:23 http://id.archive.ubuntu.com/ubuntu bionic-updates/universe DEP-11 64x64 Icons [345 kB]
Get:24 http://id.archive.ubuntu.com/ubuntu bionic-updates/multiverse i386 Packages [6,544 B]
Get:25 http://id.archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [6,388 B]
Get:26 http://id.archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 DEP-11 Metadata [2,468 B]
Get:27 http://id.archive.ubuntu.com/ubuntu bionic-backports/universe amd64 DEP-11 Metadata [7,352 B]
Get:28 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [296 kB]
Get:29 http://security.ubuntu.com/ubuntu bionic-security/main amd64 DEP-11 Metadata [204 B]
Get:30 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [131 kB]
Get:31 http://security.ubuntu.com/ubuntu bionic-security/universe i386 Packages [127 kB]
Get:32 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 DEP-11 Metadata [20.8 kB]
Get:33 http://security.ubuntu.com/ubuntu bionic-security/universe DEP-11 48x48 Icons [12.2 kB]
Get:34 http://security.ubuntu.com/ubuntu bionic-security/universe DEP-11 64x64 Icons [45.2 kB]
Get:35 http://security.ubuntu.com/ubuntu bionic-security/multiverse i386 Packages [3,900 B]
Get:36 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [3,748 B]
Get:37 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2,464 B]
Reading package lists... Done                                             
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

pi@pi-ubuntu:~/srsLTE/build$ sudo add-apt-repository -r ppa:ettusresearch/uhd-3.9.lts
 This is the long-term support (LTS) 3.9 version of UHD.
 More info: https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd-3.9.lts
Press [ENTER] to continue or Ctrl-c to cancel removing it.

pi@pi-ubuntu:~/srsLTE/build$ sudo apt-get update
Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 https://download.docker.com/linux/ubuntu bionic InRelease                
Hit:3 http://security.ubuntu.com/ubuntu bionic-security InRelease              
Hit:4 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease       
Ign:5 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Hit:6 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Err:7 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Hit:8 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease         
[... cut ...]
Reading package lists... Done                                             
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
 I connected to this laptop using BladeRF as my "SDR" while the other laptop is using USRP B210 which can run sudo srsepc and sudo srsenb

Zethernos, 2019-04-11:
pi@pi-ubuntu:~/srsLTE/build$ sudo apt-get update
Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 https://download.docker.com/linux/ubuntu bionic InRelease                
Get:3 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]    
Hit:4 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease       
Ign:5 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Hit:6 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Err:7 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Hit:8 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease         
Get:9 http://id.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]   
Get:10 http://id.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Reading package lists... Done                                  
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

What should I do with this error?

andrepuschmann, 2019-04-11:
This is not an srsLTE issue. I've told you that there is no Bionic PPA for Ettus' LTS series.

closed 2019-04-11 by andrepuschmann

Zethernos, 2019-04-11:
> This is not an srsLTE issue. I've told you that there is no Bionic PPA for Ettus' LTS series.

Forgive me,  but can you help me what should I do? I followed your advice by downloading 3.9.7 by going to ettus website to install.  I am totally new to this, can you please provide the steps? Thank you very much:).

andrepuschmann, 2019-04-11:
We recommend UHD 3.9.7 for production systems but you need to use the previous Ubuntu LTS for that. If you want to stick with 18.04 use the Ubuntu UHD version or the normal PPA.

Zethernos, 2019-04-11:
pi@pi-ubuntu:~$ sudo apt-get install libuhd-dev libuhd003 uhd-host
[sudo] password for pi: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
uhd-host is already the newest version (3.14.0.0-0ubuntu1~bionic1).
libuhd003 is already the newest version (3.13.0.1-0ubuntu1~bionic1).
libuhd-dev is already the newest version (3.14.0.0-0ubuntu1~bionic1).
The following packages were automatically installed and are no longer required:
  aufs-tools cgroupfs-mount linux-headers-4.15.0-29
  linux-headers-4.15.0-29-generic linux-image-4.15.0-29-generic
  linux-modules-4.15.0-29-generic linux-modules-extra-4.15.0-29-generic pigz
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

I got this as the output.
pi@pi-ubuntu:~$ sudo apt-get install libuhd-dev libuhd003 uhd-host
[sudo] password for pi: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
uhd-host is already the newest version (3.14.0.0-0ubuntu1~bionic1).
libuhd003 is already the newest version (3.13.0.1-0ubuntu1~bionic1).
libuhd-dev is already the newest version (3.14.0.0-0ubuntu1~bionic1).
The following packages were automatically installed and are no longer required:
  aufs-tools cgroupfs-mount linux-headers-4.15.0-29
  linux-headers-4.15.0-29-generic linux-image-4.15.0-29-generic
  linux-modules-4.15.0-29-generic linux-modules-extra-4.15.0-29-generic pigz
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

Thank you very much for helping me regarding this. I appreciate it. Besides that, I need to find the USIM programmable into a phone right? May I ask what particular USIM do you recommend? So far I have seen the one in ebay which sells about $12.08 USD