srsran/srsRAN_4G #329: UE not able to Ping from EPC

Avi-Dekra, 2019-04-10:
Hi,

I installed EPC on a linux box running Ubuntu 17.0.4. I am able to form a S1 link between eNodeB and successfully attach a UE to it. UE got an IP address of 172.16.0.2 and EPC has a SGW interface with IP address 172.16.0.1. I did added the apn in to the UE. 

But i am not able to ping the UE from the EPC PC and vice versa. I used if_masq.sh to configure a internet interface with not luck. Can suggest how i can unblock this.

Avi-Dekra, 2019-04-10:
Adding the configuration file
[epc.conf.zip](https://github.com/srsLTE/srsLTE/files/3064521/epc.conf.zip)

andrepuschmann, 2019-04-24:
I assume you're talking about a COTS UE. If so, for most UE's the Linux firewall settings are such that you can't ping them, i.e. they block ICMP by default.

closed 2019-04-24 by andrepuschmann

Avi-Dekra, 2019-04-24:
I am using Google Pixel 3 as an UE also tried with several other UEs with same result. Finally the UE is not able to access internet as well even though laptop running EPC has internet access

andrepuschmann, 2019-04-24:
If you have a rooted phone, you can enable ICMP so you can ping the phone with iptables. I.e., open a root shell using adb and then do:
'iptables -I INPUT 2 -p icmp --icmp-type echo-request -j ACCEPT'