srsran/srsRAN_4G #331: Unsure of the error from srsUE

Zethernos, 2019-04-15:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
Currently using B210, two laptops. Ubutnu 18.04.1 LTS, 64 bit(currently resolving this issue) with BladeRFx40, using this website for instructions.

## Expected Behavior ##
Unsure the reason behind this

## Actual Behaviour ##
pi@pi-ubuntu:~$ sudo srsue
Reading configuration file /home/pi/.config/srslte/ue.conf...

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1695] AGC not supported by FPGA. Manual gain control will be used instead.
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1696] To enable AGC, download and install FPGA v0.7.0 or newer from https://nuand.com/fpga/, then see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
....
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=0.5 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Found PLMN:  Id=00101, TAC=7
Setting PDN protocol to IPv4
Random Access Transmission: seq=19, ra-rnti=0x2
Random Access Transmission: seq=32, ra-rnti=0x2
Random Access Transmission: seq=22, ra-rnti=0x2
Random Access Transmission: seq=42, ra-rnti=0x2
Random Access Transmission: seq=43, ra-rnti=0x2
Random Access Transmission: seq=38, ra-rnti=0x2
Random Access Transmission: seq=5, ra-rnti=0x2
Random Access Transmission: seq=3, ra-rnti=0x2
Random Access Transmission: seq=49, ra-rnti=0x2
Random Access Transmission: seq=36, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Setting PDN protocol to IPv4

## Steps to reproduce the problem ##
Running srsepc and srsenb on the other laptop with b210 from ettus

## Additional Information ##
bladeRF> version

  bladeRF-cli version:        1.7.1-git-1da130cb
  libbladeRF version:         2.2.0-git-1da130cb

  Firmware version:           1.9.0
  FPGA version:               0.1.2

bladeRF> info

  Board:                    Nuand bladeRF (bladerf1)
  Serial #:                 54548816f628f3ba6703dc275e26b9d2
  VCTCXO DAC calibration:   0x9121
  FPGA size:                115 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit (assumed)
  USB bus:                  2
  USB address:              8
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0
pi@pi-ubuntu:~$ sudo srsue
[sudo] password for pi: 
Reading configuration file /home/pi/.config/srslte/ue.conf...

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1695] AGC not supported by FPGA. Manual gain control will be used instead.
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1696] To enable AGC, download and install FPGA v0.7.0 or newer from https://nuand.com/fpga/, then see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.............................................................

I understand FPGA version is outdated. I have tried to install in the computer but to avail. Is there any command that I can add in, i have been looking online?

Zethernos, 2019-04-15:
Does the Automatic Gain Control needed for this project? Currently I am trying to build srsUE successfully, before connecting to mobile device.

Zethernos, 2019-04-15:
I have tried to reinstall bladeRF and run the sudo srsue.
Is there any particular command i should run?
Received this error:

SYNC:  Receiving from radio while in IDLE_RX
RX failed: Invalid operation or parameter
pi@pi-ubuntu:~$ bladeRF-cli -i
bladeRF> version

  bladeRF-cli version:        1.7.1-git-1da130cb
  libbladeRF version:         2.2.0-git-1da130cb

  Firmware version:           1.9.0
  FPGA version:               0.0.3

Zethernos, 2019-04-15:
UPDATED:  pi@pi-ubuntu:~$ cd srsLTE
pi@pi-ubuntu:~/srsLTE$ cd srsLTE
bash: cd: srsLTE: No such file or directory
pi@pi-ubuntu:~/srsLTE$ cd srsue
pi@pi-ubuntu:~/srsLTE/srsue$ ls
CMakeLists.txt  hdr  src  test  ue.conf.example
pi@pi-ubuntu:~/srsLTE/srsue$ sudo srsue ue.conf.example
Reading configuration file ue.conf.example...

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
..........................................................
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=0.2 KHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
Found PLMN:  Id=00101, TAC=7
Setting PDN protocol to IPv4
Random Access Transmission: seq=27, ra-rnti=0x2
Random Access Transmission: seq=32, ra-rnti=0x2
Random Access Transmission: seq=34, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=42, ra-rnti=0x2
Random Access Transmission: seq=31, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=17, ra-rnti=0x2
Random Access Transmission: seq=41, ra-rnti=0x2
Random Access Transmission: seq=26, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Setting PDN protocol to IPv4
Random Access Transmission: seq=43, ra-rnti=0x2
Random Access Transmission: seq=15, ra-rnti=0x2
Random Access Transmission: seq=17, ra-rnti=0x2
Random Access Transmission: seq=36, ra-rnti=0x2
Random Access Transmission: seq=4, ra-rnti=0x2
Random Access Transmission: seq=29, ra-rnti=0x2
Random Access Transmission: seq=3, ra-rnti=0x2
Random Access Transmission: seq=20, ra-rnti=0x2
Random Access Transmission: seq=24, ra-rnti=0x2
Random Access Transmission: seq=1, ra-rnti=0x2
Warning: Detected Radio-Link Failure
^CStopping srsUE... Press Ctrl+C 9 more times to force stop
---  exiting  ---

Zethernos, 2019-04-15:
The FPGA is  now at version 0.8.0.

Zethernos, 2019-04-16:
pi@pi-ubuntu:~/srsLTE/build/lib/examples$ sudo ./pdsch_ue -f 2450000000 -r 1234
Opening RF device with 1 RX antennas...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Starting AGC thread...
Tunning receiver to 2450.000 MHz
set RX frequency to 2449999999
Searching for cell...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Could not find any cell in this frequency
Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Could not find any cell in this frequency

This error is similar : http://www.softwareradiosystems.com/pipermail/srslte-users/2016-July/000404.html however this is no solution. Is there anyone able to help me resolve this problem?

frankist, 2019-04-16:
Hey Zethernos,

Why are you using UHD 3.15.0? Can you use the UHD 3.9.0 LTS version instead?
Also, you should be using the srs mailing list for this. This page is only to report issues in the code.

Zethernos, 2019-04-17:
> Hey Zethernos,
> 
> Why are you using UHD 3.15.0? Can you use the UHD 3.9.0 LTS version instead?
> Also, you should be using the srs mailing list for this. This page is only to report issues in the code.

May I know does the two computers need to be in the same version? I am trying to create a "fake" base station while I am aiming to build the srsue first. Currently, one of the computers it is running ubuntu 16.04 and uhd 3.14 meant for building srsepc and srsenb.

frankist, 2019-04-17:
Both the srsenb and srsue require UHD 3.9.0 LTS

Zethernos, 2019-04-17:
> Both the srsenb and srsue require UHD 3.9.0 LTS
> […](#)

Alright thanks. I will just downloaded it 
I received this for the computer using srsue
i@pi-ubuntu:~/srsLTE/build$ sudo apt-get update
Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 https://download.docker.com/linux/ubuntu bionic InRelease                
Hit:3 http://id.archive.ubuntu.com/ubuntu bionic InRelease                     
Ign:4 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic InRelease
Get:5 http://id.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]   
Err:6 http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release
  404  Not Found [IP: 23.253.215.39 80]
Get:7 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]    
Ign:8 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic InRelease
Get:9 http://id.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB] 
Hit:10 http://ppa.launchpad.net/ettusresearch/uhd/ubuntu bionic InRelease      
Hit:11 http://ppa.launchpad.net/srslte/releases/ubuntu bionic InRelease        
Err:12 http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done
E: The repository 'http://files.ettus.com/binaries/uhd/repo/uhd/ubuntu/bionic bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/ettusresearch/uhd-3.9.lts/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
 Should I ignore the error for uhd 3.9 lts?

frankist, 2019-04-17:
I don't think that looks right. Just install from source.

git clone https://github.com/EttusResearch/uhd.git
cd uhd
git checkout UHD-3.9.LTS
cd host
mkdir build
cmake ..
make
sudo make install
sudo ldconfig

Now when you run "uhd_usrp_probe" you should see 3.9LTS

suttonpd, 2019-04-17:
This is not an issue, please move discussion to mailing list http://www.softwareradiosystems.com/mailman/listinfo/srslte-users

closed 2019-04-17 by suttonpd

Zethernos, 2019-04-18:
> I don't think that looks right. Just install from source. git clone https://github.com/EttusResearch/uhd.git cd uhd git checkout UHD-3.9.LTS cd host mkdir build cmake .. make sudo make install sudo ldconfig Now when you run "uhd_usrp_probe" you should see 3.9LTS
> […](#)

I followed your steps then I achieved this as the result( when I used bladeRF to connect)
uhd_usrp_probe 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.009.007-5-ge8202128

Error: LookupError: KeyError: No devices found for ----->
Empty Device Address
...................................
When I used b210 to connect
uhd_usrp_probe 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.009.007-5-ge8202128

-- Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
-- Detected Device: B210
-- Loading FPGA image: /usr/local/share/uhd/images/usrp_b210_fpga.bin... done
-- Operating over USB 3.
-- Detecting internal GPSDO.... No GPSDO found
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   revision: 4
|   |   product: 2
|   |   serial: 3136D68
|   |   name: MyB210
|   |   FW Version: 8.0
|   |   FPGA Version: 13.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX2
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: B
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B210 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 1
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX2
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

I think I got 3.9.7 however, when I build sudo srsue, it just shows UHD 3.15 the original version(ubuntu 18.04). I am sorry to trouble you but I am totally lost. I also build ue/epc/enb conf example respectively sometimes it doesn't well, only after restarting the computer or plugging the usb into other port or running the command.  However, when the sudo srsue builds successfully,
Opening USRP with args:
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Starting AGC thread...
Tunning receiver to 2450.000 MHz
set RX frequency to 2449999999
Searching for cell...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz

frankist, 2019-04-18:
Hey, sorry didnt realise you were using bladerf in the computer you are
running the ue. UHD is only for usrps.

Move this conversation to the SRS mailing list, and send the logs (found in
the /tmp/ue.log and /tmp/enb.log files) as well.

Regards,
Francisco

Zethernos, 2019-04-29:
Hi thanks a lot for your help. I would thank you for helping me. However, I have followed the steps you have provided the one with bladeRF doesn't seem to be working. Is there an alternative method where I can utilise 3.09LTS on the Ubuntu 18.04. Currently the Ubuntu 16.04 on the b210 is able to run with sudo srsue with UHD 3.09.07 while the Ubuntu 18.04 on the bladeRF with sudo srsepc and srsenb. I understand I shouldn't be posting here, however, I have sending to the srslte mail and it has been several days that my posts are not posted or no response from the moderator. I am grateful if these is something that I can rectify because I have read several posts.

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /home/pi/.config/srslte/epc.conf...
HSS Initialized.
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- IMSI Style Attach request
Attach request -- IMSI: 001010123456789
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 1
Attach Request -- UE Network Capabilities EEA: 11100000
Attach Request -- UE Network Capabilities EIA: 01100000
Attach Request -- MS Network Capabilities Present: false
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: false
Downlink NAS: Sending Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 001010123456789
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001010123456789
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 1010123456789
Creating Session Response -- MME control TEID: 1
SPGW: Allocated Ctrl TEID 1
SPGW: Allocated User TEID 1
SPGW: Allocate UE IP 172.16.0.2
Received Create Session Response
Create Session Response -- SPGW control TEID 1
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.2 to ISMI 001010123456789
Adding attach accept to Initial Context Setup Request
Initial Context Setup Request -- eNB UE S1AP Id 1, MME UE S1AP Id 1
Initial Context Setup Request -- E-RAB id 5
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100
Initial Context Setup Request -- QCI 7
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x500003; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 1010123456789
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Sending EMM Information
^CDeleting eNB context. eNB Id: 0x19b
Deleting UE EMM context. IMSI: 001010123456789
Deleting SP-GW GTP-C Tunnel. IMSI: 1010123456789
Deleting UE context in HSS. IMSI: 001010123456780
Deleting UE context in HSS. IMSI: 001010123456789

---  exiting  ---

Deleting UE context in HSS. IMSI: 001010123456789
[... 86 lines cut ...]
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
....
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-1.7 KHz
Found PLMN:  Id=00101, TAC=7
Setting PDN protocol to IPv4
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=22, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=26, ra-rnti=0x2
Random Access Transmission: seq=28, ra-rnti=0x2
Random Access Transmission: seq=34, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=25, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=39, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Setting PDN protocol to IPv4
Random Access Transmission: seq=37, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x50, ta=0
Network attach successful. IP: 172.16.0.2
Software Radio Systems LTE (srsLTE)
Warning: Detected Radio-Link Failure
RRC Connection Reestablishment
RRC IDLE