srsran/srsRAN_4G #332: UE doesn't connect

GonCasMor, 2019-04-25:
Hello, 

We have two LimeSDRs and we execute the srsLTE perfectly with a mobile phone.
But, we have problems connecting the UE by doing "srsue". It seems that UE sometimes doesn't create the subnetwork (in fact, one time we got the subnetwork but we were unable to make pings).
I don't know if it could be possible because the AGC is not supported?
We let some part of the shell messages:
---------------------------

Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3650, f_dl=945.0 MHz, f_ul=900.0 MHz
Tuned Rx to 945.00 MHz
Tuned Tx to 900.00 MHz
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Tx calibration finished
..
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-0.0 KHz
Set master clock rate to 11.52 MHz
[INFO] RX LPF configured
Set Rx bandwidth to 8.64 MHz
[INFO] Rx calibration finished
SoapySDRDevice_readStream returned -6: TIME_ERROR
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 8.64 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 8.64 MHz
[INFO] Tx calibration finished
Found PLMN:  Id=90170, TAC=7
Setting PDN protocol to IPv4
Random Access Transmission: seq=32, ra-rnti=0x2
[WARNING] L
[WARNING] L
Random Access Transmission: seq=33, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x2e0, ta=0
Warning: Detected Radio-Link Failure
RRC Connection Reestablishment
LRandom Access Transmission: seq=7, ra-rnti=0x2
Random Access Transmission: seq=33, ra-rnti=0x2
Random Access Transmission: seq=49, ra-rnti=0x2
Random Access Transmission: seq=12, ra-rnti=0x2
**Warning: Detected Radio-Link Failure
RRC Connection Reestablishment**
LRandom Access Transmission: seq=17, ra-rnti=0x2
Random Access Transmission: seq=12, ra-rnti=0x2
Random Access Transmission: seq=37, ra-rnti=0x2
Random Access Transmission: seq=3, ra-rnti=0x2
**RRC IDLE
Failed to Attach**
**[WARNING] popping from TX, samples popped 597/1020
[WARNING] popping from TX, samples popped 0/1020**
Setting PDN protocol to IPv4
Random Access Transmission: seq=2, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x2f9, ta=0
Scheduling request failed: releasing RRC connection...
Random Access Transmission: seq=23, ra-rnti=0x2
Random Access Transmission: seq=2, ra-rnti=0x2
Random Access Transmission: seq=22, ra-rnti=0x2
Random Access Transmission: seq=35, ra-rnti=0x2
Random Access Transmission: seq=45, ra-rnti=0x2
Random Access Transmission: seq=39, ra-rnti=0x2
Warning: Detected Radio-Link Failure
RRC Connection Reestablishment

Thank you so much for your support.
Gon_Cas_Mor

andrepuschmann, 2019-04-25:
For the Lime, try to compile with LTE rates, i.e. `cmake -DUSE_LTE_RATES=True ..`

andrepuschmann, 2019-04-25:
Closed because its not a code issue. Please use the mailing list for this.

closed 2019-04-25 by andrepuschmann