srsran/srsRAN_4G #333: Uncontrolled commercial UE TX power growth.

legden, 2019-04-29:
At certain conditions commercial UE constantly increase its transmit power and decrease PHR until it reach its limit. It's seems that behaviour explained by lack of closed loop power control in srsenb. TPC is always set to 1 in DCI messages from srsenb. With help of srsue i had logging TPC value received at UE side. It seems that in bad channel conditions blind search sometimes detect wrong DCI messages with TPC=3. Don't sure is it expected behaviour, but it happens. Uplink power control set to accumulated mode for PUSCH and it always accumlated for PUCCH. So rarely wrongly received TPC=3 cause UE to change power in one way only.

andrepuschmann, 2019-10-15:
Thanks for reporting. We are aware that the eNB power control isn't fully working yet. We'll be addressing this in one of the upcoming releases.

joeldelosangeles, 2020-07-30:
Has closed loop power control been implemented in the latest srsenb version?  Thanks

xavierarteaga, 2020-07-30:
No, it is not implemented. It is definitely in our TODO list but I cannot give you an exact release when this feature will be implemented. Once again, many thanks for your interest.

andrepuschmann, 2020-10-27:
As has been mentioned, power control will be added in one of the upcoming releases.

closed 2020-10-27 by andrepuschmann

joeldelosangeles, 2021-09-29:
Hi,

Has this been addressed in 21.04?

• 21.04
– Add UL power control, measurement gaps and a new proportional fair scheduler to srsENB

We designed our own power amplifier and LNA.  In the lab, without the PA/LNA, we were able to achieve 21 Mbps UDP at 10 MHz in the uplink.  However when we connected our frontend in the lab with the enodeb and UE 4 meters away from each other, we were only able to get 2 Mbps.  We added up to 50 dB of attentuation (using physical attenuators) but the result is the same.
Our PA works fine as confirmed by the power output.  In the field we can get youtube streaming up to 315 meters at 3 and 5 MHz (to avoid interference).   But uplink throughput is very low (kbps) at these conditions.  Of course, we are troubleshooting our LNA but I wonder, could this issue be related to UL power control?

As a side note we are following this table for rs_power in sib.conf at 5 watts output power per port after our PA.

Bandwidth (MHz)	rs_power
10	                            12
5	                            15
3	                            17

Will appreciate your insights.

Thanks

ismagom, 2021-09-29:
Can you set all logs to info in enb.conf and share your config files and enb.log please?

joeldelosangeles, 2021-09-29:
Hi Ismael,

Really appreciate your reply.  You will find the information files in the attachment.  Hope you can assist me.  Thanks, Joel
[test results.zip](https://github.com/srsran/srsRAN/files/7250754/test.results.zip)
[srsenb confs.zip](https://github.com/srsran/srsRAN/files/7250755/srsenb.confs.zip)
[enb.zip](https://github.com/srsran/srsRAN/files/7250757/enb.zip)

ismagom, 2021-09-29:
You send logs that don't correspond to console output in time. However I've been able to identify the following issues:

1) You have RT issues (message `RF status O=3, U=2, L=60` in the screen comes from the USRP not being able to process samples fast enough)
2) It looks like you are pushing 1 Mbps UDP throughput only. that's why you only get 1 Mbps for user 0x97. The BLER for this user is 0%
3) For user 0x96, you are doing full DL traffic simultaneously to UL. You seem to have cross-interference. Make sure the TX and RX antennas are orthogonal or use a duplexor.
4) In both cases, the power control seems to work well, since both UEs are reporting PHR=40 dB, which means they are transmitting their lowest power
5) Whenever you send console output logs, use text mode not screen capture

Hope that helps
thanks

joeldelosangeles, 2021-09-29:
Hi Ismael, we are pushing 20 Mbps from the iperf client (the UE).  What you saw is the iperf server side.  We are using duplexers and a 2x2 MIMO antenna.  We were able to increase the uplink throughput by connecting 50 db attenuators after the power amplifiers and before the duplexers.   So we are thinking that the issue is related to uplink power control.  The PHR adjusted to lower levels and the uplink SNR improved.  Our current version is srs 20.4.1.  See attachments in text files.  The config files are unchanged. Thanks again for the prompt response.  

[srsenb trace.txt](https://github.com/srsran/srsRAN/files/7251096/srsenb.trace.txt)
[uplink throughput testing.txt](https://github.com/srsran/srsRAN/files/7251099/uplink.throughput.testing.txt)

joeldelosangeles, 2021-10-01:
Hi,

We decided to upgrade to UHD 3.15-LTS and srsRAN 21.4.0.   Right UE's cannot camp to the enodeb.  Attached is the enb.log.  Hope you can help. Thanks
[Phones can't camp to srslte 21.04.txt](https://github.com/srsran/srsRAN/files/7265753/Phones.can.t.camp.to.srslte.21.04.txt)

joeldelosangeles, 2021-10-05:
Hi,

I am using UHD 3.14 and srsRAB 21.4.0.   I hope you can check the attached files for any errors since I still can't connect a UE although S1 is OK.  Thanks
[enb.log](https://github.com/srsran/srsRAN/files/7284110/enb.log)
[enb.zip](https://github.com/srsran/srsRAN/files/7284113/enb.zip)

yc541, 2022-02-18:
@joeldelosangeles 

We have been using PA/LNA (PA max output is 43dBm) as well and experienced similar UL power control issues (v21.10 NSA mode). From the trace the UL SNR is really bad and has high UL loss rate. 

Since PA is used the rs_power in sib.conf should be adjusted (it's 0dBm by default) to reflect the amplified reference signal power, otherwise the estimated PL would be too small therefore resulting reduced UL power (which results bad UL SNR). p0_nominal_pusch and alpha in sib.conf also contribute to the UL power control, which need to be tweaked according to the RF conditions.

avalori1, 2024-04-11:
Hi @andrepuschmann,
in which version has this been solved (if it has been solved)? Version 22.4.0 seems to be still affected...

Thanks!

avalori1, 2024-10-15:
@andrepuschmann and @xavierarteaga , is there any update on this? I moved now to 23.4.0 and seems that is still happening...
Is there something wrong with my setting, or the plan to add power control was removed?
Thanks