srsran/srsRAN_4G #334: Unable to connect to the internet with the laptop using sudo srsue

Zethernos, 2019-05-03:
Hi everyone, I appreciate for the help here so far, I understand that I am not supposed to post here instead of srslte mailing systems, I just sent an email there. I am currently trying to connect the laptop running with sudo srsue to go into the internet, while building sudo srsepc and sudo srsenb simultaneously. Do correct me if I am wrong, before using a mobile device as the ue, I heard it is possible to connect the computer running the sudo srsue to connect to the internet as a "test". I hope I can troubleshoot this error before using a mobile phone with USIM as the ue. As I am tasked to connect to the internet using the laptop connect to the srsue successfully, I am unsure of what did I went wrong.

 Before that, I have entered several configurations inside using IP address 172.16.0.2.  I am able to see the tun_srsue with the ip address with 172.16.0.2 while looking for active network connections while running the three commands. I have looked into several posts in srslte regarding this matter, please do let me know if I should amend. I will appreciate if anyone can add the steps too. I was told to make changes to the epc.conf file, however I am only able to identify srsue/ srsenb. Down below are the outputs while building sudo srsepc, sudo srsenb and sudo srsue. If you do need any necessary files please let me know. Thanks

pi@pi-ubuntu:~$ sudo srsenb

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/pi/.config/srslte/enb.conf...
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args:
Error opening UHD: code 11( from the srslte)
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
set TX frequency to 2684999936
set RX frequency to 2564999936
Setting Sampling frequency 11.52 MHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz

==== eNodeB started ===
Type <t> to view trace
RACH:  tti=1151, preamble=7, offset=0, temp_crnti=0x46( from the srslte)
SCHED: Could not transmit RAR within the window (RA TTI=1151, Window=10, Now=1165)
RACH:  tti=1181, preamble=49, offset=0, temp_crnti=0x47
SCHED: Could not transmit RAR within the window (RA TTI=1181, Window=10, Now=1195)
RACH:  tti=1211, preamble=23, offset=0, temp_crnti=0x48
SCHED: Could not transmit RAR within the window (RA TTI=1211, Window=10, Now=1225)
Disconnecting rnti=0x46.
RACH:  tti=1241, preamble=0, offset=0, temp_crnti=0x49
SCHED: Could not transmit RAR within the window (RA TTI=1241, Window=10, Now=1260)
RACH:  tti=1271, preamble=6, offset=0, temp_crnti=0x4a
Disconnecting rnti=0x47.
Disconnecting rnti=0x48.
Disconnecting rnti=0x49.
User 0x4a connected
Disconnecting rnti=0x4a.

pi@pi-ubuntu:~$ sudo srsepc

Built in Release mode using commit 429ee901 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /home/pi/.config/srslte/epc.conf...have consulted others some tell me to configure the
HSS Initialized.
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- GUTI Style Attach request
Attach request -- M-TMSI: 0x7ec9d66e
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 1have consulted others some tell me to configure the
Attach Request -- UE Network Capabilities EEA: 11100000
Attach Request -- UE Network Capabilities EIA: 01100000
Attach Request -- MS Network Capabilities Present: false
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: false
UL NAS: Received Identity Response
ID Response -- IMSI: 001010123456789
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 001010123456789
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001010123456789
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 1010123456789
Creating Session Response -- MME control TEID: 1
SPGW: Allocated Ctrl TEID 1
SPGW: Allocated User TEID 1
SPGW: Allocate UE IP 172.16.0.2
Received Create Session Response
Create Session Response -- SPGW control TEID 1
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.2 to ISMI 001010123456789
Adding attach accept to Initial Context Setup Request
Initial Context Setup Request -- eNB UE S1AP Id 1, MME UE S1AP Id 1have consulted others some tell me to configure the
Initial Context Setup Request -- E-RAB id 5
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100
Initial Context Setup Request -- QCI 7
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x4a0003; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 1010123456789
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Sending EMM Information
Received UE Context Release Request. MME-UE S1AP Id 1
There are active E-RABs, send release access bearers request
UE is ECM IDLE.
Received UE Context Release Complete. MME-UE S1AP Id 1
UE is not ECM connected. No need to release S1-U. MME UE S1AP Id 1
UE Context Release Completed.
SCTP Association Shutdown. Association: 24
Deleting eNB context. eNB Id: 0x19b
Releasing UEs context
Releasing UE ECM context. UE-MME S1AP Id: 0
^CDeleting UE EMM context. IMSI: 001010123456789
Deleting UE context in HSS. IMSI: 001010123456780
Deleting UE context in HSS. IMSI: 001010123456789

pi@pi:~$ sudo srsuehave consulted others some tell me to configure the

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.007-5-ge8202128

Reading configuration file /home/pi/.config/srslte/ue.conf...

Built in Release mode using commit 429ee90 on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...

Opening USRP with args: type=b200,master_clock_rate=30.72e6

-- Detected Device: B210

-- Operating over USB 3.

-- Initialize CODEC control...

-- Initialize Radio control...

-- Performing register loopback test... pass

-- Performing register loopback test... pass

-- Performing CODEC loopback test... pass

-- Performing CODEC loopback test... pass

-- Asking for clock rate 30.720000 MHz...

-- Actually got clock rate 30.720000 MHz.

-- Performing timer loopback test... pass

-- Performing timer loopback test... pass

Waiting PHY to initialize...

...

Attaching UE...

Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz

...

Found Cell:  PCI=1, PRB=50, Ports=1, CFO=-0.4 KHz

Found PLMN:  Id=00101, TAC=7

Setting PDN protocol to IPv4

Random Access Transmission: seq=7, ra-rnti=0x2

Random Access Transmission: seq=49, ra-rnti=0x2

Random Access Transmission: seq=23, ra-rnti=0x2

Random Access Transmission: seq=0, ra-rnti=0x2

Random Access Transmission: seq=6, ra-rnti=0x2

RRC Connected

Random Access Complete.     c-rnti=0x4a, ta=0

Network attach successful. IP: 172.16.0.2

Software Radio Systems LTE (srsLTE)

Warning: Detected Radio-Link Failure

RRC Connection Reestablishment

RRC IDLE

andrepuschmann, 2019-05-03:
You are right, you should use the mailing list for this.

closed 2019-05-03 by andrepuschmann

jin-github1002, 2020-04-23:
im sorry for leaving comments here. Was this issue included in the mailing list? if it was, how can i find it? i have the same problem now. It is so boring for me. Thanks for your help @Zethernos @andrepuschmann