srsran/srsRAN_4G #337: UE does pings with enb but UE hasn't internet

GonCasMor, 2019-05-18:
Hi again,

Finally we have connected the UE with a Usim Card (sysmocom) with our two LimeSDR:

Random Access Transmission: seq=19, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=1
Network attach successful. IP: 172.16.0.2
Software Radio Systems LTE (srsLTE)

tun_srsue: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 172.16.0.2  netmask 255.255.255.0  destination 172.16.0.2
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1  bytes 48 (48.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Then we did a ping from the UE to ENB and it was successful:
ping 172.16.0.1
PING 172.16.0.1 (172.16.0.1) 56(84) bytes of data.
64 bytes from 172.16.0.1: icmp_seq=1 ttl=64 time=28.1 ms
64 bytes from 172.16.0.1: icmp_seq=2 ttl=64 time=16.8 ms
64 bytes from 172.16.0.1: icmp_seq=3 ttl=64 time=14.0 ms
64 bytes from 172.16.0.1: icmp_seq=4 ttl=64 time=11.8 ms

But when we tried to do a ping to internet, failed
ping 8.8.8.8
connect: Network is unreachable

We don't understand why is not working because with a mobile phone works Ok and we can surf in the internet (The masquerade is done).

Please, help us.

We will grateful.

Thanks

frankurcrazy, 2019-05-28:
Have you tried `ip route` and see if the default gateway is configured to `172.16.0.1`?

GonCasMor, 2019-06-02:
Yes. 
We have execute: sudo route add default gw 172.16.0.1 tun_srsue, and as a result we got this:

lab138@NUC8i5BEH:/etc/network$ ip route
default via 172.16.0.1 dev tun_srsue 
169.254.0.0/16 dev tun_srsue scope link metric 1000 
172.16.0.0/24 dev tun_srsue proto kernel scope link src 172.16.0.5 

Any suggestion?

Thank you so much

ismagom, 2019-06-06:
Use mailist please. Thanks

closed 2019-06-06 by ismagom