srsran/srsRAN_4G #339: UHD ERROR 44 -- No compatible RF frontend found

pgupta408, 2019-05-26:
Why was this closed without any explanation? 

As per UHD documentation, this is application programming error. 

Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
[Describe the issue in detail]

--- Software Radio Systems LTE eNodeB ---

Reading configuration file /etc/srslte/enb.conf...
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.13.0.1-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [B200] Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
Opening USRP with args: type=b200,master_clock_rate=30.72e6
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
Error opening UHD: code 44
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_imp.c.126: No compatible RF frontend found

/home/ubuntu/srsLTE/lib/src/radio/radio.cc.38: Error opening RF device

Failed to find device auto with args auto

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2019-05-26:
I am closing this again. Please follow the guidelines listed before filing an issue and please don't file issues multiple times. If you want help you need to do your homework and provide sufficient information for the community to help.

closed 2019-05-26 by andrepuschmann

pgupta408, 2019-05-26:
This is very rude to close without explanation or help. 

I have not reached this far without homework. 

Help solve the issue or offer hints to solve or ask questions for more logs. 
 
Your installation description needs updating to align proper installation of UHD drivers or other SDR drivers for different OS supported. 

Community can help in documentation if community understands your installation process better. 

I can provide notes after I am able to run successfully.. 

Thanks

Closed #339.

pgupta408, 2019-05-27:
Hi Andre,

 

Just FYI.. Solved issues.. Problem was around ZMQ.. Lib is different for ubuntu 16.04 Vs 17.04.. 

 

Since UBUNTU-17.04 is no longer supported and 18.04 is not a stable release. So, Ubuntu 16.04 was only option. 

 

I had to use Ubuntu 16.04 for UHD for ZMQ and use 17.04 lib for ZMQ.. 

 

Thanks

-Praveen

https://www.mobilestack.com <https://www.mobilestack.com/> 

 

Article:-

Ten steps of adding LTE to a Wearable -  <https://www.mobilestack.com/blog/ten-steps-of-adding-lte-to-a-wearable/> https://www.mobilestack.com/blog/ten-steps-of-adding-lte-to-a-wearable/

5G Pillars for network densification - https://www.mobilestack.com/blog/5g-infrastructure-pillers-for-densification/

5G for Cable Industry -  <https://www.mobilestack.com/blog/5g-business-opportunity-for-cable-industry/> https://www.mobilestack.com/blog/5g-business-opportunity-for-cable-industry/

5G in India -   <https://www.mobilestack.com/blog/5g-in-india-need-to-create-new-public-private-partnership-projects/> https://www.mobilestack.com/blog/5g-in-india-need-to-create-new-public-private-partnership-projects/

Video:

SDR-based Private LTE demo - https://www.youtube.com/watch?v=ZyGyreFG5wI

Intelligent Edge GW Demo – coming soon

 

 <https://www.mobilestack.com/> 

Follow us at:   <https://twitter.com/mobilestackinc>  <https://www.linkedin.com/company/Mobilestack-inc>   <https://www.facebook.com/MobilestackInc/> 

This e-mail message, and any attachments, is intended only for the use of the individual or entity identified in the alias address of this message and may contain information that is confidential, privileged and subject to legal restrictions and penalties regarding its unauthorized disclosure and use. Any unauthorized review, copying, disclosure, distribution or use is strictly prohibited. If you have received this e-mail message in error, please notify the sender immediately by reply e-mail and delete this message, and any attachments, from your system. Thank you for your cooperation.

Mkishore1, 2020-01-09:
@andrepuschmann andrepuschmann :
Hey I have the same problem with bladeRFx40running on Ubuntu 16.04
steps to reproduce the error:
~/srsLTE-master/srsue$ sudo srsue ue.conf.example
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Reading configuration file ue.conf.example...

Built in Release mode using 19.12.0.

Opening 1 RF devices with 1 RF channels...
Opening USRP with args: 
Error opening UHD: code 11
/home/crrao/srsLTE-master/lib/src/phy/rf/rf_imp.c.130: No compatible RF frontend found

/home/crrao/srsLTE-master/lib/src/radio/radio.cc.36: Error opening RF device

Failed to find device auto with args auto
Error initializing radio.
Waiting PHY to initialize ... terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---
Please help.

andrepuschmann, 2020-01-09:
If you have a bladeRF you need to install the bladerf driver, not UHD.

Mkishore1, 2020-01-13:
@andrepuschmann :
I have installed BladeRF libraries.
In my setup srsenb & srsepc  running using B210 and srsue using bladeRFx40.I am trying to connect UE TO ENB but got this error.

steps to reproduce the problem:
~/srsLTE-master/srsue$ sudo srsue ue.conf.example
Reading configuration file ue.conf.example...

Built in Release mode using 19.12.0.

Opening 1 RF devices with 1 RF channels...
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Waiting PHY to initialize ... /srsLTE-master/srsue/src/phy/sync.cc.722: Error stopping AGC: not implemented

done!
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2685000000
set TX frequency to 2565000000
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.
Found Cell:  Mode=FDD, PCI=1, PRB=15, Ports=1, CFO=4.9 KHz
Set RX sampling rate 3.84 Mhz, filter BW: 3.84 Mhz
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=18, ra-rnti=0x2
Random Access Transmission: seq=19, ra-rnti=0x2
Random Access Transmission: seq=4, ra-rnti=0x2
Random Access Transmission: seq=31, ra-rnti=0x2
Random Access Transmission: seq=8, ra-rnti=0x2
Random Access Transmission: seq=4, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=48, ra-rnti=0x2
Random Access Transmission: seq=2, ra-rnti=0x2
Random Access Transmission: seq=44, ra-rnti=0x2
Warning: Detected Radio-Link Failure
RF status: O=11, U=0, L=0
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.
Found Cell:  Mode=FDD, PCI=1, PRB=15, Ports=1, CFO=5.0 KHz
Set RX sampling rate 3.84 Mhz, filter BW: 3.84 Mhz
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
.
Found Cell:  Mode=FDD, PCI=1, PRB=15, Ports=1, CFO=4.8 KHz
Set RX sampling rate 3.84 Mhz, filter BW: 3.84 Mhz
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=34, ra-rnti=0x2
Random Access Transmission: seq=50, ra-rnti=0x2
Random Access Transmission: seq=30, ra-rnti=0x2
Random Access Transmission: seq=36, ra-rnti=0x2
Random Access Transmission: seq=49, ra-rnti=0x2
Random Access Transmission: seq=27, ra-rnti=0x2
Random Access Transmission: seq=10, ra-rnti=0x2
Random Access Transmission: seq=40, ra-rnti=0x2
Random Access Transmission: seq=22, ra-rnti=0x2
Random Access Transmission: seq=25, ra-rnti=0x2
Warning: Detected Radio-Link Failure
^CStopping srsUE... Press Ctrl+C 9 more times to force stop
---  exiting  ---
please help.

Mkishore1, 2020-01-14:
hey anyone please respond !!

Mkishore1, 2020-01-23:
hey @andrepuschmann please help me to resolve this error!!

yagoda, 2020-01-23:
Hey Mkishore1,

By looking at your console output, it seems that your problem could be caused by CFO but I cant be sure without looking at the logs.

Please send an email to the srsLTE mailing list srslte-users@lists.softwareradiosystems.com and include logs from both the UE and ENB.

Regards,
Justin.