srsran/srsRAN_4G #342: UE and eNodeB Without Hardware

infinitydon, 2019-06-11:
I just want to ask if it is possible to emulate both UE and eNodeB (without using hardware devices)?

infinitydon, 2019-06-12:
I noticed the srsUE can be run simply as a standalone application.. Does anyone have an idea of how to simulate an eNodeB?

andrepuschmann, 2019-06-12:
The no-RF capabilities are still under development and not yet ready for general use.

closed 2019-06-12 by andrepuschmann

ChrisJiang, 2020-02-26:
Any update on this feature now?

nhnghia, 2020-02-26:
I tested the communication between UE and eNodeB using ZeroMQ-based fake RF drive. It works well.

victory377, 2020-04-23:
> I tested the communication between UE and eNodeB using ZeroMQ-based fake RF drive. It works well.

Could you share the more detailed information such ue.conf, enb.conf... using ZeroMQ-base fake RF driver to make connection between ue and enb? 

thanks.

nhnghia, 2020-04-23:
You may find detailed information here: https://docs.srslte.com/en/latest/app_notes/source/zeromq/source/index.html

victory377, 2020-04-24:
> You may find detailed information here: https://docs.srslte.com/en/latest/app_notes/source/zeromq/source/index.html

thank for your info.

Have you experienced one machine epc and one machine ue and enb?

UE and enb use ZeroMQ-base fake RF driver to communicate each other.

Enb another side uses ethernet interface to communicate with epc.

topology as below,
[UE <--zeroMQ--> enb ]<--- ethernet ---> [epc]

infinitydon, 2020-04-25:
@victory377 - This is a sample of what I used:

```
[enb]
enb_id = 0x19B
cell_id = 0x01
phy_cell_id = 1
tac = 0x3039
mcc = 208
mnc = 93
mme_addr = 10.1.1.121
gtp_bind_addr = 10.0.0.192
s1c_bind_addr = 10.0.0.192

[rf]
device_name=zmq
device_args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6"

[expert]
nof_phy_threads=1

[enb_files]
sib_config = sib.conf
rr_config  = rr.conf
drb_config = drb.conf
```

UE/eNodeB was running on a different system, the mme was running in a separate network.

You can check my repo for the file:

https://bitbucket.org/infinitydon/virtual-4g-simulator/src/master/srsLTE/srsenb/enb.conf

Hope this helps.

alam2018, 2021-01-13:
I have no RF device installed. I am trying to run the ue in one machine and enb plus epc in another machine. The topology is below:
[ue] <--- ethernet ---> [enb + epc]

For ue i am using the following command: sudo srsue --rf.device_name zmq --rf.device_args "rx_port=tcp://<ipenb>:5554,tx_port=tcp://*:5555,id=enb,base_srate=1.92e6"

For enb i am using the following command: sudo srsenb --rf.device_name zmq --rf.device_args "rx_port=tcp://<ipUE>:5554,tx_port=tcp://*:5555,id=enb,base_srate=1.92e6"

But the ue cannot connect to the enb. I have checked the connectivity between the machines and that works fine. 

With the ZMQ library is it possible to run the ue and enb on different machines. if yes, what would be the command for starting the ue and enb?

shakthivels300, 2023-03-09:
@infinitydon Hi! I wish to use your Virtual 4g simulator. Is that in a working condition still. Are the links alive? Shall I give it a try? Can I get your mail ID?