srsran/srsRAN_4G #344: short circuit return leaves mutex locked

jgiovatto, 2019-06-13:
Im running into a condition where tbs < 4. Note this is in an emulated rf environment which Im still looking into the inputs.

```
 SCHED: Allocation of TBS=2 that does not account header
```
After this log the enb stops transmitting so I think this needs some attention.

```
srsenb/src/mac/scheduler_ue.cc

int sched_ue::generate_format1(dl_harq_proc* h, sched_interface::dl_sched_data_t* data, uint32_t tti, uint32_t cfi)
{
  pthread_mutex_lock(&mutex);
...
    if (need_conres_ce and cell.nof_prb < 10) { // SRB0 Tx. Use a higher MCS for the PRACH to fit in 6 PRBs
      tbs = srslte_ra_tbs_from_idx(srslte_ra_tbs_idx_from_mcs(MCS_FIRST_DL, false), nof_prb) / 8;
      mcs = MCS_FIRST_DL;
    } else if (fixed_mcs_dl < 0) {
      tbs = alloc_tbs_dl(nof_prb, nof_re, req_bytes, &mcs);
    } else {
      tbs = srslte_ra_tbs_from_idx(srslte_ra_tbs_idx_from_mcs(fixed_mcs_dl, false), nof_prb) / 8;
      mcs = fixed_mcs_dl;
    }
...
    if (tbs < MIN_DATA_TBS) {
      log_h->warning("SCHED: Allocation of TBS=%d < %d that does not account header\n", tbs, MIN_DATA_TBS);
      return 0;   <<<< short circuit return
    }
...
 
  pthread_mutex_unlock(&mutex); <<<< normal return
  return tbs;
}
```

andrepuschmann, 2019-06-14:
Very nice catch Joe! Thanks for that. We look after it. Will be fixed in the next release.

jgiovatto, 2019-06-14:
We try to use these when possible.

https://en.cppreference.com/w/cpp/thread/lock_guard

andrepuschmann, 2019-06-14:
Yes, we'll migrate all pthread mutexes to std::mutex and lock_guard/unique_lock

andrepuschmann, 2019-07-01:
Fixed in new release.

closed 2019-07-01 by andrepuschmann