srsran/srsRAN_4G #345: Function to calculate the DCI format length

YaxiongXiePrinceton, 2019-06-13:
Dear all,

When I am reading the code of calculating the DCI format length, I got one problem.

According to the code (i.e., the function dci_format1B_sizeof()), the length of DCI format 1B depends on the length of 1A, 
  uint32_t n = dci_format1A_sizeof(cell, sf, cfg) - 1 + tpmi_bits(cell->nof_ports) + 1 + (IS_TDD ? 2 : 0);

The length of DCI format 1A depends on whether the SRS request flag is set or not. If the SRS flag is set, then the length of DCI format 1A is 1 bit longer. 

But, DCI format 1B doesn't have the SRS request flag, so its length doesn't depend on this flag. 

Therefore, according to the implementation of the function dci_format1B_sizeof(), the length of DCI format 1B depends on the length of format 1A and hence the SRS flag. But theoretically, length DCI format 1B doesn't depend on this flag. 

Am I understanding this correctly? Or is there any problem in the implementation?

xavierarteaga, 2019-08-19:
Hi @YaxiongXiePrinceton, many thanks for pointing this out. Your understanding is correct and this is an implementation defect that needs to get a fix.

Xavier

andrepuschmann, 2019-10-14:
Closing issue. Will be tackled in next release.

closed 2019-10-14 by andrepuschmann