srsran/srsRAN_4G #347: enb lib error messages when ebable mbsfn and prb < 50

jgiovatto, 2019-06-14:
Noted when running fake rf driver enb output especially noticeable with UL traffic.

```
lib/src/phy/enb/enb_dl.c.385: Error encoding UL DCI message
srsenb/src/phy/sf_worker.cc.710: Error putting PUSCH 0
lib/src/phy/phch/pdcch.c.621: Illegal DCI message nCCE: 16, L: 1, nof_cce: 13, nof_bits=25
lib/src/phy/enb/enb_dl.c.385: Error encoding UL DCI message
srsenb/src/phy/sf_worker.cc.710: Error putting PUSCH 0
lib/src/phy/phch/pdcch.c.621: Illegal DCI message nCCE: 12, L: 1, nof_cce: 13, nof_bits=25
lib/src/phy/enb/enb_dl.c.385: Error encoding UL DCI message
srsenb/src/phy/sf_worker.cc.710: Error putting PUSCH 0
lib/src/phy/phch/pdcch.c.621: Illegal DCI message nCCE: 18, L: 1, nof_cce: 13, nof_bits=25
```

ue.conf changes
```
-#mbms_service        = -1
+mbms_service         = 0
-#snr_estim_alg       = refs
+snr_estim_alg        = pss
-#interpolate_subframe_enabled = false
+interpolate_subframe_enabled = true
```

enb.conf changes

```
-n_prb = 50
+n_prb = 25
-sib_config = sib.conf
+sib_config = sib.conf.mbsfn
-nof_ctrl_symbols = 3
+nof_ctrl_symbols = 2
-#enable_mbsfn = false
+enable_mbsfn = true
```

jgiovatto, 2019-06-25:
The above logs were printed to stderr via library ERROR logger.
Additional MAC warning logs below seem to be related:
```11:30:38.001207 [MAC ] [W] [07036] ACK=0 received pid=4, tb_idx=0, n_rtx=3, max_retx=4
11:30:38.001229 [MAC ] [W] [07036] SCHED: discarting TB 0 pid=4, tti=7036, maximum number of retx exceeded (4)
```

yagoda, 2019-07-25:
Hey Joe,

I think if you set the number of control symbols in the enodeb to 2 this problem should go away.

jgiovatto, 2019-07-25:
Hi

I did set it to 2, changes to the default config were noted in the isuue.

|-nof_ctrl_symbols = 3 +nof_ctrl_symbols = 2 Joe |

yagoda, 2019-07-25:
OK noted we will see if we can reproduce the issue.
Thanks!
Justin

andrepuschmann, 2019-10-15:
I am closing the issue for the moment. Please reopen if it is still present in the current release.

closed 2019-10-15 by andrepuschmann

Borjis131, 2020-02-04:
Im getting the same errors with 19.12 release. Any update about this?