srsran/srsRAN_4G #349: Invalid use of CHECK_LIBRARY_EXISTS() in FindbladeRF.cmake

dottorespina, 2019-07-04:
## Issue Description ##
CMake permits to easily build and install different *platforms*, each providing different versions of the same set of locally built libraries and tools.
To achieve this, one can apply a simple pattern:
* consistently use CMake with the `-DCMAKE_INSTALL_PREFIX=/path/to/custom/prefix` parameter, where `/path/to/custom/prefix` is NOT `/usr/local`, etc  
* set environment accordingly (`PATH`, `LD_LIBRARY_PATH`, and the likes) to select a platform at runtime

An invalid use of `CHECK_LIBRARY_EXISTS()` in `FindbladeRF.cmake` prevents CMake to **_find_** the libbladeRF library, and thus srsLTE to pick the bladeRF driver up when it's installed to a custom prefix.

AFAICT, `FindbladeRF.cmake` would currently succeed only in a couple of context:
* libbladeRF is installed at system level (for e.g. with a package manager), AND no `DCMAKE_INSTALL_PREFIX` parameter is given to CMake
* libbladeRF is installed to `/usr/local`,  AND no `DCMAKE_INSTALL_PREFIX` parameter is given to CMake, AND the linker is configured to search this location

These above requirements do not play well with some common usages.

## Setup Details ##
Linux 5.1.15
CMake 3.14.5
GCC 9.1.1
libbladeRF 2.2.0
srsLTE: commit 5343b33f (2019-05-13)

## Expected Behavior ##
CMake should **_find_** the libbladeRF library when it's installed to the same custom prefix we use when building srsLTE.

## Actual Behaviour ##
Running `cmake -DCMAKE_INSTALL_PREFIX=/path/to/custom/prefix` fails to *find* the bladeRF driver library:
```
-- Checking for module 'libbladeRF'
--   Package 'libbladeRF', required by 'virtual:world', not found
-- Looking for bladerf_get_board_name in bladeRF
-- Looking for bladerf_get_board_name in bladeRF - not found
-- libbladeRF found but not compatible. Upgrade your driver or use SoapySDR.
```

Note that the bladeRF library is correctly installed to:
* `/path/to/custom/prefix/include/libbladeRF.h`
* `/path/to/custom/prefix/lib/libbladeRF.so`

## Steps to reproduce the problem ##
1. Build and install the libbladeRF library to a custom prefix, for e.g. `/opt/sdr-platform` 
2. `git clone https://github.com/srsLTE/srsLTE.git`
3. `cd srsLTE`
4. `mkdir build && cd build`
5. `cmake -DCMAKE_INSTALL_PREFIX=/opt/sdr-platform ../`
6. CMake does not *find* the bladeRF library: `libbladeRF found but not compatible.`

## Additional Information ##
Adding some debug messages in `cmake/modules/FindbladeRF.cmake`, it appears that the calls to both `find_path()` and  `find_libray()` actually do succeed (thus the `libbladeRF found` part):
* `BLADERF_INCLUDE_DIRS=/opt/sdr-platform/include`
* `BLADERF_LIBRARIES=/opt/sdr-platform/lib/libbladeRF.so` 

But, using the CMake option `--debug-trycompile`, we can see that `FindbladeRF.cmake` turns into a nonsense linker flag when testing for `bladerf_get_board_name()` existence (thus the `but not compatible` part):
```
/usr/lib64/ccache/cc  -DCHECK_FUNCTION_EXISTS=bladerf_get_board_name [...] -L/path/to/srsLTE/build/CMakeFiles/CMakeTmp/BLADERF_LIBRARIES [...] -lbladeRF
/usr/bin/ld: cannot find -lbladeRF
```
The nonsense is obviously `-L/path/to/srsLTE/build/CMakeFiles/CMakeTmp/BLADERF_LIBRARIES`, which feed the linker with a non existing path.  

Note that this invalid linker flag has no consequence when `libbladeRF.so` is installed to a path the linker *already knows of*, for e.g. as configured in `/etc/ld.so.conf` and the likes.

IMHO, this is caused by an invalid use of `CHECK_LIBRARY_EXISTS()` in `FindbladeRF.cmake`.

From CMake (3.8.2) API documentation:
```
CHECK_LIBRARY_EXISTS (LIBRARY FUNCTION LOCATION VARIABLE)

LIBRARY  - the name of the library you are looking for
FUNCTION - the name of the function
LOCATION - location where the library should be found
VARIABLE - variable to store the result
```

AFAICT, this does not match how this macro is called from `FindbladeRF.cmake`:
```
CHECK_LIBRARY_EXISTS(bladeRF bladerf_get_board_name BLADERF_LIBRARIES BLADERF_VERSION_OK)
```
It seems to me that `BLADERF_LIBRARIES` definitely is an invalid value for the parameter `LOCATION`.
Using `${BLADERF_LIBRARIES}` is a bit better, as it turns out to `-L/opt/sdr-platform/lib/libbladeRF.so`, but still invalid.

To get the correct linker flag, which is obviously `-L/opt/sdr-platform/lib`, one can do:
```
get_filename_component(bladerf_lib_dir ${BLADERF_LIBRARIES} DIRECTORY)
CHECK_LIBRARY_EXISTS(bladeRF bladerf_get_board_name ${bladerf_lib_dir} BLADERF_VERSION_OK)
```
I can confirm the above approach works, though I can't tell for sure it's the *right fix*.

I think this issue, if confirmed, should be treated as a bug, since it prevents srsLTE to pick the bladeRF driver up, although it's correctly installed *somewhere* CMake should *find* it.
And, clearly, the call to `CHECK_LIBRARY_EXISTS()` seems invalid. 

As a side note, other issues may also relate to UHD or bladeRF drivers being installed at a non standard prefix.
For e.g., reading issue #241:
* reporting user does not provide any `CMAKE_INSTALL_PREFIX`, so we might assume it's `/usr/local`
* in last comment, @andrepuschmann says: *It means that you don't have any RF drivers installed (or at least CMake doesnt pick them up).*

OP didn't tell which device/driver should have been *picked up*, nor which operating system is used, but AFAIK a lot of Linux distributions do not by default add `/usr/local/lib` to the paths the linker will search in; and on these systems, `FindbladeRF.cmake` would also fail at link time if the bladeRF library is installed to `/usr/local/lib`.

I had a quick look at `FindUHD.cmake`, and though it does not rely upon `CHECK_LIBRARY_EXISTS()`, i can't tell for sure it would succeed if the UHD driver is installed:
* in `/usr/local/lib` on a platform where the linker does not search in this location
* to a custom prefix using the `CMAKE_INSTALL_PREFIX ` parameter

I'll test this, and open a new issue if useful.

Thanks.

andrepuschmann, 2019-07-05:
Thanks for the comprehensive description. Feel free to open a pull request if you made it work. Thanks

dottorespina, 2019-07-08:
I've just opened PR #350 that contains the simple fix already described.
Let me know if it works for you.
Thanks.

andrepuschmann, 2019-10-08:
Fix has been merged. Thanks for this again

closed 2019-10-08 by andrepuschmann