srsran/srsRAN_4G #351: Running srsLTE at ZC702

shahram-WINS, 2019-07-15:
Hello
I am trying to run srsLTE on xilinx ZC702, I am using soapysdr driver and I am using PlutoSoapy because both board use the same chipset.
I couldn’t install the latest version of srsLTE on the ARM chipset of board but I managed to run its previous version (18.12.0). 
When I run it srsENB, I face with following output and the enb is not discoverable by any other devices:
Built in Release mode using 18.12.0.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...
Soapy has found device #0: device=plutosdr, driver=plutosdr, uri=local:, 
set_TX_1)
Set Tx bandwidth to 0.20 MHz
TX done!
set_RX_1)
set_RX_2)
set_RX_3)
set_RX_4)
Set Rx bandwidth to 2.50 MHz
set_RX_1)
Setting up RX stream
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Setting up TX stream
[INFO] Using format CF32.
[INFO] Has direct TX copy: 1
soapy5
Available device sensors: 
 - xadc_temp0
 - xadc_voltage0
 - xadc_voltage1
 - xadc_voltage2
 - xadc_voltage3
 - xadc_voltage4
 - xadc_voltage5
 - xadc_voltage6
 - xadc_voltage7
 - xadc_voltage8
 - adm1177_current0
 - adm1177_voltage0
 - ad9361-phy_temp0
 - ad9361-phy_voltage2
Available sensors for RX channel 0: 
State of gain elements for Rx channel 0 (AGC supported):
 - PGA: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PGA: 80.00 dB
Rx antenna set to A_BALANCED
Tx antenna set to A

Warning burst preamble is not calibrated for device plutosdr. Set a value manually

Receiving async metadata not supported by device. Exiting thread.
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
Tuned Tx to 2685.00 MHz
Tuned Rx to 2565.00 MHz
Set master clock rate to 0.00 MHz
Setting Sampling frequency 1.92 MHz
set_RX_1)
Failed to connect to MME - retrying in 10 seconds
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
set_RX_2)
set_RX_3)
set_RX_4)
Set Rx bandwidth to 2.50 MHz
set_RX_1)
set_TX_1)
Set Tx bandwidth to 0.20 MHz

==== eNodeB started ===
Type <t> to view trace

yagoda, 2019-07-25:
Hey! 

Nice work getting this far with the ZCU702! I assume you are using the ad9361 or something similar as a frontend.
Unfortunately, in order to use srsENB or srsUE with a zynq board with an analog devices frontend, you need to implement timestamping in the FPGA which is not supported by default by the analog devices firmware.

If you are interested in getting something working with this board it is possible to run the pdsch examples using a combination of the soapy drivers, and pluto drivers. Although some tweaking may be required in order to find success depending on the specific rf you are using?

If you want more details on this, post a message on the mailing list and I will reply as soon as possible.

Regards,
Justin

sinill57, 2019-07-26:
Hi!

I am too very interested in using zc702 as a SDR platform for srsLTE and would like to contribute to this issue. I have been working with ad9361 software and FPGA stack for quite a few years already, so if I knew more about what is required, could tailor the FPGA part a bit. Could you please tell more about the required timestamps?

Regards,
Ilya

shahram-WINS, 2019-08-02:
> Hey!
>
> Nice work getting this far with the ZCU702! I assume you are using the ad9361 or something similar as a frontend.
> Unfortunately, in order to use srsENB or srsUE with a zynq board with an [...]
Hello again
I tried to run pdsch example, although pdsch_enb works well but pdsch_ue will be stucked in the middle of run as follows:
ubuntu@arm:~/srsLTE-release_18_12/build/lib/examples$ sudo ./pdsch_ue -f 2560000000
Opening RF device with 1 RX antennas...
Soapy has found device #0: device=plutosdr, driver=plutosdr, uri=local:, 
soapy1=plutosdr
Set Tx bandwidth to 0.20 MHz
TX done!
Set Rx bandwidth to 2.50 MHz
Setting up RX stream
[INFO] Using format CF32.
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Setting up TX stream
[INFO] Using format CF32.
[INFO] Has direct TX copy: 1
Available device sensors: 
 - xadc_temp0
 - xadc_voltage0
 - xadc_voltage1
 - xadc_voltage2
 - xadc_voltage3
 - xadc_voltage4
 - xadc_voltage5
 - xadc_voltage6
 - xadc_voltage7
 - xadc_voltage8
 - adm1177_current0
 - adm1177_voltage0
 - ad9361-phy_temp0
 - ad9361-phy_voltage2
Available sensors for RX channel 0: 
State of gain elements for Rx channel 0 (AGC supported):
 - PGA: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PGA: 79.00 dB
Rx antenna set to A_BALANCED
Tx antenna set to A
Starting AGC thread...
Receiving async metadata not supported by device. Exiting thread.
rate  is  30720000.00
Set master clock rate to 0.00 MHz
Tunning receiver to 2560.000 MHz
Tuned Rx to 2560.00 MHz
Searching for cell...
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Set Rx bandwidth to 2.50 MHz
Could not find any cell in this frequency
Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
[INFO] Auto setting Buffer Size: 32768
[INFO] Set MTU Size: 32768
Set Rx bandwidth to 2.50 MHz

this is the same case for cell_search example.

yagoda, 2019-08-06:
Hey guys,

What is required of the timestamping for the UE and EnodeB to function properly is the ability to timestamp the exact moment a packet was received and to transmit at a precise moment, based on a shared clock between the transmit and receive chains.
To put it another way, if I receive a packet a time X, I need to be able to tell my transmitter to transmit at X + Y milliseconds and for this to happen with an accuracy of the order of microseconds.

We have an implementation of FPGA based timestamping as part of our commercial offering if this is something that may be of interest to you.
Regards,
Justin

shahram-WINS, 2019-08-16:
First of all, thank you so much for your response.
In order to implement time-stamping shall we apply it over "rf_soapy_send_timed" and "rf_soapy_recv_with_time_multi" functions in rf_soapy_imc module?
Do we need to use a hardware synchronization module such as GPS or can we do it with NTP as well?

andrepuschmann, 2019-10-15:
Closing the issue. We've got a commercial solution that adds timestamping for the above-mentioned board. Please contact sales@softwareradiosystems.com if you're interested.

closed 2019-10-15 by andrepuschmann

RamzuYunus, 2021-04-01:
> Hey guys,
>
> What is required of the timestamping for the UE and EnodeB to function properly is the ability to timestamp the exact moment a packet was received and to transmit at a precise moment, [...]

i am interested. If you can reply before i take this Pluto back for a refund. Thanks

ArielRFF, 2022-07-13:
Hello !

I am trying to run srsRAN on Pluto.  I succeeded to execute pdsch examples using SoapyPlutoSDR drivers, but can´t run srsENB.
I am exploring Pluto's iio drivers to add the required timestamps.  I am also making some research to adapt Pluto's Zynq 7010 FPGA.
Has somebody tried these approaches ?

ofontbach, 2022-07-13:
Hey, we have in our plans to open some code to add timestamping (both HDL and C code) to Zynq-based devices in the following months (a version using libiio shall be there for sure). So I'd recommend following our updates. Thanks!

ArielRFF, 2022-07-13:
> Hey, we have in our plans to open some code to add timestamping (both HDL and C code) to Zynq-based devices in the following months (a version using libiio shall be there for sure). So I'd recommend following our updates. Thanks!

Thanks @ofontbach I will following.  Regards.
Ariel.

Johnsel, 2022-11-02:
> Hey, we have in our plans to open some code to add timestamping (both HDL and C code) to Zynq-based devices in the following months (a version using libiio shall be there for sure). So I'd recommend following our updates. Thanks!

Any timeline on this? I'd like to build a project with a pluto, so it would be useful to have, and a prerequisite to build myself if it takes too long...

ofontbach, 2022-11-03:
We're tidying up the code and documentation, so should be available in the next few weeks. Regards

LunaWuna, 2022-11-16:
how is progress on this going? im planning to do a project and im waiting on this. thanks

ofontbach, 2022-11-18:
Hi @LunaWuna we aim at releasing by the end of next week. Cheers

ofontbach, 2022-11-28:
Hey, the repo is now live at https://github.com/srsran/zynq_timestamping.
Cheers