srsran/srsRAN_4G #352: Make failed, rf_zmq_imp.c:177:40

jonah01, 2019-07-16:
Hi,
i want to install srsLTE on an ubuntu 16.04 System, but the make process failed. I had searched for an issue like that, but i didn't find something like that.

I don't know what to do, maybe someone can give me a solution.
Best regards
jonah

Here is the message of the terminal:

after cmake ../

ubuntu@ubuntu-ThinkPad-T550:~/srsLTE/build$ make
Scanning dependencies of target rrc_asn1
[  0%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o
[  1%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1_enum.cc.o
[  1%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/asn1_utils.cc.o
[  1%] Linking CXX static library librrc_asn1.a
[  1%] Built target rrc_asn1
Scanning dependencies of target srslte_asn1
[  1%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_common.cc.o
[  1%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_mme.cc.o
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_s1ap.cc.o
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_m2ap.cc.o
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/gtpc.cc.o
[  2%] Linking CXX static library libsrslte_asn1.a
[  2%] Built target srslte_asn1
Scanning dependencies of target gen_build_info
-- Generating build_info.h
[  2%] Built target gen_build_info
Scanning dependencies of target srslte_common
[  3%] Building C object lib/src/common/CMakeFiles/srslte_common.dir/threads.c.o
[  3%] Building C object lib/src/common/CMakeFiles/srslte_common.dir/gen_mch_tables.c.o
[  3%] Building C object lib/src/common/CMakeFiles/srslte_common.dir/version.c.o
[  3%] Building C object lib/src/common/CMakeFiles/srslte_common.dir/crash_handler.c.o
[  4%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/task_dispatcher.cc.o
[  4%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/s1ap_pcap.cc.o
[  4%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/pdu_queue.cc.o
[  4%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/thread_pool.cc.o
[  5%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/snow_3g.cc.o
[  5%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/logger_file.cc.o
[  5%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/buffer_pool.cc.o
[  5%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/tti_sync_cv.cc.o
[  6%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/security.cc.o
[  6%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/pdu.cc.o
[  6%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/log_filter.cc.o
[  6%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/nas_pcap.cc.o
[  7%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/rlc_pcap.cc.o
[  7%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/mac_pcap.cc.o
[  7%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/liblte_security.cc.o
[  7%] Building CXX object lib/src/common/CMakeFiles/srslte_common.dir/arch_select.cc.o
[  8%] Linking CXX static library libsrslte_common.a
[  8%] Built target srslte_common
Scanning dependencies of target arch_select
[  8%] Building CXX object lib/src/common/CMakeFiles/arch_select.dir/arch_select.cc.o
[  9%] Linking CXX executable arch_select
[  9%] Built target arch_select
Scanning dependencies of target srslte_enb
[ 10%] Building C object lib/src/phy/enb/CMakeFiles/srslte_enb.dir/enb_ul.c.o
[ 10%] Building C object lib/src/phy/enb/CMakeFiles/srslte_enb.dir/enb_dl.c.o
[ 10%] Built target srslte_enb
Scanning dependencies of target srslte_agc
[ 10%] Building C object lib/src/phy/agc/CMakeFiles/srslte_agc.dir/agc.c.o
[ 10%] Built target srslte_agc
Scanning dependencies of target srslte_ch_estimation
[ 11%] Building C object lib/src/phy/ch_estimation/CMakeFiles/srslte_ch_estimation.dir/refsignal_ul.c.o
[ 11%] Building C object lib/src/phy/ch_estimation/CMakeFiles/srslte_ch_estimation.dir/chest_dl.c.o
[ 11%] Building C object lib/src/phy/ch_estimation/CMakeFiles/srslte_ch_estimation.dir/refsignal_dl.c.o
[ 11%] Building C object lib/src/phy/ch_estimation/CMakeFiles/srslte_ch_estimation.dir/chest_ul.c.o
[ 12%] Building C object lib/src/phy/ch_estimation/CMakeFiles/srslte_ch_estimation.dir/chest_common.c.o
[ 12%] Built target srslte_ch_estimation
[... 230 lines cut ...]
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:253:11: error: too many arguments to function ‘zmq_recv’
       n = zmq_recv(handler->receiver, handler->buffer_rx, BUFFER_SIZE, 0);
           ^
In file included from /home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:35:0:
/usr/include/zmq.h:230:16: note: declared here
 ZMQ_EXPORT int zmq_recv (void *s, zmq_msg_t *msg, int flags);
                ^
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c: In function ‘rf_zmq_open_multi’:
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:427:24: warning: implicit declaration of function ‘zmq_ctx_new’ [-Wimplicit-function-declaration]
     handler->context = zmq_ctx_new();
                        ^
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:427:22: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     handler->context = zmq_ctx_new();
                      ^
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c: In function ‘rf_zmq_close’:
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:580:5: warning: implicit declaration of function ‘zmq_ctx_destroy’ [-Wimplicit-function-declaration]
     zmq_ctx_destroy(handler->context);
     ^
cc1: some warnings being treated as errors
lib/src/phy/rf/CMakeFiles/srslte_rf.dir/build.make:158: die Regel für Ziel „lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_zmq_imp.c.o“ scheiterte
make[2]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_zmq_imp.c.o] Fehler 1
CMakeFiles/Makefile2:2985: die Regel für Ziel „lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all“ scheiterte
make[1]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all] Fehler 2
Makefile:160: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2

velichkov, 2019-07-16:
Hi @jonah01,

> i want to install srsLTE on an ubuntu 16.04 System, but the make process failed. I

> [ 41%] Building C object lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_zmq_imp.c.o
> /home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c: In function ‘rf_zmq_tx’:
> /home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:177:40: error: passing argument 2 of ‘zmq_recv’ from incompatible pointer type [-Werror=incompatible-pointer-types]
> n = zmq_recv(handler->transmitter, &dummy, sizeof(dummy), 0);
> ^
> In file included from /home/ubuntu/srsLTE/lib/src/phy/rf/rf_zmq_imp.c:35:0:
> /usr/include/zmq.h:230:16: note: expected ‘zmq_msg_t * {aka struct *}’ but argument is of type ‘uint8_t * {aka unsigned char *}’
> ZMQ_EXPORT int zmq_recv (void *s, zmq_msg_t *msg, int flags);
> ^

You probably have installed the older version of zeromq (`libzmq-dev`) which has incompatible C API. Installing the newer version (`libzmq3-dev` package in Ubuntu 16.04) should solve this problem

jonah01, 2019-07-17:
Hi @velichkov,

Thank you for your fast answer, that was the problem.
Now it is working :)

closed 2019-07-17 by jonah01