srsran/srsRAN_4G #355: cell_search not working

777arc, 2019-07-29:
Just an FYI, I think the huge commit earlier in the year that added TDD broke (or partially broke) cell_search, because I can't get cell_search to actually find any cells, when using master, but when I checkout release_18_12 it finds the cells fine.

I'm also curious why cell_measurent was removed.

jonah01, 2019-09-02:
Hi @777arc 

i have the same problem. 
Have you find a solution to run the latest version ? 
I am using the LimeSDR :)

777arc, 2019-09-02:
Nope, I just checked out the older version, since I didn't need TDD support anyway.  I was using a USRP B210.

andrepuschmann, 2019-09-02:
Can you guys please reopen the issue and use the issue template and fill out all the information. Thanks

777arc, 2019-09-02:
I reopened one using the template

closed 2019-09-02 by 777arc