srsran/srsRAN_4G #356: eNB keeps transmitting even if S1 backhaul is down

laf0rge, 2019-08-04:
This may be usual to people working with srsLTE, but generally the eNB I've been working with don't start transmitting a carrier beforet the  S1AP to the MME is up (i.e. sucessful S1Setup procedure), and they stop transmitting if the link is lost.  I'm not sure if there's any strict requirement about this in the 3GPP specs, but I could start digging if this is relevant to you.

Sure, this is not the most important detail of an eNB, but it at least gives a clear error behavior: If the backhaul is down, the RF carrier also goes down.

So maybe keep this issue around as a reminder about this feature request.

andrepuschmann, 2019-08-05:
Hey, we actually use this quite a lot because the UE already finds the cell then and gets a connection reject when it tries to attach but the S1 is not yet setup. So I'd say it's a feature, not a bug. Now if this is against any 3GPP spec is another question.

dchard, 2019-09-01:
Harald is correct (good to see you here btw).

The default operation is if the S1 connection is down (usually detected via the SCTP link status), the RF transmitters are shutting down immediately. The reason behind this is to allow other neighbor sites to have a chance and take over from a site where a transmission fault occurs.
Some vendors do have an option to allow to bring the cells up even when the S1 connection is offline, but this is really there for testing during commissioning, where there might be no TRM available to allow the site to go on air. None the less these are special flags and not part of the default operation.

andrepuschmann, 2019-09-02:
Cool. This makes a lot of sense. So IMHO we should go for making it optional then as well. As said before, I (we) make use of it during development so this can't entirely go away.

dchard, 2019-09-02:
Yes, I can totally understand that. A nice solution would be to create a flag or config option that this can be enabled, and by default the RF transmission should stop in case the SCTP/S1 link is down.

HungBui-cp72, 2019-12-09:
I do not believe the RAN should be working as the S1-AP indicates which PLMNid the eNodeB should be working for.  If all S1-AP connection to MMEs serving a PLMNid is not present, that PLMNid should be taken out of SIB-1.  Hence if no S1-AP connection at all, there should be no PLMNid broadcasted in SIB-1!  The RAN layer should be off then.