srsran/srsRAN_4G #358: [ENB] No support for CSFB

laf0rge, 2019-08-04:
I would argue that in every production LTE network today, CSFB (Circuit Switched Fallback) is deployed.  This relates to the fact that many early UEs don't implement VoLTE/IMS, and even those that do often only have explicit white-lists of MCCs/MNCs on which they use VoLTE.  For all other MCC/MNC tuples, CSFB to UTRAN or GERAN is used.

Most of the related work is handled by the MME and the VLR/MSC, see https://osmocom.org/projects/osmomsc/wiki/SGs_Interface for more information.

However, some small bits of this task also need to be performed by the eNB.  There are three different ways how the eNB can send the UE back to GERAN or UTRAN, and they are described in 3GPP TS 23.272 Section 6.3 under 3a/3b/3c.  
* "3a" uses inter-RAT CCO (possibly with NACC) which is rather complex. 
* "3b" uses a simple RRC connection release with rediect to GERAN or UTRAN
* "3c" is an extended version of "3b" as it adds the GSM system information for the target cells to accelerate the process (UE doesn't have to wait for receiving them)

So the simplest approach is "3b".  Supporting it in srsENB would entail not much other than:
* determining that a given RRC connection / UE is doing a CSFB (this can be seen from the InitialContextSetupReq containing the CSFB Indicator)
* including the "redirect" part when sending the RRC Connection Release

I've started to hack on this, and this issue is to keep track of this new feature.

closed 2019-09-09 by andrepuschmann

joeldelosangeles, 2021-09-22:
Hi, is CSFB already supported in srsENB if we will interface to a third-party EPC and GSM core?

joeldelosangeles, 2021-09-22:
Is the support enough and tested?  Will there be no problems with higher version. (e.g. 20.04 and 20.10)?

## 19.09
  * Add support for CSFB in srsENB

PR #379 (2019-09-09): [ENB] Add CFSB support
This commit introduces CSFB (circuit switched fall-back) capabilities to
srsLTE.  Actually, all the eNB has to do is to send a
RrcConnectionRelease with the RedirectedCarrierInfo IE.

The MME triggers this by the S1AP CS Fallback Indicator IE, which may
be present either in the Initial Context Setup Requst or in the
UE Context Modification Request.

As srsLTE has no support for the UE Context Modification Request at all
yet (!), basic support for this message is introduced in this patch.

Both Mobile Originated and Mobile Terminated CSFB with a coupel of
different UE mod [... cut]
commit c4bcd6e2 (2019-09-09): [ENB] Add CFSB support

This commit introduces CSFB (circuit switched fall-back) capabilities to
srsLTE.  Actually, all the eNB has to do is to send a
RrcConnectionRelease with the RedirectedCarrierInfo IE.

The MME triggers this by the S1AP CS Fallback Indicator IE, which may
be present either in the Initial Context Setup Requst or in the
UE Context Modification Request.

As srsLTE has no support for the UE Context Modification Request at all
yet (!), basic support for this message is introduced in this patch.

Both Mobile Originated and Mobile Terminated CSFB with a coupel of
different UE mo [... cut]