srsran/srsRAN_4G #362: Can I use srsUE to connect to a commercial network?

xietian1, 2019-08-05:
## Issue Description ##
My question is same to the previous post on https://github.com/srsLTE/srsUE/issues/11. 
Can I use srsUE installed on USRP B210 to connect to a commercial network (i.e., AT&T) wireless?

I saw someone was working on this thing as well.
http://www.softwareradiosystems.com/pipermail/srslte-users/2019-June/002695.html

According to the post in the first link, I think srsUE should work. However, is there any specific requirement to connect the srsUE to the operators' network? E.g., which antenna and which board I need to use? Is there any specific step to set the srsUE? Thanks!

antonsb, 2019-10-05:
You will need K keys and op/opc codes that can't be acquired from commercial sim card

andrepuschmann, 2019-10-08:
This has been discussed multiple times and is also covered in the docs [here](http://docs.srslte.com/en/latest/srsue/source/4_ue_advanced.html#external-usim).

closed 2019-10-08 by andrepuschmann