srsran/srsRAN_4G #372: multiple calls to std::move in lib/include/srslte/common/block_queue.h

jgiovatto, 2019-08-16:
lib/include/srslte/common/block_queue.h
```
std::pair<bool, myobj> push_(myobj&& value, bool block)
  {
    if (!enable) {
      return std::make_pair(false, std::move(value));
    }
    pthread_mutex_lock(&mutex);
    bool ret = check_queue_space_unlocked(block);
    if (ret) {
      if (mutexed_callback) {
        mutexed_callback->pushing(value);
      }
      q.push(**std::move(value))**;
      pthread_cond_signal(&cv_empty);
      ++qmetrics.num_push;
      qmetrics.currsize = q.size();
      qmetrics.highwater = std::max(qmetrics.highwater, q.size());
    } else {
      ++qmetrics.num_push_fail;
    }
    pthread_mutex_unlock(&mutex);
    return std::make_pair(ret, **std::move(value));**
  }
```

frankist, 2019-08-17:
The idea is to return back the value if the push to the queue fails so we can reuse the object for example for logging. In the case the function succeeds to push the value to the queue, it returns an invalid object (moved from), which should not be checked. It is not technical wrong code, but maybe there is a more elegant way to do this to avoid confusion. @andrepuschmann @alvasMan any ideas?

andrepuschmann, 2020-10-27:
This has been fixed in recent versions.

closed 2020-10-27 by andrepuschmann