srsran/srsRAN_4G #373: rf_blade_imp.c.456: RX failed: Operation timed out

allanjinhui, 2019-08-18:
## Issue Description ##
I run the srsLTE UE on the BladeRF, and always meet this "time out" error. 
The log information is below:
--------------------------------------------
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
set RX frequency to 2684999998
set TX frequency to 2564999998
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
.
Found Cell: Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=-4.0 KHz
Set RX sampling rate 15.36 Mhz, filter BW: 12.29 Mhz
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.456: RX failed: Operation timed out

SYNC: Receiving from radio while in IDLE_RX
/srsLTE/lib/src/phy/rf/rf_blade_imp.c.456: RX failed: Operation timed out

/srsLTE/lib/src/phy/ue/ue_sync.c.727: Error receiving samples

/srsLTE/lib/src/phy/rf/rf_blade_imp.c.456: RX failed: Operation timed out

SYNC: Receiving from radio while in IDLE_RX
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

limlynn, 2019-09-26:
I have also same issue. Have you solved it?

antonsb, 2019-10-05:
Check if bladerf is set up correctly

andrepuschmann, 2019-10-15:
We've tested with the bladeRF and it seems to work fine.

closed 2019-10-15 by andrepuschmann

lcr1017, 2020-03-31:
https://github.com/Nuand/bladeRF/issues/751

same problem

TheDuchy, 2020-04-27:
Same problem here, tested on multiple machines, different OSes and different x40 boards

andrepuschmann, 2020-04-27:
Are you guys using the srsLTE native bladeRF lib or Soapy?

lcr1017, 2020-04-27:
native bladeRF lib

TheDuchy, 2020-04-27:
native, from their git

il-vladislav, 2020-04-27:
Native bladeRF lib, x40 board, experiencing the same problem

andrepuschmann, 2020-04-27:
I was testing with SoapyRF over the last couple of days with mixed results (see [this](https://github.com/Nuand/bladeRF/issues/778) issue). I hope we can get that working before the new 20.04 release. Any chance you guys can test with Soapy as well?

TheDuchy, 2020-04-27:
@andrepuschmann Will test in the following days and update here.

yassine2018, 2020-04-28:
same pb

TheDuchy, 2020-05-11:
Sorry for the slight delay, here is the console output with Soapy. (all latest git versions, Ubuntu 18.04.4)

```
duchy@workstation:~$ sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/duchy/.config/srslte/enb.conf...

Built in Release mode using commit e3e384821 on branch master.

Opening 1 channels in RF device=default with args=default
Soapy has found device #0: backend=libusb, device=0x04:0x04, driver=bladerf, instance=0, label=BladeRF #0 [c1cd78b9..040d9897], serial=c1cd78b9f1600963934c23ad040d9897, 
Selecting Soapy device: 0
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = c1cd78b9f1600963934c23ad040d9897
[INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
Available device sensors: 
Available sensors for Rx channel 0: 
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1705] RX DC calibration table not found. Manual gain control will be used instead.
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1706] To enable AGC, see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
State of gain elements for Rx channel 0 (AGC not supported):
 - lna: 6.00 dB
 - rxvga1: 30.00 dB
 - rxvga2: 3.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - txvga1: -14.00 dB
 - txvga2: 0.00 dB
Rx antenna set to RX
Tx antenna set to TX
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1615] set_tx_gain: unable to achieve requested gain 80 (missed by 7)
[INFO] setSampleRate(Rx, 0, 11.520000 MHz), actual = 11.520000 MHz
connect(): Connection refused
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds

==== eNodeB started ===
Type <t> to view trace
[INFO] setSampleRate(Tx, 0, 11.520000 MHz), actual = 11.520000 MHz
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
0000000[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1090] Transfer timed out for buffer 0x55e23b615cf0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1090] Transfer timed out for buffer 0x55e23b679e80
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1090] Transfer timed out for buffer 0x55e23b67de90
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^CStopping ..
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
^C^C^C[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:306] wait_for_buffer: Timed out waiting for buf_ready after 400 ms
Couldn't stop after 5s. Forcing exit.
^C[1]    10146 killed     sudo srsenb
```

il-vladislav, 2020-05-11:
Hi, 
as there are many people with the same error could we re-open this ticket? Maybe more people would be able to find this issue and provide more logs?

andrepuschmann, 2020-05-11:
Hi guys,
we've reported the issue to Nuand, see [here](https://github.com/Nuand/bladeRF/issues/778). It hasn't been really solved just yet but playing around with `export BLADERF_DEFAULT_TUNING_MODE=[fpga/host]` does seem to change things.

@il-vladislav This particular issue has diverged now talking about the native bladeRF driver and the Soapy version. But feel free to open a new version and link the Nuand issue tracker. It seems to be related anyway.

Cheers
Andre

nawood10, 2021-07-15:
Also having the same operation timed out issue. Has anyone had any luck figuring this out?