srsran/srsRAN_4G #380: cell_search won't find cells anymore

777arc, 2019-09-02:
## Issue Description ##
The cell_search example/app runs, but does not find any cells.  It works fine (finds cells) when I check out release_18_12 and use all the same settings. 

## Setup Details ##
USRP B210, Ubuntu 18 Desktop, antenna meant for the band I'm looking at.

## Expected Behavior ##
With release_18_12 and prior I can run cell_search and cell_measurement and get it to find LTE cells nearby me.  

## Actual Behaviour ##
The cell_search runs but does not find any cells.  cell_measurement seems to have been removed for some reason.

## Steps to reproduce the problem ##
Check out the latest code, build it like normal, run it with a B210.  In the US, 751 MHz is a common cell that usually shows up, but your mileage may vary depending on location.

## Additional Information ##
Seems like it broke right when TDD support was added to srsLTE.

igorskh, 2019-09-03:
Hi all, 

Here are some logs from cell search run on `release_18_12-93-g54315fe` and `release_19_06-6-g8c16bca` that might be helpful.

Absolutely identical setup for both runs, just different Docker containers:
Radio frontend: USRP X310
Kernel/Ubuntu: Linux 4.15.0-58-lowlatency #64~16.04.1-Ubuntu SMP PREEMPT 

Also tested with LimeSDR USB and RPI 3, the same effect.

Command:
```
 ./cell_search -b 3 -s 1280 -e 1330 -v > logfile.log
```

Log tails:
```
tail release_18_12-93-g54315fe_cell_search.log

[INFO]: PSS: id=2, peak_pos=677, peak_value=1.094841
[INFO]: SYNC FIND: sf_idx=10, ret=0, next_state=0

Found 3 cells
Found CELL 1815.0 MHz, EARFCN=1300, PHYID=69, 100 PRB, 2 ports, PSS power=-58.0 dBm
Found CELL 1815.3 MHz, EARFCN=1303, PHYID=0, 100 PRB, 2 ports, PSS power=-59.8 dBm
Found CELL 1815.4 MHz, EARFCN=1304, PHYID=240, 15 PRB, 1 ports, PSS power=-64.0 dBm

Bye
```

```
tail release_19_06-6-g8c16bca.log

[INFO]: PSS: id=2, peak_pos=5193, peak_value=1.151303
[INFO]: SYNC FIND: sf_idx=1, ret=0, next_state=0
[INFO]: CP-CFO: estimated=-0.266817, mean=-0.265167
[INFO]: PSS: id=2, peak_pos=2537, peak_value=1.423934
[INFO]: SYNC FIND: sf_idx=1, ret=0, next_state=0

Found 0 cells

Bye
```

[release_18_12-93-g54315fe_cell_search.log](https://github.com/srsLTE/srsLTE/files/3568283/release_18_12-93-g54315fe_cell_search.log)

[release_19_06-6-g8c16bca.log](https://github.com/srsLTE/srsLTE/files/3568284/release_19_06-6-g8c16bca.log)

ayubgustavo, 2019-09-27:
In my case the application detect the PSS But it can not decode PBCH. I tried different EARFCN.

cryptik, 2019-09-30:
Same issue here when running the master baseline... no cells found when it should have found one at 839MHz.  I will try the 18_12 release and compare.

andrepuschmann, 2019-10-01:
Hi guys, thanks for flagging this issue. It will be resolved in the upcoming release which will be out shortly. In the meanwhile, feel free to apply this trivial patch manually.
```
diff --git a/lib/examples/cell_search.c b/lib/examples/cell_search.c
index 1d2a1c481..cbdf77a59 100644
--- a/lib/examples/cell_search.c
+++ b/lib/examples/cell_search.c
@@ -223,7 +223,7 @@ int main(int argc, char **argv) {
       exit(-1);
     } else if (n > 0) {
       for (int i=0;i<3;i++) {
-        if (found_cells[i].psr > 10.0) {
+        if (found_cells[i].psr > 2.0) {
           srslte_cell_t cell;
           cell.id = found_cells[i].cell_id;
           cell.cp = found_cells[i].cp;
```

closed 2019-10-01 by andrepuschmann

cryptik, 2019-10-02:
Thanks for the patch Andre.  This works like a charm.  It's actually a very simple change....I should have looked at the code first :)

andrepuschmann, 2019-10-02:
Your welcome. Thanks guys for flagging it. Tests of the examples are currently missing in our CI/CD unfortunately so we rely on reports like this.