srsran/srsRAN_4G #384: [srsenb] Not handling octet strings longer than 16383 octets

ashwinipatankar, 2019-09-17:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
I enabled the UE Capability Enquiry message in srsenb. The UE is sending UE Capability information and the srsenb is throwing the below warning message:
Not handling octet strings longer than 16383 octets
[/SOMEPATH/srslte/lib/src/asn1/asn1_utils.cc][893] Decoding failure.

## Setup Details ##
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic
RF Frontend - USRP B210
linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.0-release

## Expected Behavior ##
Decode the UE Capability message properly

## Actual Behaviour ##
Throwing error - Not handling octet strings longer than 16383 octets

## Steps to reproduce the problem ##
Enable Ue Capability Enquiry message and attach the UE.

## Additional Information ##
The hexa stream of the UE Capability information:
12:21:23.415494 [RRC ] [I] Rx SRB1 PDU
             0000: 38 01 08 22 fc 99 80 0e 1c d8 20 c0 c0 84 00 2c
             0010: 61 02 69 53 a8 1d 3b ff f9 1f f9 df ff c8 ff ce
12:21:23.415518 [RRC ] [D] Activity registered rnti=0x46
12:21:23.415591 [RRC ] [D] SRB1 - Rx ueCapabilityInformation (564 B)
             0000: 38 01 08 22 fc 99 80 0e 1c d8 20 c0 c0 84 00 2c
             0010: 61 02 69 53 a8 1d 3b ff f9 1f f9 df ff c8 ff ce
12:21:23.415601 [RRC ] [D] Content:
[
  {
    "UL-DCCH-Message": {
      "message": {
        "c1": {
          "ueCapabilityInformation": {
            "rrc-TransactionIdentifier": 0,
            "criticalExtensions": {
              "c1": {
                "ueCapabilityInformation-r8": {
                  "ue-CapabilityRAT-ContainerList": [
                    {
                      "rat-Type": "eutra",
                      "ueCapabilityRAT-Container": "c99800e1cd820c0c084002c61026953a81d3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fdff3ffafa208008c870ca74a92a3f17ec0000000701ff3ffafb00000001c27fcffebec0000000ba8900c700040036c001000c2000400318001000c3000400304001000c4000400300001000cb000400330001000d000040034c001000e500040039c001000e800040071c001011820028e10318001001c700040420800a3840c200040071c001010020028e103000010016708184000800e98002021040054c206100020038c000808410014608184000800e100020210400308001201d300040460800a9840c6000400718001011820018c000900630002203a6000808010015308180000800f38002033840039c0012124efffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fe77fff23ff3bfff91ff9dfffc8ffcefffe47fd00900003f2201200001100900001200008bc5e3f8fe2f1743c3c3c390f0f1fc5c21d129000000000000000000000000000000000000000425000080"
                    }
                  ]
                }
              }
            }
          }
        }
      }
    }
  }
]
12:21:23.415605 [RRC ] [I] UECapabilityInformation transaction ID: 0
12:21:23.415697 [RRC ] [E] Failed to unpack EUTRA capabilities message
12:21:23.415719 [RRC ] [D] SRB0 - rnti=0x46 - Tx rrcConnectionReject (3 B)
             0000: 41 20 3c

andrepuschmann, 2019-10-02:
Can you tell what UE it is and attach the PCAP?

ashwinipatankar, 2019-10-02:
UE: Samsung galaxy A30S

I will upload the pcap soon. Its the UE Capablity Information message which is huge.

ashwinipatankar, 2019-10-30:
Thanks for the patience.  Please find the pcap attached. The zip also contains the enb.log

Thanks
[enboctetissue.zip](https://github.com/srsLTE/srsLTE/files/3790501/enboctetissue.zip)

closed 2023-04-25 by ismagom