srsran/srsRAN_4G #386: make failed on srslte: liblte_security.cc:(.text+0x1): undefined reference to `sha256_hmac'

Monikaprak, 2019-10-03:
I got the following error when I tried to do make in srslte/build/. 

Linking CXX executable test_eea1
../../src/common/libsrslte_common.a(liblte_security.cc.o): In function `sha256(unsigned char const*, unsigned long, unsigned char const*, unsigned long, unsigned char*, int)':
liblte_security.cc:(.text+0x1): undefined reference to `sha256_hmac'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/test/common/test_eea1] Error 1
make[1]: *** [lib/test/common/CMakeFiles/test_eea1.dir/all] Error 2

My system is fedora 20. I installed all the required packages using the following command
dnf install cmake fftw3-devel polarssl-devel lksctp-tools-devel libconfig-devel boost-devel

velichkov, 2019-10-03:
Hi @Monikaprak, 

> My system is fedora 20. 

Fedora 20 is quite old and is [end of life since June 2015](https://lists.fedoraproject.org/pipermail/announce/2015-June/003279.html). 

> I installed all the required packages using the following command
> dnf install cmake fftw3-devel polarssl-devel lksctp-tools-devel libconfig-devel boost-devel

> I got the following error when I tried to do make in srslte/build/.
>
> Linking CXX executable test_eea1
> ../../src/common/libsrslte_common.a(liblte_security.cc.o): In function `sha256(unsigned char [...]

The `polarssl` version in Fedora 20 does not suport `sha256`

```
$ rpm -q polarssl-devel
polarssl-devel-1.2.12-3.fc20.x86_64

$ rpm -ql polarssl-devel | grep include | grep sha
/usr/include/polarssl/sha1.h
/usr/include/polarssl/sha2.h
/usr/include/polarssl/sha4.h
```
The `mbedtls` version is also old and incompatible.
```
$ rpm -q mbedtls-devel
mbedtls-devel-1.3.11-1.fc20.x86_64
$ rpm -ql mbedtls-devel | grep include | grep sha
/usr/include/polarssl/sha1.h
/usr/include/polarssl/sha256.h
/usr/include/polarssl/sha512.h
```
So my advice is to update your system to Fedora 30 or Ubuntu 18.04 LTS.

andrepuschmann, 2019-10-07:
Closing this as suggestions have been given.

closed 2019-10-07 by andrepuschmann