srsran/srsRAN_4G #390: srsepc not working

noobmaster777, 2019-10-10:
Hello I am a noob trying to learn srslte. Any help would be greatly appreciated.

## Issue Description ##
Whenever I start srsepc. I get this error.
“Could not initialize SGi interface”
“Could not initialize the SPGW’s GTP-U”
“Error initializing SP-GW”

## Setup Details ##
Gentoo linux 4.19.72 
BladeRF

## Additional Information ##
I already added GTP through kernel menuconfig but still getting same error.

alvasMan, 2019-10-10:
Hi, did you run srsepc with sudo? srsEPC creates a TUN interface and requires sudo for that.
You can see that something went wrong in the initialization of the TUN in the "Could not initialize SGi interface" message. You can check what the function that initializes the SGi in "srsepc/src/spgw/gtpu.cc" 

On another note, srsEPC uses a user-plane implementation of the GTP-U, so you should not have to enable GTP in the kernel menuconfig (no harm in doing it though).

Finally, this does not seem like an issue, please use the mailing list for support:
http://www.softwareradiosystems.com/mailman/listinfo/srslte-users

noobmaster777, 2019-10-10:
Thanks so much.

suttonpd, 2019-10-10:
Hi, please refer to docs.srslte.com and readme files in the code for guidance on getting up and running with srsLTE. If you still experience problems, you can ask for help through the srslte-users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users). Please use Github issues for reporting verified bugs/issues only.

closed 2019-10-10 by suttonpd

noobmaster777, 2019-10-11:
To those who encounter the same problem, I solved it through enabling TUN/TAP support in the kernel.