srsran/srsRAN_4G #392: build error on latest (6d62e15d2ae1969e01371dba4ea918966cb0e3d4)

jgiovatto, 2019-10-14:
Hi folks, 
I ran into this error on Fedora 29:

srsLTE-master/lib/src/phy/rf/rf_zmq_imp_tx.c:40:5: error: ‘strncpy’ specified bound 16 equals destination size [-Werror=stringop-truncation]

srsLTE-master/lib/src/phy/rf/rf_zmq_imp_rx.c:104:5: error: ‘strncpy’ specified bound 16 equals destination size [-Werror=stringop-truncation]

Upon further investigation Im not quite understanding the order here, I think the order should be swapped and a safer approach using strncpy be used.
```
-    // Copy id
-    strncpy(q->id, id, 16);
-
     // Zero object
     bzero(q, sizeof(rf_zmq_tx_t));
 
+    // Copy id
+    strncpy(q->id, id, sizeof(q->id) - 1);
+
```

andrepuschmann, 2019-10-15:
You are too quick ;-) Pedro also detected this after we've pushed. Just updated `master`. Thanks anyways

closed 2019-10-15 by andrepuschmann

jgiovatto, 2019-10-15:
might be a bug there,

looking at lib/src/phy/rf/rf_zmq_imp_rx.c

int rf_zmq_rx_open(rf_zmq_rx_t* q, char* id, void* zmq_ctx, char* sock_args)
{
  int ret = SRSLTE_ERROR;

  if (q) {
    // Zero object
    bzero(q, sizeof(rf_zmq_tx_t));

type mismatch between tx_t and rx_t

Joe

andrepuschmann, 2019-10-16:
Again, much appreciated Joe. It's fixed now in `master`.