srsran/srsRAN_4G #393: ue logging info level in 19.X

jgiovatto, 2019-10-18:
Hi folks
I've been seeing info level logs in the ue log file even though I've set the levels to warning and/or error as in previous versions. I see some changes have been made w/r to setting log levels in ue.cc over the 18.X series. The enb seems ok, could be due to a subsequent call to 
```
 pool_log.init("POOL", logger);
 pool_log.set_level(srslte::LOG_LEVEL_ERROR);
 pool->set_log(&pool_log);
```
that  ue.cc does not have.

As a temporary measure to reduce overhead I've made this change to ue.cc which seems to work but I doubt is the final solution.
```
diff --git a/srsue/src/ue.cc b/srsue/src/ue.cc
index 7fd84ca..6d4d113 100644
--- a/srsue/src/ue.cc
+++ b/srsue/src/ue.cc
@@ -136,6 +136,8 @@ int ue::init(const all_args_t& args_, srslte::logger* logger_)
   phy->wait_initialize();
   log.console("done!\n");
 
+  log.set_level(srslte::LOG_LEVEL_WARNING);
+
   return SRSLTE_SUCCESS;
 }
```

andrepuschmann, 2019-10-18:
Thanks Joe for the report. We'll investigate ASAP.

andrepuschmann, 2020-10-27:
We've updated the entire logging architecture in 20.10. Please check if the issue still exists for you. If so, please feel free to reopen.

closed 2020-10-27 by andrepuschmann