srsran/srsRAN_4G #395: Could srsLTE work with USRP N310??

zmcrazy, 2019-10-21:
## Issue Description ##
I found that in ReadMe, srsLTE was tested with USRP B210 and USRP X310, but N310 is not tested yet, right??
How to install it in USRP N310? N310 support 2 mode: network mode and embedded mode, which mode can be used with srsLTE ?

zmcrazy, 2019-10-24:
I tried srsUE on N310, but it failed, what can I do?? By the way, the time_adv_nsamples value is not set in srsLTE code to adapt N310, see  radio::set_tx_srate function in file Radio.cc, so I choose 45 for this value, because it is used for X310:
INFO] [MPM.PeriphManager] init() called with device args `mgmt_addr=172.23.2.116,product=n310,master_clock_rate=122.88e6'.
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1343 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1346 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1342 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000002)
Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=1400, f_dl=1835.5 MHz, f_ul=1740.5 MHz
srslte_rf_name(&rf_device): uhd_n300Setting manual TX/RX offset to 45 samples
.SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
.SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX
SYNC:  Receiving from radio while in IDLE_RX

[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1332 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000002)
Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=1400, f_dl=1835.5 MHz, f_ul=1740.5 MHz
srslte_rf_name(&rf_device): uhd_n300Setting manual TX/RX offset to 45 samples
.
Found Cell:  Mode=FDD, PCI=201, PRB=25, Ports=1, CFO=1.0 KHz
srslte_rf_name(&rf_device): uhd_n300Setting manual TX/RX offset to 45 samples
Found PLMN:  Id=48020, TAC=3
Random Access Transmission: seq=20, ra-rnti=0x2
Random Access Transmission: seq=3, ra-rnti=0x2
Random Access Transmission: seq=4, ra-rnti=0x2
Random Access Transmission: seq=7, ra-rnti=0x2
Random Access Transmission: seq=24, ra-rnti=0x2
Random Access Transmission: seq=8, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=16, ra-rnti=0x2
Random Access Transmission: seq=6, ra-rnti=0x2
Random Access Transmission: seq=10, ra-rnti=0x2
Warning: Detected Radio-Link Failure

jonakarl, 2019-10-24:
We also have a n310 but we don't even get this far but we might have a different problem as we get: 
`Error opening UHD: code 11
/build/srslte-EIQesT/srslte-19.06/lib/src/phy/rf/rf_imp.c.126: No compatible RF frontend found`

We are using the deb version (19.06-0ubuntu1~srslte1~18.04) and the UHD version used seams to differ from the one installed. 
```
 uhd_find_devices 
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.1-release
<snip>
srsue
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown
```

zmcrazy, 2019-10-25:
@jonakarl UHD version need the same, I think.
Do you know how to upgrade the UHD version of N 310??

andrepuschmann, 2019-10-25:
Are you guys building from source or using the binaries? If you update UHD (or use a more recent than the one the binary was built with) you have to build srsLTE from source, too.

zmcrazy, 2019-10-25:
@andrepuschmann I use UHD FPGA images.

jonakarl, 2019-11-01:
@zmcrazy I used the official procedure to upgrade the filesystem and FPGA of the n310 https://files.ettus.com/manual/page_usrp_n3xx.html (we use the XG image for dual sfp+)

@andrepuschmann I tried to build form source and while the compilation worked I get this error when trying to use srsue: 
` srsue
srsue: symbol lookup error: srsue: undefined symbol: srslte_rf_set_master_clock_rate`

zmcrazy, 2019-11-05:
@jonakarl Thank you very much.
I use the official procedure to upgrade the filesystem.
there is a problem, do you get this error??

---
 mender –commit
INFO[0000] Configuration file does not exist: /var/lib/mender/mender.conf  module=config
INFO[0000] Loaded configuration file: /etc/mender/mender.conf  module=config
INFO[0000] Mender running on partition: /dev/mmcblk0p3   module=main
ERRO[0000] Must give one of -rootfs, -commit, -bootstrap or -daemon arguments  module=main

zmcrazy, 2019-11-05:
root@ni-n3xx-31A364B:/data/mender# cat /etc/mender/artifact_info
artifact_name=v3.14.1.1_n3xx

jonakarl, 2019-11-15:
@zmcrazy 
We did not use mender but used a sd card, see "Updating the file system"

zmcrazy, 2019-11-20:
@jonakarl 
Do you have any progress??
I still hava following error messages, and can not connect srsUE to the network, and I have no idea...
Found Cell:  Mode=FDD, PCI=201, PRB=25, Ports=1, CFO=-0.1 KHz
In void radio::set_tx_srate(double srate) srslte_rf_name(&rf_device): uhd_n300Setting manual TX/RX offset to 0 samples
Found PLMN:  Id=48020, TAC=8
Random Access Transmission: seq=17, ra-rnti=0x2
Random Access Transmission: seq=4, ra-rnti=0x2
Random Access Transmission: seq=9, ra-rnti=0x2
Random Access Transmission: seq=6, ra-rnti=0x2
Random Access Transmission: seq=27, ra-rnti=0x2
Random Access Transmission: seq=12, ra-rnti=0x2
Random Access Transmission: seq=8, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=26, ra-rnti=0x2
Random Access Transmission: seq=17, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Random Access Transmission: seq=9, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2

maxpenner, 2019-11-29:
I have tested srsLTE with two USRPs N310 and it ran very stable for multiple hours.

i7-4770 srsENB
i5-2500 srsUE
Ubunu 16.04.6 LTS with 4.4.0-169-lowlatency kernel
UHD-3.15.LTS
N310/N321 with 10 GbE interface (HG or XG image)
SISO
Everything build from source

I have also connected an N321 and it works well too, if you switch the master clock to 245.76 MS/s in srsLTE/lib/src/phy/rf/rf_uhd_imp.c, as the N321 does not support 122.88 MS/s.

jonakarl, 2019-11-29:
@maxpenner 
Can you share a example of commandline options or config files used to get this running?

maxpenner, 2019-11-29:
Here's a minimal functional configuration I've tested:

I connect both USRPs N310 with 30cm coax cables, a DC blocker and 30dB attenuation. As for the configuration files, I only change ue.conf and enb.conf (attached as .txt). The other files are left in their original state. The value time_adv_nsamples=50 at the srsUE works but is not optimal, the terminal shows ta_us=5.2 (output attached). I trigger one 10kB ping per second from the srsUE to the srsENB.

With this setup, I can easily stream a 1080p Yotube video in the downlink.

[enb.conf.txt](https://github.com/srsLTE/srsLTE/files/3905449/enb.conf.txt)
[ue.conf.txt](https://github.com/srsLTE/srsLTE/files/3905452/ue.conf.txt)
[terminal_output_srsENB.txt](https://github.com/srsLTE/srsLTE/files/3905450/terminal_output_srsENB.txt)
[terminal_output_srsUE.txt](https://github.com/srsLTE/srsLTE/files/3905451/terminal_output_srsUE.txt)

zmcrazy, 2019-11-30:
@maxpenner 
Hi, which version of srsLTE are you use??  I still got following error msgs:

Reading configuration file n310_band3_2.0.conf...

Built in Release mode using 19.9.0.

Opening 1 RF devices with 1 RF channels...
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.0-0-unknown
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP with args: "type=n3xx",master_clock_rate=122.88e6
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=172.23.2.116,type=n3xx,product=n310,serial=31A364B,claimed=False,addr=192.168.10.2,"type=n3xx",master_clock_rate=122.88e6
[INFO] [MPM.main] Launching USRP/MPM, version: 3.15.0.0-gdbaf4132
[INFO] [MPM.main] Spawning RPC process...
[INFO] [MPM.PeriphManager] Device serial number: 31A364B
[INFO] [MPM.PeriphManager] Initialized 2 daughterboard(s).
[INFO] [MPM.PeriphManager] init() called with device args `time_source=internal,clock_source=internal'.
[INFO] [MPM.RPCServer] RPC server ready!
[INFO] [MPM.RPCServer] Spawning watchdog task...
[INFO] [MPM.PeriphManager] init() called with device args `product=n310,mgmt_addr=172.23.2.116,"type=n3xx",clock_source=internal,master_clock_rate=122.88e6,time_source=internal'.
[INFO] [0/Replay_0] Initializing block control (NOC ID: 0x4E91A00000000004)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/FIFO_0] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_1] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_2] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_3] Initializing block control (NOC ID: 0xF1F0000000000000)
Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=1400, f_dl=1835.5 MHz, f_ul=1740.5 MHz
Setting manual TX/RX offset to 50 samples
.
Found Cell:  Mode=FDD, PCI=201, PRB=50, Ports=1, CFO=0.2 KHz
Setting manual TX/RX offset to 50 samples
Found PLMN:  Id=48020, TAC=8
Random Access Transmission: seq=24, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=8, ra-rnti=0x2
Random Access Transmission: seq=8, ra-rnti=0x2
Random Access Transmission: seq=27, ra-rnti=0x2
Random Access Transmission: seq=11, ra-rnti=0x2
Random Access Transmission: seq=14, ra-rnti=0x2
Random Access Transmission: seq=1, ra-rnti=0x2
Random Access Transmission: seq=8, ra-rnti=0x2
Random Access Transmission: seq=7, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Setting manual TX/RX offset to 50 samples

maxpenner, 2019-11-30:
I'm using the latest version 19.09 build from source.

I don't know what the cause of the error message is, but I would suggest that you try the lowlatency kernel. In fact, if you are using Ubuntu 16.04.x LTS or 18.04.x LTS, you can follow the [OAI instructions](https://gitlab.eurecom.fr/oai/openairinterface5g/wikis/OpenAirKernelMainSetup#ubuntu-1604-ltslinux-kernel-version-48-or-higher) up to and including "Disable CPU Frequency scaling". Maybe it's a real-time issue.

zmcrazy, 2019-12-03:
@maxpenner 
Thank you very much.
It is solved.
I changed the 1GB connection from HOST to N310 to 10GB speed, and it is become normal.

zmcrazy, 2019-12-03:
@maxpenner 
By the way, I find that time_adv_nsamples this value , I set it to Zero, it seems normal too.
So, I think it is no need to set it.

andrepuschmann, 2020-10-27:
Issue seems to be fixed for all users. Closing.

closed 2020-10-27 by andrepuschmann