srsran/srsRAN_4G #397: Error opening UHD: code 11 "No compatible RF frontend found".

asp2k5, 2019-10-23:
hi srsLTE team,
I'm having issues getting srsENB or srsUE to detect my SDR hardware
(BladeRF x40),
In short, when starting srsENB or srsUE it looks for a USRP and then says
"No compatible RF frontend found".

>
>Built in Release mode using commit  on branch .
>
>---  Software Radio Systems LTE eNodeB  ---
>
>Reading configuration file /root/.srs/enb.conf...
>**Opening USRP with args:
>Error opening UHD: code 11
>No compatible RF frontend found
>Error opening RF device
>Failed to find device auto with args auto**

jonakarl, 2019-10-24:
We have the same problem but with usrp n310. 

What we discovered is that the uhd versions differ between `srsue` and `udh_find_device`  
`UHD_3.14.1.1-release` vs `UHD_003.010.003.000-0-unknown `

btw: uhd_find_devices works and finds the device.

andrepuschmann, 2019-10-25:
@asp2k5's issue is that you are trying to run a bladeRF using the UHD, which doesn't work. Install the bladerf libs. @jonakarl you may need to install srsLTE from source as you've updated UHD and the original deb package has been built with UHD 3.10.3

asp2k5, 2019-10-25:
thank you for reply but i have already intstalled all the reqd libs of bladeRF. error is not UHD based but the "No compatible RF frontend found" after UHD it is trying to look for BladeRF but its failing to recognise.

velichkov, 2019-10-25:
Hi @asp2k5,

> i have already intstalled all the reqd libs of bladeRF. 

How did you installed them, which versions and before or after building srsLTE?
What Linux distribution do you use and which version?

> after UHD it is trying to look for BladeRF but its failing to recognise.

How do you know this? It's not evident from the logs you've provided as there is no "Opening bladeRF..." message so it seems you have compiled without bladeRF support.

Go do your build directory, open CMakeCache.txt and make sure `BLADERF_FOUND` is set to `TRUE`. If it is set to `FALSE` first remove the CMakeCache.txt and then build as usual - `cmake ..`, ...

JayakumarCK2, 2021-02-12:
[image]

Hi, I am getting this error. What does this mean?

asp2k5, 2021-02-17:
FGPA is not compatible. Burn the required FGPA into your SDR, it will work.

JayakumarCK2, 2021-02-17:
Can you please elaborate, what are FPGA images, and what exactly they do? Any link for the relevant information would be helpful.

Jayck-1096, 2021-03-16:
[image]

I have installed UHD libraries, still getting this error. Any leads ?

Chi-Hsin-Wu, 2022-06-22:
[image]
I have installed UHD libraries，but it says that it's available，and uhd_init failed

RogerLin88124, 2022-12-26:
@Chi-Hsin-Wu   hello,I have the same problem as you, is there a solution?

stuitee, 2023-11-19:
@Chi-Hsin-Wu hello,I have the same problem as you, is there a solution?

andyraf, 2024-09-27:
I have the same problem as well. I've even reinstalled Ubuntu 22.04 from scratch and rerun through all the dependencies and builds according to instructions with no success.