srsran/srsRAN_4G #401: [srsenb] crash when exit

v1c0r, 2019-10-30:
## Issue Description ##
srsenb crashes at exit when active UE connections exists

## Setup Details ##
Ubuntu 18.04.3 LTS; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.1-release

## Expected Behavior ##
Correct program exit

## Actual Behaviour ##
Console output:
Built in Release mode using commit 62d29567 on branch HEAD.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/sdr/config/enb.conf...
Opening 1 RF devices with 1 RF channels...
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.1-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP with args: type=b200,master_clock_rate=23.04e6
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO , Firmware Rev 0.929a
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting frequency: DL=2665.0 Mhz, UL=2545.0 MHz
Setting Sampling frequency 11.52 MHz

==== eNodeB started ===
Type <t> to view trace
RACH:  tti=1411, preamble=31, offset=1, temp_crnti=0x46
SCHED: Could not transmit RAR within the window (RA TTI=1411, Window=10, Now=1425)
RACH:  tti=1431, preamble=21, offset=1, temp_crnti=0x47
RACH:  tti=1451, preamble=44, offset=0, temp_crnti=0x48
SCHED: Could not transmit RAR within the window (RA TTI=1451, Window=10, Now=1465)
RACH:  tti=1471, preamble=39, offset=1, temp_crnti=0x49
RACH:  tti=1491, preamble=6, offset=1, temp_crnti=0x4a
Disconnecting rnti=0x46.
RACH:  tti=1511, preamble=16, offset=1, temp_crnti=0x4b
Disconnecting rnti=0x47.
Disconnecting rnti=0x48.
Disconnecting rnti=0x49.
Disconnecting rnti=0x4a.
RACH:  tti=1601, preamble=36, offset=1, temp_crnti=0x4c
RACH:  tti=1691, preamble=6, offset=1, temp_crnti=0x4d
RACH:  tti=1791, preamble=11, offset=1, temp_crnti=0x4e
^CStopping srsENB... Press Ctrl+C 9 more times to force stop
Disconnecting rnti=0x4b.
Disconnecting rnti=0x4c.
double free or corruption (out)
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---

Contents of srsLTE.backtrace.crash:
--- command='./srsenb /home/sdr/config/enb.conf' version=19.9.0 signal=6 date='29/10/2019 20:04:37' ---
	./srsenb(+0x254a78) [0x5628ce9d4a78]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3ef20) [0x7f74d3deaf20]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7f74d3deae97]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7f74d3dec801]
	/lib/x86_64-linux-gnu/libc.so.6(+0x89897) [0x7f74d3e35897]
	/lib/x86_64-linux-gnu/libc.so.6(+0x9090a) [0x7f74d3e3c90a]
	/lib/x86_64-linux-gnu/libc.so.6(cfree+0x525) [0x7f74d3e43e75]
	./srsenb(_ZN6srsenb9sf_worker8rem_rntiEt+0xe6) [0x5628ce980716]
	./srsenb(_ZThn8_N6srsenb3phy8rem_rntiEt+0x43) [0x5628ce973eb3]
	./srsenb(_ZN6srsenb3mac6ue_remEt+0xa7) [0x5628ce9a34f7]
	./srsenb(_ZN6srsenb3rrc8rem_userEt+0xbe) [0x5628ce9c250e]
	./srsenb(_ZN6srsenb3rrc10run_threadEv+0x49b) [0x5628ce9c896b]
	./srsenb(_ZN6thread21thread_function_entryEPv+0x14d) [0x5628ce93fd4d]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f74d5b1c6db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f74d3ecd88f]

## Steps to reproduce the problem ##
Press Ctrl+C in srsenb console when active UE RRC connection exist

## Additional Information ##
As I can see, rrc layer calls rem_user() when phy layer with vector of sf_worker already stopped. Calling sf_worker::rem_rnti() after sf_worker::stop() produces memory corruption in srslte_enb_dl_rem_rnti().

A quick fix for that is to clear ue_db map after deleting its nodes in sf_worker::stop(), so call of sf_worker::rem_rnti() will do nothing but error reporting, because such rnti does not exist anymore.

asp2k5, 2019-12-28:
Can you pls tell what earcn value and which corresponding mobile/ue are you using?

andrepuschmann, 2020-04-29:
Should be fixed in the upcoming 20.04 release.

closed 2020-04-29 by andrepuschmann