srsran/srsRAN_4G #402: srsenb outputs "killed"

valerietyj, 2019-11-07:
Running srsenb returns killed. 
Currently have a LimeSDR Mini and srsLTE is set up on VMWare Fusion, running Ubuntu 18.04 64-bit OS. 

Output: 
Built in Release mode using commit c2de0c67 on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/yj/.config/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.15.0.0-96-g0408f47b
[INFO] [LOGGING] Fastpath logging disabled at runtime.
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Opening USRP with args: 
[INFO] [UHDSoapyDevice] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00492A2A'
[INFO] [UHDSoapyDevice] Reference clock 30.72 MHz
[INFO] [UHDSoapyDevice] Device name: LimeSDR-USB
[INFO] [UHDSoapyDevice] Reference: 30.72 MHz
[INFO] [UHDSoapyDevice] LMS7002M calibration values caching Disable
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
Killed

andrepuschmann, 2019-11-08:
you're using the Soapy wrapper for UHD. Have you tried to use the SoapySDR directly?

valerietyj, 2019-11-08:
Hello, thank you for your quick reply! I’m not really sure how I can do that, will you be able to guide me?

andrepuschmann, 2019-11-08:
deinstall soapy-uhd

andrepuschmann, 2020-04-29:
Closing issue due to lack of feedback

closed 2020-04-29 by andrepuschmann