srsran/srsRAN_4G #403: No signal found with bladerf

Thanosss, 2019-11-12:
Bladerf has established a connection with PC. after enabling srsenb, enter view trace to output the following contents, but it is unable to find signals through mobile devices
The operation log is as follows:
```
sudo srsenb 

Built in Release mode using commit c2de0c6 on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/thanos/.config/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
set TX frequency to 2684999936
set RX frequency to 2564999936
Setting Sampling frequency 11.52 MHz

==== eNodeB started ===
Type <t> to view trace
S1 Setup Failure. Cause: misc - unknown-PLMN
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz
t
Enter t to stop trace.
RF status: O=982, U=0, L=0
RF status: O=978, U=0, L=0
RF status: O=984, U=0, L=0
```

suttonpd, 2019-11-12:
The eNodeB was unable to establish an S1 connection with the EPC ("S1 Setup Failure. Cause: misc - unknown-PLMN").

closed 2019-11-12 by suttonpd

Thanosss, 2019-11-13:
Hello, what is the cause of this problem?(Sending S1 Setup Failure - Unknown PLMN)

alvasMan, 2019-11-13:
Can you double check if the MCC and the MNC match on the eNB and the EPC?

kschepps, 2023-12-20:
I'm experiencing the same issue. I have checked the MCC and MNC values, and everything seems to be matching as best as I can tell. I have even tried wiping and reloading my OS for clean installs, but I'm still unable to get srsenb and open5gs to talk to each other

enb.conf:
`#####################################################################
#                   srsENB configuration file
#####################################################################

#####################################################################
# eNB configuration
#
# enb_id:               20-bit eNB identifier.
# mcc:                  Mobile Country Code
# mnc:                  Mobile Network Code
# mme_addr:             IP address of MME for S1 connnection
# gtp_bind_addr:        Local IP address to bind for GTP connection
# gtp_advertise_addr:   IP address of eNB to advertise for DL GTP-U Traffic
# s1c_bind_addr:        Local IP address to bind for S1AP connection
# s1c_bind_port:        Source port for S1AP connection (0 means any)
# n_prb:                Number of Physical Resource Blocks (6,15,25,50,75,100)
# tm:                   Transmission mode 1-4 (TM1 default)
# nof_ports:            Number of Tx ports (1 port default, set to 2 for TM2/3/4)
#
#####################################################################
[enb]
enb_id = 0x19B
mcc = 001
mnc = 01
#mme_addr = 127.0.1.100
mme_addr = 127.0.0.2
gtp_bind_addr = 127.0.1.1
s1c_bind_addr = 127.0.1.1
s1c_bind_port = 0
n_prb = 50
#tm = 4
#nof_ports = 2

#####################################################################
# eNB configuration files
#
# sib_config:  SIB1, SIB2 and SIB3 configuration file 
# note: When enabling MBMS, use the sib.conf.mbsfn configuration file which includes SIB13
# rr_config:   Radio Resources configuration file 
# rb_config:   SRB/DRB configuration file 
#####################################################################
[enb_files]
sib_config = sib.conf
rr_config  = rr.conf
rb_config = rb.conf

#####################################################################
# RF configuration
#
# dl_earfcn: EARFCN code for DL (only valid if a single cell is configured in rr.conf)
# tx_gain: Transmit gain (dB).
# rx_gain: Optional receive gain (dB). If disabled, AGC if enabled
#
# Optional parameters:
# dl_freq:            Override DL frequency corresponding to dl_earfcn
# ul_freq:            Override UL frequency corresponding to dl_earfcn (must be set if dl_freq is set)
# device_name:        Device driver family
#                     Supported options: "auto" (uses first driver found), "UHD", "bladeRF", "soapy", "zmq" or "Sidekiq"
# device_args:        Arguments for the device driver. Options are "auto" or any string.
#                     Default for UHD: "recv_frame_size=9232,send_frame_size=9232"
#                     Default for bladeRF: ""
# time_adv_nsamples:  Transmission time advance (in number of samples) to compensate for RF delay
#                     from antenna to timestamp insertion.
#                     Default "auto". B210 USRP: 100 samples, bladeRF: 27
#####################################################################
[rf]
dl_earfcn = 2100
[... 585 lines cut ...]
#    - plmn_id:
#        mcc: 008
#        mnc: 08
#      tac: 60
#    - plmn_id:
#        mcc: 009
#        mnc: 09
#      tac: [70, 80]
#
#  o Access Control
#  access_control:
#    - default_reject_cause: 13
#    - plmn_id:
#        reject_cause: 15
#        mcc: 001
#        mnc: 01
#    - plmn_id:
#        mcc: 002
#        mnc: 02
#    - plmn_id:
#        mcc: 999
#        mnc: 70
#
#  o Relative Capacity
#  relative_capacity: 100`

kschepps, 2023-12-20:
I'm also seeing Sending S1 Setup Failure - Unknown PLMN. I have wiped this PC numerous times and followed various tutorials, but this is as far as I'm able to get.

Edit: Sorry about the formatting. I can't figure out how to get the code boxes to show properly

srsenb output:
`Setting cpu: 0
Setting cpu: 1
Setting cpu: 2
Setting cpu: 3
Setting cpu: 4
Setting cpu: 5
Setting cpu: 6
Setting cpu: 7
Setting cpu: 8
Setting cpu: 9
Setting cpu: 10
Setting cpu: 11
Setting cpu: 12
Setting cpu: 13
Setting cpu: 14
Setting cpu: 15
-e BladeRF Successfully Detected!
FPGA already loaded
Active RF plugins: libsrsran_rf_blade.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Couldn't open /root/.config/srsran/enb.conf either, trying /etc/srsran/enb.conf
Reading configuration file /etc/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open /root/.config/srsran/sib.conf either, trying /etc/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open /root/.config/srsran/rr.conf either, trying /etc/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
Couldn't open /root/.config/srsran/rb.conf either, trying /etc/srsran/rb.conf
srsLog error - Unable to create log file "/tmp/enb.log": Permission denied

Built in Release mode using commit eea87b1d8 on branch master.

/files/LTE_Tower/srsRAN/srsenb/src/enb_cfg_parser.cc:1881: Force DL EARFCN for cell PCI=1 to 2100
Opening 1 channels in RF device=bladeRF with args=default
Supported RF device list: bladeRF file
Opening bladeRF...
S1setup failed
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
S1 Setup Failure. Cause: misc - unknown-PLMN
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

==== eNodeB started ===
Type <t> to view trace
Starting plot for worker_id=0
Creating plot window 'srsENB PCI 1'...
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting frequency: DL=2125.0 Mhz, UL=1725.0 MHz for cc_idx=0 nof_prb=50
set TX frequency to 2125000000
set RX frequency to 1724999998
S1setup failed
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
S1 Setup Failure. Cause: misc - unknown-PLMN
`

----------------------------------------------------------------------------------------------------------------------------------------

enb.conf:
`#####################################################################
#                   srsENB configuration file
#####################################################################

[... 650 lines cut ...]
#    - plmn_id:
#        mcc: 008
#        mnc: 08
#      tac: 60
#    - plmn_id:
#        mcc: 009
#        mnc: 09
#      tac: [70, 80]
#
#  o Access Control
#  access_control:
#    - default_reject_cause: 13
#    - plmn_id:
#        reject_cause: 15
#        mcc: 001
#        mnc: 01
#    - plmn_id:
#        mcc: 002
#        mnc: 02
#    - plmn_id:
#        mcc: 999
#        mnc: 70
#
#  o Relative Capacity
#  relative_capacity: 100`