srsran/srsRAN_4G #408: UE attach failure after attached to network

bwchacha, 2019-11-14:
## Issue Description ##
UE attach failure after attached to network 

## Setup Details ##
I’m using srsepc as core element to connect a vendor ENB, Huawei CPE as UE. 
It went well after srsepc started then S1AP setup with ENB. UE could attach to network, see log record:
14:53:43.341384 [S1AP] [I] Received Initial UE message -- Attach Request
..............
14:53:43.713877 [S1AP] [I] UL NAS: Received Attach Complete

But after a short while (about 20 minutes), UE is released, not sure either UE or EPC started release procedure. UE should kept EMM-registered after attached to network, right? See log records:
15:14:19.937758 [S1AP] [I] Received UE Context Release Request Message.
..............
15:14:27.280772 [NAS ] [E] UE is not EMM-Registered.

After that, you can see UE tried to re-attach to network time and time again, yet it failed each time.
Unless restarting the srsepc, UE would not attach successfully.

Some wired log records were observed:
15:16:52.882385 [HSS ] [I] User not found. IMSI: 000000000000000
15:16:52.882399 [HSS ] [E] User not found at HSS. IMSI: 000000000000000
15:16:52.882414 [NAS ] [I] User not found. IMSI 000000000000000 
..........
15:26:12.172341 [NAS ] [I] Attach Request -- Found previously attached UE.
15:26:12.172360 [MME GTPC] [I] Sending GTP-C Delete Session Request request. IMSI 0
15:26:12.172372 [MME GTPC] [E] Could not find GTP-C context to remove
..........
15:26:12.173542 [S1AP] [E] Unhandled S1AP intiating message: ErrorIndication
..........

There is no IMSI 0000000000000 in usr_db.csv, I have no idea where it came from.

## Steps to reproduce the problem ##
This could reproduce every time and same problem happened. UE failed to attach to network after a short while it attached. 

## Additional Information ##
Pls see the log files for more details.
[UE attach failure.zip](https://github.com/srsLTE/srsLTE/files/3846801/UE.attach.failure.zip)

alvasMan, 2019-11-14:
Seems like the UE keeps sending PDN connectivity requests because it's expecting an IPv6 connections as well.

Could you disable  IPv6 in the UE's APN to see if that helps? You just need to set the PDN type to IPv4 instead of IPv4v6.

If does not help, please send the logs and pcaps again.

bwchacha, 2019-11-20:
> Seems like the UE keeps sending PDN connectivity requests because it's expecting an IPv6 connections as well.
> 
> Could you disable IPv6 in the UE's APN to see if that helps? You just need to set the PDN type to IPv4 instead of IPv4v6.
> 
> If that does not help, please send the logs and pcaps again.

Hi, Pedro,

You are right. The Huawei CPE can't switch off ipv6, so I use another CPE which can turn off ipv6 instead. There is no such problem after turning off ipv6.
Thanks for your advice. But I still can't figure out why there is IMSI 0000000000000 in log messages when the problem happened.

closed 2019-11-20 by bwchacha

jin-github1002, 2020-04-16:
@alvasMan I guess i have the same issue with @bwchacha . The difference is that i use srsue as the UE instead of a mobile phone with a USIM card. Could you give more specification about how to disable IPv6 in the UE's side. I use srsue as the ue side in a computer. I do not use USIM card. I think srsue use the user number(such as IMSI...)  in the ue.conf.
Thanks so much.