srsran/srsRAN_4G #411: multiple cells within the eNB

slimaneAI, 2019-11-16:
Hi guys,

I would like to know if the srsLTE support multiple cells within the eNB, i would like to install a configuration with two BladeRF (one for each cell) in order to test the intra-eNB Handover,

Thanks,

frankist, 2019-11-17:
Hey slimaneAI, please use the mailing list for questions regarding srsLTE in the future. Regarding your question, our eNB does not yet support intra-eNB handover.

bwchacha, 2019-11-20:
> Hey slimaneAI, please use the mailing list for questions regarding srsLTE in the future. Regarding your question, our eNB does not yet support intra-eNB handover.

hi, Frankist,

Is there any plan to support intra-eNB handover by srsepc? 
How about inter-eNB handover, X2 or other type of handover, does srsepc support any of them?
Thank you.

frankist, 2019-11-20:
Intra-eNB, S1-Handover and X2-Handover, in this order of priority, are in our plan to support. Please use this the mailing list for further questions as we use this platform to report issues

bwchacha, 2019-11-20:
Got it. Really hope handover function can be supported soon. Thanks.

andrepuschmann, 2020-09-23:
Mobility support will be added in the upcoming 20.10 release. Let's open a new issue if needed.

closed 2020-09-23 by andrepuschmann